The-OpenROAD-Project/OpenLane #1816: magic lef write takes forever

mo-hosni, 2023-05-23:
### Description

After magic update,
[commit 541c95](https://github.com/The-OpenROAD-Project/OpenLane/commit/541c95953c90428a7d716cb9095e66ba2e92896d) 
lef writing takes forever (3 hours and still stuck in writing). 

### Expected Behavior

LEF writing should not take more than a minute.

### Environment report

```shell
Kernel: Linux v4.19.0-21-cloud-amd64
Distribution: debian 10
Python: v3.7.3 (OK)
Container Engine: docker v20.10.15 (OK)
OpenLane Git Version: 541c95953c90428a7d716cb9095e66ba2e92896d
pip: INSTALLED
python-venv: INSTALLED
---
PDK Version Verification Status: OK
---
Git Log (Last 3 Commits)

541c9595 2023-05-22T17:13:52+03:00 [BOT] Update magic (#1778) - Openlane Bot -  (HEAD -> master, tag: 2023.05.23, origin/master, origin/HEAD)
791366fe 2023-05-22T14:01:37+03:00 Fix bugs when reading views for STA (#1809) - Kareem Farid -  ()
e5e58b64 2023-05-22T13:59:40+03:00 set `SCRIPTS_DIR` after saving an environment snapshot (#1810) - Kareem Farid -  ()
---
Git Remotes

origin  git@github.com:The-OpenROAD-Project/OpenLane.git (fetch)
origin  git@github.com:The-OpenROAD-Project/OpenLane.git (push)
```

### Reproduction material

There is no reproducible, this is the openlane directory of the project. 
https://github.com/efabless/caravel/tree/caravan-mpw9-PnR/openlane/caravan_core

### Relevant log output

```shell
[INFO]: Writing abstract LEF
Generating LEF output /home/hosni/caravan/caravan-mpw9-PnR/caravel/openlane/caravan_core/runs/23_05_22_22_47/results/signoff/caravan_core.lef for cell caravan_core:
Diagnostic:  Write LEF header for cell caravan_core
Diagnostic:  Writing LEF output for cell caravan_core
Diagnostic:  Scale value is 0.005000
```

kareefardi, 2023-05-23:
Can you upload the GDS generated, before writing a LEF, and `config_in.tcl` from your run?

mo-hosni, 2023-05-23:
Sure. 
GDS: https://github.com/efabless/caravel/blob/caravan-mpw9-PnR/gds/caravan_core.gds.gz
config: https://github.com/efabless/caravel/blob/caravan-mpw9-PnR/openlane/caravan_core/config.tcl

kareefardi, 2023-05-23:
Seems like magic is taking too long on designs that somewhat have large area. Housekeeping didn't take much time but a user_project_wrapper is taking a while. I think we should go back to an older version until this is fixed.

kareefardi, 2023-05-23:
@RTimothyEdwards I believe that this was caused by fixing this issue: https://github.com/RTimothyEdwards/magic/issues/236

RTimothyEdwards, 2023-06-07:
I'm not sure what you mean by "too long".  It does take a long time to generate LEF for a large layout due to the need to trace every pin to check for antenna area.  There was an issue posted to the github issue tracker for magic by Sylvain Munaut for the same thing.  I responded with a suggestion that options can be added to skip antenna checks, and the time to process might be reduced significantly by not checking power supplies, which means another new option to specifiy nets to ignore when running antenna area checks.

RTimothyEdwards, 2023-06-07:
Does the LEF generation need to be done often such that the processing time is a blocker?

kareefardi, 2023-06-13:
The above GDS file took about 6 hours to generate an abstract lef. I believe there is room for optimization. Until then, checking antenna area should be an option.

closed 2023-06-18 by kareefardi