The-OpenROAD-Project/OpenLane #1819: Error at the routing stage "Routing congestion too high"

Pa1mantri, 2023-05-24:
### Description

After the CTS stage generated all the required reports, the routing stage results in an error stating "Routing congestion is too high. Check the congestion heat map in the GUI" 

### Expected Behavior

Any changes in the configurations that I should do for the successful completion of the routing stage.

### Environment report

```shell
Kernel: Linux v5.19.0-41-generic

Distribution: ubuntu 22.04

Python: v3.10.6 (OK)

Container Engine: docker v20.10.21 (OK)

OpenLane Git Version: 54bfaa3d0e826fe64bdc7e6e1049f3600fc28942

pip: INSTALLED

python-venv: INSTALLED

---

PDK Version Verification Status: FAILED

/home/pa1mantri/OpenLane/pdks/sky130A not found.

Traceback (most recent call last):

  File "/home/pa1mantri/OpenLane/dependencies/verify_versions.py", line 76, in verify_versions

    raise Exception(f"{pdk_dir} not found.")

Exception: /home/pa1mantri/OpenLane/pdks/sky130A not found.

Failed to verify sky130A.

---

Git Log (Last 3 Commits)

54bfaa3d 2023-05-11T21:01:58+03:00 [BOT] Update openroad_app (#1774) - Openlane Bot -  (HEAD -> master, tag: 2023.05.12, origin/master, origin/HEAD)

587fe72b 2023-05-11T14:51:16+03:00 [BOT] Update PDK (#1712) - Openlane Bot -  ()

5a291005 2023-05-10T22:30:22+03:00 Integrate back OpenROAD STA (#1784) - Kareem Farid -  (tag: 2023.05.11)

---

Git Remotes

origin	https://github.com/The-OpenROAD-Project/OpenLane (fetch)

origin	https://github.com/The-OpenROAD-Project/OpenLane (push)
```

### Reproduction material

[issue_reproducible.tar.gz](https://github.com/The-OpenROAD-Project/OpenLane/files/11551303/issue_reproducible.tar.gz)

### Relevant log output

```shell
OpenROAD 7f6c37aa57467242807155c654deb350022d75c1 
This program is licensed under the BSD-3 license. See the LICENSE file for details.
Components of this program may be licensed under more restrictive licenses which must be honored.
[INFO]: Reading ODB at '/openlane/designs/picorv32a/runs/RUN_2023.05.18_09.51.23/results/cts/picorv32.odb'…
define_corners Typical
read_liberty -corner Typical /openlane/designs/picorv32a/runs/RUN_2023.05.18_09.51.23/tmp/synthesis/resizer_sky130_fd_sc_hd__typical.lib
[WARNING STA-0163] /openlane/designs/picorv32a/runs/RUN_2023.05.18_09.51.23/tmp/synthesis/resizer_sky130_fd_sc_hd__typical.lib line 23, default_fanout_load is 0.0.
[INFO]: Setting signal min routing layer to: met1 and clock min routing layer to met1. 
[INFO]: Setting signal max routing layer to: met5 and clock max routing layer to met5. 
-congestion_iterations 50 -verbose
[INFO GRT-0020] Min routing layer: met1
[INFO GRT-0021] Max routing layer: met5
[INFO GRT-0022] Global adjustment: 30%
[INFO GRT-0023] Grid origin: (0, 0)
[INFO GRT-0043] No OR_DEFAULT vias defined.
[INFO GRT-0088] Layer li1     Track-Pitch = 0.4600  line-2-Via Pitch: 0.3400
[INFO GRT-0088] Layer met1    Track-Pitch = 0.3400  line-2-Via Pitch: 0.3400
[INFO GRT-0088] Layer met2    Track-Pitch = 0.4600  line-2-Via Pitch: 0.3500
[INFO GRT-0088] Layer met3    Track-Pitch = 0.6800  line-2-Via Pitch: 0.6150
[INFO GRT-0088] Layer met4    Track-Pitch = 0.9200  line-2-Via Pitch: 1.0400
[INFO GRT-0088] Layer met5    Track-Pitch = 3.4000  line-2-Via Pitch: 3.1100
[INFO GRT-0019] Found 285 clock nets.
[INFO GRT-0001] Minimum degree: 2
[INFO GRT-0002] Maximum degree: 17
[INFO GRT-0003] Macros: 0
[INFO GRT-0004] Blockages: 32512

[INFO GRT-0053] Routing resources analysis:
          Routing      Original      Derated      Resource
Layer     Direction    Resources     Resources    Reduction (%)
---------------------------------------------------------------
li1        Vertical            0             0          0.00%
met1       Horizontal     126400         61746          51.15%
met2       Vertical        94800         62730          33.83%
met3       Horizontal      63200         44106          30.21%
met4       Vertical        37920         24125          36.38%
met5       Horizontal      12640          6084          51.87%
---------------------------------------------------------------

[INFO GRT-0101] Running extra iterations to remove overflow.
[INFO GRT-0103] Extra Run for hard benchmark.
[INFO GRT-0197] Via related to pin nodes: 49970
[INFO GRT-0198] Via related Steiner nodes: 8906
[INFO GRT-0199] Via filling finished.
[INFO GRT-0111] Final number of vias: 148681
[INFO GRT-0112] Final usage 3D: 1194987
[ERROR GRT-0118] Routing congestion too high. Check the congestion heatmap in the GUI.
Error: resizer_routing_design.tcl, 39 GRT-0118
```

vijayank88, 2023-05-24:
@Pa1mantri 
Unable to run the test case attached.
```
openroad> source run.tcl 
Error: resizer_routing_design.tcl, 14 can't read "::env(SCRIPTS_DIR)": no such variable
while evaluating {source run.tcl }
```
Can you attach the right test case?

Pa1mantri, 2023-05-24:
Hey Vijayan, Thank you for the reply. I am new to using OPENLANE tools and Github. Can you be specific when you say test case. What do you want me to attach. Thanks in advance.

kareefardi, 2023-05-24:
```
openroad> source run.tcl 
Error: resizer_routing_design.tcl, 14 can't read "::env(SCRIPTS_DIR)": no such variable
while evaluating {source run.tcl }
```
This is caused by a bug that was fixed in https://github.com/The-OpenROAD-Project/OpenLane/pull/1810. @Pa1mantri can you update OpenLane, re-run your design and upload a new reproducible

Pa1mantri, 2023-05-26:
@kareefardi Hi Kareem, sorry for the delayed response. Here is the new reproducible after updating the OpenLane.

[issue_reproducible.tar.gz](https://github.com/The-OpenROAD-Project/OpenLane/files/11571724/issue_reproducible.tar.gz)

vijayank88, 2023-05-26:
> @kareefardi Hi Kareem, sorry for the delayed response. Here is the new reproducible after updating the OpenLane.
> 
> [issue_reproducible.tar.gz](https://github.com/The-OpenROAD-Project/OpenLane/files/11571724/issue_reproducible.tar.gz)

still same error. Unable to find `SCRIPT_DIR`

Pa1mantri, 2023-05-26:
Hi, I have deleted the entire folder and cloned the repository again. When I am running the flow without any changes in the design, not facing any issues at the routing stage. Here, I am running the flow with a custom cell(inverter) incorporated into the design by making some changes to the config.json file. This is the file after the changes. 

{

    "DESIGN_NAME": "picorv32",

    "VERILOG_FILES": "dir::src/picorv32a.v",

    "CLOCK_PORT": "clk",

    "CLOCK_NET": "clk",

    "GLB_RESIZER_TIMING_OPTIMIZATIONS": true,

    "GPL_CELL_PADDING": 2,

    "DPL_CELL_PADDING": 2,

    "CLOCK_PERIOD": 24,

    "FP_CORE_UTIL": 35,

    
     "PL_RANDOM_GLB_PLACEMENT": 1,

    "PL_TARGET_DENSITY": 0.5,

    "FP_SIZING":  "relative",       

    "LIB_SYNTH":"dir::src/sky130_fd_sc_hd__typical.lib",

    "LIB_FASTEST":"dir::src/sky130_fd_sc_hd__fast.lib",

    "LIB_SLOWEST":"dir::src/sky130_fd_sc_hd__slow.lib",

    "LIB_TYPICAL":"dir::src/sky130_fd_sc_hd__typical.lib",

    "TEST_EXTERNAL_GLOB":"dir::../iiitb_rv32i/src/*",

    "SYNTH_DRIVING_CELL":"sky130_vsdinv",

    "pdk::sky130*": {

        "SYNTH_MAX_FANOUT": 6,

        "scl::sky130_fd_sc_ms": {

            "FP_CORE_UTIL": 30

        }

    }

}

vijayank88, 2023-05-26:
If there is no issue to fix, you can close this issue. @Pa1mantri

vijayank88, 2023-05-27:
For design with more than 100 std cells, `"PL_RANDOM_GLB_PLACEMENT": 1,` this should be disabled or not required add into config file.

Pa1mantri, 2023-05-27:
Great, seems to be working. Thanks for the help @vijayank88 @kareefardi. Appreciate it.

kareefardi, 2023-05-28:
This seems resolved now.

closed 2023-05-28 by kareefardi