The-OpenROAD-Project/OpenLane #1821: OpenLane flow fails with OpenFPGA design

CristopherA96, 2023-05-24:
## Description

Working with files generated by OpenFPGA, I am trying to get the top module which contains instances of each tile.

I am having the next issues, different versions were use, because with one of them worked but now both crash.

## Error with tag 5880f35

This error

[image]

Not `issue_reproducible` directory is generated.

Doing a research about this issue, I found this one [Yosys_issue_94](https://github.com/YosysHQ/yosys/issues/94), but it means that I need to do synthesis by using yosys externally to OpenLane flow, setting SYNTH_NO_FLAT variable to 1 or 0 it is the same.

## Issue with tag e3a5189

The error is shown below:

[image]

I tried setting smaller value for density as the error say, but still appears.

The issue reproducible for this shown below:

[issue_e3a5189.zip](https://github.com/The-OpenROAD-Project/OpenLane/files/11557879/issue_e3a5189.zip)

**Note:** I added issue with tag e3a5189 because with that version was working but now both fail.

## System settings

I am using:

- Linux Mint OS
-  RAM 32-bit

Thanks for your time.

kareefardi, 2023-05-25:
For the first issue, it could be an actual error of an undriven wire. `hierarchy` is called in OL before running checks as stated in the referenced yosys issue. If you believe that you can waive the error, then disable `QUIT_ON_SYNTH_CHECKS`.

For the second issue, the netlist has, non-flat, macros inside. Classic placement flow won't work in this case. 
Side note: `init_density_penalty` is different from `PL_TAREGET_DENSITY`. `init_density_penalty`

CristopherA96, 2023-05-31:
Thank you so much @kareefardi, it has been very helpful.

closed 2023-06-19 by CristopherA96