The-OpenROAD-Project/OpenLane #1823: [ERROR ODB-0370] Attempt to disconnect iterm of dont touch.

mo-hosni, 2023-05-26:
### Description

OpenROAD fails in resizer optimizations when there is a `dont touch`. This was working in older versions of OpenLane.

### Expected Behavior

Resizer should resize whatever needs resizing except for the don't touch nets as it always did before.

### Environment report

```shell
Kernel: Linux v4.19.0-21-cloud-amd64
Distribution: debian 10
Python: v3.7.3 (OK)
Container Engine: docker v20.10.15 (OK)
OpenLane Git Version: 5880f35897d61b2eee0f1bfaf57262f238597074
pip: INSTALLED
python-venv: INSTALLED
---
PDK Version Verification Status: OK
---
Git Log (Last 3 Commits)

5880f35 2023-05-24T17:30:26+03:00 Read multi-corner libs while running resizer and CTS (#1804) - Kareem Farid -  (grafted, HEAD -> master, tag: 2023.05.25, origin/master, origin/HEAD)
---
Git Remotes

origin  git@github.com:The-OpenROAD-Project/OpenLane.git (fetch)
origin  git@github.com:The-OpenROAD-Project/OpenLane.git (push)
hosni@ciic-03:~/caravan/caravel/OpenLane$
```

### Reproduction material

[issue_reproducible.tar.gz](https://github.com/The-OpenROAD-Project/OpenLane/files/11573872/issue_reproducible.tar.gz)

### Relevant log output

```shell
[STEP 10]
[INFO]: Running Placement Resizer Design Optimizations (log: caravan_core/runs/23_05_26_02_37/logs/placement/10-resizer.log)...
[ERROR]: during executing openroad script /openlane/scripts/openroad/resizer.tcl
[ERROR]: Log: caravan_core/runs/23_05_26_02_37/logs/placement/10-resizer.log
[ERROR]: Last 10 lines:
[INFO] Instance 'pll.ringosc.iss.delayenb0' matched don't touch regular expression, setting as don't touch...                                                 
[INFO] Instance 'pll.ringosc.iss.delayenb1' matched don't touch regular expression, setting as don't touch...                                                 
[INFO] Instance 'pll.ringosc.iss.delayint0' matched don't touch regular expression, setting as don't touch...                                                 
[INFO] Instance 'pll.ringosc.iss.reseten0' matched don't touch regular expression, setting as don't touch...                                                  
[INFO RSZ-0027] Inserted 31 input buffers.                                                                                                                    
[INFO RSZ-0028] Inserted 367 output buffers.                                                                                                                  
[WARNING RSZ-0065] max wire length less than 4125u increases wire delays.                                                                                     
[ERROR ODB-0370] Attempt to disconnect iterm of dont_touch instance pll.ringosc.iss.reseten0                                                                  
Error: resizer.tcl, 55 ODB-0370                                                                                                                               
child process exited abnormally
```

kareefardi, 2023-06-07:
The issue reproducible is wrong due to a bug in a certain past version of OpenLane that got fixed. I ran into the same resizer issue using a different design later. So here is a functional reproducible.  
[issue_reproducible.tar.gz](https://github.com/The-OpenROAD-Project/OpenLane/files/11678581/issue_reproducible.tar.gz)
@openroadie can you take another look?

mo-hosni, 2023-06-15:
@openroadie @maliberty Is there any progress on this?
Since this issue, we are using an old version (March 2023) for OpenLane and OpenROAD for all caravel activities because of this issue. We cannot update either until this is resolved.

vijayank88, 2023-06-15:
@kareefardi 
`set_dont_touch net/instance` either one automatically added to dont touch list.
As per the log
```
[INFO] Instance 'dll.ringosc.iss.delayenb0' matched don't touch regular expression, setting as don't touch...
[INFO] Instance 'dll.ringosc.iss.delayenb1' matched don't touch regular expression, setting as don't touch...
[INFO] Instance 'dll.ringosc.iss.delayint0' matched don't touch regular expression, setting as don't touch...
[INFO] Instance 'dll.ringosc.iss.reseten0' matched don't touch regular expression, setting as don't touch...
[INFO RSZ-0027] Inserted 108 input buffers.
[INFO RSZ-0028] Inserted 396 output buffers.
[WARNING RSZ-0065] max wire length less than 4269u increases wire delays.
[ERROR ODB-0370] Attempt to disconnect iterm of dont_touch instance dll.ringosc.iss.reseten0
Error: resizer.tcl, 55 ODB-0370
```
`ringosc` nets already moved to dont touch list and again trying to access instance name `ringosc` due to the OpenLane flow script latest update: https://github.com/The-OpenROAD-Project/OpenLane/blob/5ca48214640dcb9f88945a3861be47716ebe187d/scripts/openroad/common/resizer.tcl#L29.
If I remove that `foreach` loop for instance flow running fine.
```
[INFO] Net 'dll.ringosc.iss.d1' matched don't touch regular expression, setting as don't touch...
[INFO] Net 'dll.ringosc.iss.d2' matched don't touch regular expression, setting as don't touch...
[INFO] Net 'dll.ringosc.iss.one' matched don't touch regular expression, setting as don't touch...
[INFO RSZ-0027] Inserted 108 input buffers.
[INFO RSZ-0028] Inserted 396 output buffers.
[WARNING RSZ-0065] max wire length less than 4269u increases wire delays.
[INFO RSZ-0034] Found 1296 slew violations.
[INFO RSZ-0035] Found 491 fanout violations.
[INFO RSZ-0036] Found 463 capacitance violations.
[INFO RSZ-0037] Found 357 long wires.
[INFO RSZ-0038] Inserted 3086 buffers in 1986 nets.
[INFO RSZ-0039] Resized 16144 instances.
[INFO RSZ-0042] Inserted 20 tie sky130_fd_sc_hd__conb_1 instances.
[INFO RSZ-0042] Inserted 2 tie sky130_fd_sc_hd__conb_1 instances.
Placement Analysis
---------------------------------
total displacement      65570.9 u
average displacement        1.1 u
max displacement          220.4 u
original HPWL         1631055.9 u
legalized HPWL        1678793.0 u
delta HPWL                    3 %

[INFO DPL-0020] Mirrored 8220 instances
[INFO DPL-0021] HPWL before          1678793.0 u
[INFO DPL-0022] HPWL after           1657593.9 u
[INFO DPL-0023] HPWL delta               -1.3 %
Setting global connections for newly added cells…
Writing OpenROAD database to './tmp/placement/10-resizer.odb'…
Writing netlist to './out.v'…
Writing powered netlist to './tmp/placement/10-resizer.pnl.v'…
Writing layout to './tmp/placement/10-resizer.def'…
Writing timing constraints to './tmp/placement/10-resizer.sdc'…
area_report

===========================================================================
report_design_area
============================================================================
Design area 860642 u^2 29% utilization.
area_report_end
```

vijayank88, 2023-06-15:
While accessing following instance it is throwing an error:
```
 sky130_fd_sc_hd__einvp_1 \dll.ringosc.iss.reseten0  (.A(\dll.ringosc.iss.one ),
    .TE(\dll.ireset ),
    .VGND(VGND),
    .VNB(VGND),
    .VPB(VPWR),
    .VPWR(VPWR),
    .Z(\dll.ringosc.dstage[0].id.in ));
```
@mo-hosni your requirement is `net` or `instance` or `both` to add into `set_dont_touch`?

openroadie, 2023-06-15:
@mo-hosni I will take a look. I guess you are looking for the resizer to not terminate abnormally ?

openroadie, 2023-06-20:
I tried to run the provided testcase and here is the output:

```
openroad run.tcl 
OpenROAD 0048a8121ba928841891df877f811ddaea9075ff 
This program is licensed under the BSD-3 license. See the LICENSE file for details.
Components of this program may be licensed under more restrictive licenses which must be honored.
Error: resizer.tcl, 14 can't read "::env(SCRIPTS_DIR)": no such variable
openroad> 
```

@mo-hosni  Can we please get a test case that demonstrates the problem.

vijayank88, 2023-06-21:
> The issue reproducible is wrong due to a bug in a certain past version of OpenLane that got fixed. I ran into the same resizer issue using a different design later. So here is a functional [...]

@openroadie Try with Kareem's attachment.

openroadie, 2023-06-21:
Thank you @vijayank88 . I can see the issue now :). Will take a deeper look.

openroadie, 2023-06-22:
So here is what I think the problem is. We are fixing  fanout on this net:

Net:  **dll.ireset**
Instance: **dll.ringosc.iss.reseten0**
Pin: **dll.ringosc.iss.reseten0/TE** 

The check for fan out fixing checks that the net it not a don't touch (and it is not) but then it has to connect the pin to a different instance (and this is where it throws an error). I think we should either mark the net dll.ireset as don't touch or we can unmark the instance involved.

@maliberty @vijayank88 @mo-hosni : What do you think?

maliberty, 2023-06-22:
I don't think we should unmark an instance as dont_touch as that's user intent.  Making the net dont_touch is possible but adds extra restriction.  

Let's just not disconnect the dont_touch instance.  Unless there are so many of them that we can't fix max fanout it isn't a problem (and if there are then the problem is unfixable).

openroadie, 2023-06-22:
@maliberty we can do that but then you run the risk of the instance being connected to a buffer that is far away... but I guess since this is a reset line it does not matter all that much.

maliberty, 2023-06-22:
You are free to modify the net attached to the dont_touch instance - you just can't disconnect it.  So when inserting the buffer you need to take some care about which is the new net and which is the old.

maliberty, 2023-06-22:
I'm pretty sure this has come up before so I wouldn't be surprised if there is some code trying to deal with this already.

openroadie, 2023-06-22:
I can skip the pins that connect to don't touch instances in the code. Maybe we got lucky before? Anyways, small PR ready in another hour or less.

openroadie, 2023-06-22:
https://github.com/The-OpenROAD-Project/OpenROAD/pull/3505

kareefardi, 2023-06-26:
Closing this as OpenROAD no longer throws this error.

closed 2023-06-26 by kareefardi