The-OpenROAD-Project/OpenLane #1827: yosys synth script inserts buffers after tristates

mattvenn, 2023-05-29:
### Description

We are using tristates in the design. We noticed while running STA that a buffer followed the tristate.
We already have the option turned off in the config to add output buffers:

`set ::env(PL_RESIZER_BUFFER_OUTPUT_PORTS) `

We tracked it down to the yosys synth script calling insbuf:

```   splitnets
    opt_clean -purge
    insbuf -buf {*}$::env(SYNTH_MIN_BUF_PORT)

    set stat_libs ""
```

We are able to work around it with this patch: https://github.com/TinyTapeout/tinytapeout-03p5/blob/main/tt_mux.patch 
And here's the tt_mux source: https://github.com/TinyTapeout/tt-multiplexer/blob/main/proto/tt_mux.v 

### Expected Behavior

openlane shouldn't insert buffers after tristates. 

### Environment report

```shell
Kernel: Linux v5.15.0-71-generic
Distribution: ubuntu 20.04
Python: v3.8.10 (OK)
Container Engine: docker v20.10.21 (OK)
OpenLane Git Version: 4cd0986b3ae550cdf7a6d0fba4e0657012f635d8
pip: INSTALLED
python-venv: INSTALLED
---
PDK Version Verification Status: OK
---
Git Log (Last 3 Commits)

4cd0986 2023-02-13T14:32:59+02:00 Bugfixes (#1658) - Kareem Farid -  (grafted, HEAD, tag: 2023.02.14)
---
Git Remotes

origin  https://github.com/The-OpenROAD-Project/OpenLane (fetch)
origin  https://github.com/The-OpenROAD-Project/OpenLane (push)
```

### Reproduction material

n/a

### Relevant log output

```shell
n/a
```

kareefardi, 2023-05-29:
Please provide any steps to reproduce and show where was the buffer inserted.

mattvenn, 2023-05-30:
the outputs should be <some logic> -> <tristate> -> <output pin>

but instead they are <some logic> -> <tristate> -> <buffer> -> <output pin>

closed 2025-09-15 by donn