The-OpenROAD-Project/OpenLane #1835: test_sram_macro demo design fails

mguthaus, 2023-06-04:
### Description

There are three issues (so far) why it fails:
1) Bitwidths are not specified for single bit constants in the Verilog file.
Can be worked around by changing them to "1'b0" and "1'b1" for clk1, csb1, and addr1

2) The corners aren't defined for the SRAM lib:
Can be worked around by adding "/// sta-blackbox" in the Verilog file.

3) The power and ground straps are too close to the pins and prevent detailed routing from passing.
[image]

### Expected Behavior

It should be able to route the two macros. I tried modifying `FP_PDN_HORIZONTAL_HALO` and `FP_PDN_VERTICAL_HALO` but these don't seem to have any effect.

### Environment report

```shell
Kernel: Linux v6.0.12-76060012-generic
Distribution: pop 20.04
Python: v3.8.10 (OK)
Container Engine: docker v23.0.1 (OK)
OpenLane Git Version: 2ec3bfdd25143ee6b1973d31fdbb612f838594cb
pip: INSTALLED
python-venv: INSTALLED
---
PDK Version Verification Status: MISMATCH
The version of open_pdks used in building the PDK does not match the version OpenLane was tested on (installed: e6f9c8876da77220403014b116761b0b2d79aab4, tested: af3485525297d5cbe93c129ea853da2d588fac41)
This may introduce some issues. You may want to re-install the PDK by invoking `make pdk`.
---
Git Log (Last 3 Commits)

2ec3bfdd 2023-05-16T21:06:51+03:00 Do not skip STA when `CLOCK_PORT` is not specified (#1797) - Kareem Farid -  (HEAD -> master, tag: 2023.05.17, origin/master, origin/HEAD)
54bfaa3d 2023-05-11T21:01:58+03:00 [BOT] Update openroad_app (#1774) - Openlane Bot -  (tag: 2023.05.12)
587fe72b 2023-05-11T14:51:16+03:00 [BOT] Update PDK (#1712) - Openlane Bot -  ()
---
Git Remotes

origin  https://github.com/The-OpenROAD-Project/OpenLane (fetch)
origin  https://github.com/The-OpenROAD-Project/OpenLane (push)
```

### Reproduction material

diff --git a/designs/test_sram_macro/src/test_sram_macro.v b/designs/test_sram_macro/src/test_sram_macro.v
index 685a36fe..fdd433b8 100644
--- a/designs/test_sram_macro/src/test_sram_macro.v
+++ b/designs/test_sram_macro/src/test_sram_macro.v
@@ -41,9 +41,9 @@ sky130_sram_1kbyte_1rw1r_32x256_8 sram0(
     .din0(datain[31:0]),
     .dout0(dataout_int[31:0]),

-    .clk1(0),
-    .csb1(1),
-    .addr1(0),
+    .clk1(1'b0),
+    .csb1(1'b1),
+    .addr1(1'b0),
     .dout1(dout1[31:0])
 );

@@ -56,9 +56,9 @@ sky130_sram_1kbyte_1rw1r_32x256_8 sram1(
     .din0(datain[63:32]),
     .dout0(dataout_int[63:32]),

-    .clk1(0),
-    .csb1(1),
-    .addr1(0),
+    .clk1(1'b0),
+    .csb1(1'b1),
+    .addr1(1'b0),
     .dout1(dout1[63:32])
 );

@@ -79,4 +79,4 @@ module test_sram_macro(

 test_sram_macro_unwrapped submodule (.*);

-endmodule
\ No newline at end of file
+endmodule

also add "/// sta-blackbox" to the SRAM macro:
/openlane/pdks/sky130B/libs.ref/sky130_sram_macros/verilog/sky130_sram_1kbyte_1rw1r_32x256_8.v

### Relevant log output

```shell
[INFO]: Running Detailed Routing (log: designs/test_sram_macro/runs/foo/logs/routing/24-detailed.log)...
[ERROR]: There are violations in the design after detailed routing.
[ERROR]: Total Number of violations is 12
[ERROR]: Step 24 (routing) failed with error:
-code 1 -level 0 -errorcode NONE -errorinfo {
    while executing
"throw_error"
    (procedure "quit_on_tr_drc" line 7)
    invoked from within
"quit_on_tr_drc"
    (procedure "detailed_routing_tritonroute" line 19)
    invoked from within
"detailed_routing_tritonroute {*}$args"
    (procedure "detailed_routing" line 2)
    invoked from within
"detailed_routing"
    (procedure "run_routing" line 35)
    invoked from within
"run_routing"
    (procedure "run_routing_step" line 7)
    invoked from within
"run_routing_step"} -errorline 1
[
```

vijayank88, 2023-06-05:
it fails during synthesis itself with latest master
```
[INFO]: Running Verilator (log: designs/test_sram_macro/runs/gh1835/logs/synthesis/verilator.log)...
[ERROR]: 3 errors found by Verilator
[ERROR]: Step 0 (verilator_lint_check) failed with error:
-code 1 -level 0 -errorcode NONE -errorinfo {
    while executing
"throw_error"
    (procedure "run_verilator" line 35)
    invoked from within
"run_verilator"
    (procedure "run_verilator_step" line 3)
    invoked from within
"run_verilator_step"} -errorline 1
[INFO]: Calculating runtime...
[INFO]: Saving current set of views in 'designs/test_sram_macro/runs/gh1835/results/final'...
[INFO]: Generating final set of reports...
[INFO]: Created manufacturability report at 'designs/test_sram_macro/runs/gh1835/reports/manufacturability.rpt'.
[INFO]: Created metrics report at 'designs/test_sram_macro/runs/gh1835/reports/metrics.csv'.
[INFO]: Saving runtime environment...
[ERROR]: Flow failed.
```

mguthaus, 2023-06-05:
It seems like this is a recurring bug that was also in https://github.com/The-OpenROAD-Project/OpenLane/issues/640

If I add 

`set ::halo [list $::env(FP_PDN_HORIZONTAL_HALO) $::env(FP_PDN_VERTICAL_HALO)]
`
before the th emacro "define_pdn_grid -macro " in pdn_cfg.tcl, it seems to work. So, pdngen is not using the environment values supplied to the command.

mguthaus, 2023-06-05:
> it fails during synthesis itself with latest master

This is due to issue 1 with the int constants.
```
Bitwidths are not specified for single bit constants in the Verilog file.
Can be worked around by changing them to "1'b0" and "1'b1" for clk1, csb1, and addr1
```

kareefardi, 2023-06-06:
@vijayank88 I attached a test case. [issue_reproducible.tar.gz](https://github.com/The-OpenROAD-Project/OpenLane/files/11664371/issue_reproducible.tar.gz). Varying halo values has no impact on the inserted pdn. Furthermore, upon disabling the tapdecap insertion met1 straps were passing through the met1 obstruction in the macro. See the screenshot:   
[image]

vijayank88, 2023-06-06:
@mguthaus 
Implemented your changes and still it fail during verilator
```
%Error-NEEDTIMINGOPT: //openlane/pdks/sky130B/libs.ref/sky130_sram_macros/verilog/sky130_sram_1kbyte_1rw1r_32x256_8.v:58:5: Use --timing or --no-timing to specify how delays should be handled
                                                                                                                          : ... In instance test_sram_macro.submodule.sram1
   58 |     #(T_HOLD) dout0 = 32'bx;
      |     ^
                      ... For error description see https://verilator.org/warn/NEEDTIMINGOPT?v=5.009
%Error-NEEDTIMINGOPT: //openlane/pdks/sky130B/libs.ref/sky130_sram_macros/verilog/sky130_sram_1kbyte_1rw1r_32x256_8.v:76:5: Use --timing or --no-timing to specify how delays should be handled
                                                                                                                          : ... In instance test_sram_macro.submodule.sram1
   76 |     #(T_HOLD) dout1 = 32'bx;
      |     ^
%Error-NEEDTIMINGOPT: //openlane/pdks/sky130B/libs.ref/sky130_sram_macros/verilog/sky130_sram_1kbyte_1rw1r_32x256_8.v:104:17: Use --timing or --no-timing to specify how timing controls should be handled
                                                                                                                            : ... In instance test_sram_macro.submodule.sram1
  104 |        dout0 <= #(DELAY) mem[addr0_reg];
      |                 ^
%Error-NEEDTIMINGOPT: //openlane/pdks/sky130B/libs.ref/sky130_sram_macros/verilog/sky130_sram_1kbyte_1rw1r_32x256_8.v:112:17: Use --timing or --no-timing to specify how timing controls should be handled
                                                                                                                            : ... In instance test_sram_macro.submodule.sram1
  112 |        dout1 <= #(DELAY) mem[addr1_reg];
      |                 ^
%Warning-WIDTHEXPAND: //openlane/designs/test_sram_macro/src/test_sram_macro.v:46:6: Input port connection 'addr1' expects 8 bits on the pin connection, but pin connection's CONST '1'h0' generates 1 bits.
                                                                                   : ... In instance test_sram_macro.submodule
   46 |     .addr1(1'b0),
      |      ^~~~~
                      ... Use "/* verilator lint_off WIDTHEXPAND */" and lint_on around source to disable this message.
%Warning-WIDTHEXPAND: //openlane/designs/test_sram_macro/src/test_sram_macro.v:61:6: Input port connection 'addr1' expects 8 bits on the pin connection, but pin connection's CONST '1'h0' generates 1 bits.
                                                                                   : ... In instance test_sram_macro.submodule
   61 |     .addr1(1'b0),
      |      ^~~~~
%Error: Exiting due to 4 error(s)
```

mguthaus, 2023-06-06:
@vijayank88 I added a PR so that this can be replicated:
https://github.com/The-OpenROAD-Project/OpenLane/pull/1836

mguthaus, 2023-06-06:
@vijayank88 This also required a change to synthesis.tcl to enable --timing for verilator. Or, you can make the SRAM Verilog a blackbox with no contents.

mguthaus, 2023-06-06:
This is another test that is basically the same as test_sram_macro but with the macro locally rather than in the PDK.
[mytest.tar.gz](https://github.com/The-OpenROAD-Project/OpenLane/files/11664927/mytest.tar.gz)

kareefardi, 2023-06-06:
@vijayank88 For the sake of the testcase you can disable `QUIT_ON_VERILATOR_ERRORS`. This what I did in order to generate the testcase above.

vijayank88, 2023-06-06:
Resolved those verilator errors and get struck at global routing congestion. I ran `issue_reproducible` with latest OpenROAD commit, it passed GRT stage.

vijayank88, 2023-06-06:
@kareefardi 
How do I update OpenLane with latest OpenROAD commit?
Because current OpenROAD commit throws GRT congestion.

maliberty, 2023-06-06:
@gadfort any comments on -halo?

gadfort, 2023-06-06:
@maliberty `-halo` should be honored. I can take a look to see if there is something wrong when I look at the other testcase later today

gadfort, 2023-06-06:
@maliberty @mguthaus @kareefardi there is something very wrong with this layout. I noticed in the picture that the rows appear to be going through the macros. Looking at the layout there are sites defined where the macros are placed, so PDN is attempting to connect to those rows. You need to ensure you cut the rows after placement otherwise PDNgen doesn't really know how to proceed.

[image]

kareefardi, 2023-06-06:
> @maliberty @mguthaus @kareefardi there is something very wrong with this layout. I noticed in the picture that the rows appear to be going through the macros. Looking at the layout there are sites [...]

@gadfort I was experimenting and disabled tapdecap insertion. I guess this resulted in no row cuts. That's good to know.

If needed, here is a cleaner testcase with tapdecap insertion enabled and consequently row cuts.  
[issue_reproducible.tar.gz](https://github.com/The-OpenROAD-Project/OpenLane/files/11666604/issue_reproducible.tar.gz)

mguthaus, 2023-06-06:
@gadfort can you look at my testcase instead? 
https://github.com/The-OpenROAD-Project/OpenLane/issues/1835#issuecomment-1578726623

gadfort, 2023-06-06:
@mguthaus @kareefardi  Could you check if: https://github.com/The-OpenROAD-Project/OpenROAD/pull/3432 works for you.
The halo was getting lost in the mix.

kareefardi, 2023-06-06:
Looks like the halo is applied to met4 strips correctly. I am wondering though if it should also apply to the metal rails. If not can you describe how is the halo considered by PDNgen

mguthaus, 2023-06-06:
@kareefardi How do you check this so quickly? I'm still building a custom OpenROAD pdngen and openlane image... is there an automated shortcut I'm missing?

gadfort, 2023-06-06:
> 

@kareefardi I believe the followpins are allowed to enter the halo since they are following the standard cell rows (if we cut them, then instances would be disconnected from power or ground).
The halo is applied to the obstructions in the macro and used to trim back the straps and other power grid objects (followpin being a little special).

kareefardi, 2023-06-06:
> @kareefardi How do you check this so quickly? I'm still building a custom OpenROAD pdngen and openlane image... is there an automated shortcut I'm missing?

@mguthaus Once I have the reproducible. I use a local build of OpenROAD that I update regularly. Updating OpenROAD doesn't take much time especially if I have build files from past builds.

> @kareefardi I believe the followpins are allowed to enter the halo since they are following the standard cell rows (if we cut them, then instances would be disconnected from power or ground). The [...]

@gadfort Got it. Thank you.

mguthaus, 2023-06-07:
Can confirm it looks good for me too. Thanks!

closed 2023-06-07 by mguthaus

stevvik, 2023-06-27:
> @maliberty @mguthaus @kareefardi there is something very wrong with this layout. I noticed in the picture that the rows appear to be going through the macros. Looking at the layout there are sites [...]

So, finally, is this a normal behavior that after applying "add_pdn_connect" command I'm getting metal1 stripes through the macro? These metal1 stripes shorts everything in my macro. And I don't know how to "cut the rows after placement" under the macro as yuo proposed.

maliberty, 2023-06-27:
The rows should have been cut before pdn is run.  Normally it happens during tapcell insertion

stevvik, 2023-06-27:
Thank you, I got it! I have no tapcells in my technology. I'll try filler cell as tapcell.

stevvik, 2023-06-27:
But anyway, description of "add_pdn_connect" says nothing about automatic stripes stretching. :(

maliberty, 2023-06-27:
add_pdn_connect is not stretching any stripes.  Stripes cover all valid placement sites.  Its just that some of your 'valid' sites aren't really valid.

maliberty, 2023-06-28:
In general we connect from above but if a top level stripe exactly overlaps a block stripe then we allow it.  However it is up to you to ensure they exactly align.