The-OpenROAD-Project/OpenLane #1838: Redundant verilog blackbox options

mguthaus, 2023-06-06:
### Description

To flag a Verilog module as blackbox, you need two things:
1) specify it in `VERILOG_FILES_BLACKBOX`
2) add a comment of `/// sta-blackbox`

These seem redundant. Why not either blackbox all the files in the TCL list or look for the comment in all Verilog files?

### Expected Behavior

Use either `VERILOG_FILES_BLACKBOX` *or* `sta-blackbox` but not both.

### Environment report

```shell
Not relevant
```

### Reproduction material

N/A

### Relevant log output

```shell
N/A
```

kareefardi, 2023-06-06:
Please use "Feature Requests and Enhancements" github template next time.
The reason for having both is that `/// sta-blackbox` is a fallback for a special case, when the included file is RTL and not a gatelevel netlist. If we blackbox designs in `VERILOG_FILES_BLACKBOX` during sta, it would make hierarchical sta practically not functional since all the macros will be blackboxed.

mguthaus, 2023-06-06:
I think that I'm missing something.  A gatelevel netlist in hierarchical sta won't get blackboxed unless that comment is present also. Both conditions are required to blackbox a module: 1) in the list 2) have the comment.

kareefardi, 2023-06-06:
> I think that I'm missing something. A gatelevel netlist in hierarchical sta won't get blackboxed unless that comment is present also. Both conditions are required to blackbox a module: 1) in the list 2) have the comment.

If we remove `/// sta-blackbox` then we would blackbox all modules in `VERILOG_FILES_BLACKBOX`. Then we will run into the issue I mentioned above. The other case is removing `VERILOG_FILES_BLACKBOX`.

The other option is to replace and deprecate `VERILOG_FILES_BLACKBOX` which I think would be a bit of unnecessary challenge. 

I understand the confusion but we need a solution that would allow selection of specific modules to blackbox during sta. We considered adding another variable but thought that it would be more confusing.

mguthaus, 2023-06-06:
But you only add files to `VERILOG_FILES_BLACKBOX` if you wanted them to be black-boxed. If not, don't add them and they won't be black-boxed. Are people trying to just add all Verilog files to `VERILOG_FILES_BLACKBOX`?

kareefardi, 2023-06-06:
To blackbox the macro during synthesis and PnR but not for hierarchical sta.

mguthaus, 2023-06-06:
Ohhh, I get it. Thanks.

mguthaus, 2023-06-06:
One more question though:
Why does a macro have to be gate-level if there is a .lib file corresponding to it? For memories, for example, you usually have a behavioral model and not a gate-level model. If I make this an "sta-blackbox" then it doesn't use the lib file for timing in it.

closed 2023-06-06 by mguthaus

mguthaus, 2023-06-06:
My problem is that the read_verilog fails on the behavioral SRAM Verilog files because they use parameters. I had to convert all my parameters to constants and I can read the SRAM Verilog files without sta-blackbox.