The-OpenROAD-Project/OpenLane #1842: Error in routing for picorv32a

mariamrakka, 2023-06-07:
### Description

I am running the design flow example using picorv32a design. I am using commit e910d115dc75c51407f652330bce8ed997c45946 and I have all steps successful up to run_routing, where I am getting the below log of error during detailed routing (note that I am using pdks i have installed separately in open_pdks directory):

```
OpenROAD 9a713f0e8a51769207918da28adfd3c02b854375 
This program is licensed under the BSD-3 license. See the LICENSE file for details.
Components of this program may be licensed under more restrictive licenses which must be honored.
[INFO]: Reading ODB at '/openlane/designs/picorv32a/runs/RUN_2023.06.03_07.02.57/tmp/routing/25-fill.odb'…
define_corners Typical
read_liberty -corner Typical /home/mariam/OpenLane/designs/picorv32a/src/sky130_fd_sc_hd__typical.lib
[WARNING STA-0163] /home/mariam/OpenLane/designs/picorv32a/src/sky130_fd_sc_hd__typical.lib line 23, default_fanout_load is 0.0.
[INFO ORD-0030] Using 2 thread(s).
[INFO DRT-0149] Reading tech and libs.

Units:                1000
Number of layers:     13
Number of macros:     442
Number of vias:       25
Number of viarulegen: 25

[INFO DRT-0150] Reading design.

Design:                   picorv32
Die area:                 ( 0 0 ) ( 546215 556935 )
Number of track patterns: 12
Number of DEF vias:       0
Number of components:     39492
Number of terminals:      411
Number of snets:          2
Number of nets:           10163

[INFO DRT-0167] List of default vias:
  Layer mcon
    default via: L1M1_PR
  Layer via
    default via: M1M2_PR
  Layer via2
    default via: M2M3_PR
  Layer via3
    default via: M3M4_PR
  Layer via4
    default via: M4M5_PR
[INFO DRT-0162] Library cell analysis.
[INFO DRT-0163] Instance analysis.
  Complete 10000 instances.
  Complete 20000 instances.
  Complete 30000 instances.
[INFO DRT-0164] Number of unique instances = 265.
[INFO DRT-0168] Init region query.
[INFO DRT-0018]   Complete 10000 insts.
[INFO DRT-0018]   Complete 20000 insts.
[INFO DRT-0018]   Complete 30000 insts.
[INFO DRT-0024]   Complete FR_MASTERSLICE.
[INFO DRT-0024]   Complete Fr_VIA.
[INFO DRT-0024]   Complete li1.
[INFO DRT-0024]   Complete mcon.
[INFO DRT-0024]   Complete met1.
[INFO DRT-0024]   Complete via.
[INFO DRT-0024]   Complete met2.
[INFO DRT-0024]   Complete via2.
[INFO DRT-0024]   Complete met3.
[INFO DRT-0024]   Complete via3.
[INFO DRT-0024]   Complete met4.
[INFO DRT-0024]   Complete via4.
[INFO DRT-0024]   Complete met5.
[INFO DRT-0033] FR_MASTERSLICE shape region query size = 0.
[INFO DRT-0033] FR_VIA shape region query size = 0.
[INFO DRT-0033] li1 shape region query size = 453622.
[INFO DRT-0033] mcon shape region query size = 456210.
[... 359 lines cut ...]

```shell
{
    "DESIGN_NAME": "picorv32",
    "VERILOG_FILES": "dir::src/picorv32a.v",
    "CLOCK_PORT": "clk",
    "CLOCK_NET": "clk",
    "GLB_RESIZER_TIMING_OPTIMIZATIONS": true,
    "GPL_CELL_PADDING": 2,
    "DPL_CELL_PADDING": 2,
    "CLOCK_PERIOD": 24,
    "FP_CORE_UTIL": 35,
    "LIB_SYNTH": "/home/mariam/OpenLane/designs/picorv32a/src/sky130_fd_sc_hd__typical.lib",
    "LIB_SLOWEST": "/home/mariam/OpenLane/designs/picorv32a/src/sky130_fd_sc_hd__slow.lib",
    "LIB_FASTEST": "/home/mariam/OpenLane/designs/picorv32a/src/sky130_fd_sc_hd__fast.lib",
    "LIB_TYPICAL": "/home/mariam/OpenLane/designs/picorv32a/src/sky130_fd_sc_hd__typical.lib",
    "EXTRA_LEFS": "/home/mariam/OpenLane/designs/picorv32a/src/sky130_vsdinv.lef",
    "pdk::sky130*": {
        "SYNTH_MAX_FANOUT": 6,
        "scl::sky130_fd_sc_ms": {
            "FP_CORE_UTIL": 30
        }
    }
}
```

maliberty, 2023-06-07:
The problem is coming from openroad but the issue file is for magic.  Please upload a reproduce that shows the routing error.

```
% cat run.sh 
#!/bin/sh
dir=$(cd -P -- "$(dirname -- "$0")" && pwd -P)
cd $dir;
export DESIGN_NAME='picorv32';
export MAGIC_GDS='./results/signoff/picorv32.magic.gds';
export MAGIC_MAGICRC='pdk/sky130A/libs.tech/magic/sky130A.magicrc';
export MAGIC_SCRIPT='openlane/scripts/magic/gds/mag_with_pointers.tcl';
export PACKAGED_SCRIPT_0='openlane/scripts/magic/wrapper.tcl';
export PACKAGED_SCRIPT_1='openlane/scripts/magic/gds/mag_with_pointers.tcl';
export signoff_tmpfiles='./tmp/signoff';
TOOL_BIN=${TOOL_BIN:-magic}
$TOOL_BIN -dnull -noconsole -rcfile $MAGIC_MAGICRC < $PACKAGED_SCRIPT_0
```

mariamrakka, 2023-06-07:
I reran the steps ensuring I use gen_pdn and it worked.

closed 2023-06-07 by mariamrakka