The-OpenROAD-Project/OpenLane #1848: Routing congestion to high in user_project_wrapper

CristopherA96, 2023-06-12:
Hi, I've posted on caravel user project repository, but no feedback about this was received.

## Description

I have the next issue about Routing congestion to high. Check the congestion heatmap in the GUI.
I am working on adding an OpenFPGA.

This is the issue_reproducible.zip.

[issue_reproducible.zip](https://github.com/efabless/caravel_user_project/files/11628281/issue_reproducible.zip)

The log file says that pins are outside die area, but we have increase the DIE_AREA and still says that.

I was trying by changing some variables in order to remove this congestion:

FP_CORE_UTIL from 50 to 40.
PL_TARGET_DENSITY from 0.55 to 0.45.
SYNTH_STRATEGY from AREA 0 to DELAY 1

As this issue https://github.com/The-OpenROAD-Project/OpenLane/issues/955 in OpenLane say.

## Setup

I am working on a machine with next specs:

Linux Mint OS
RAM 32GB

maliberty, 2023-06-12:
You have the project instance jammed up against the bottom edge leaving little room to access all those pins.  Is there a reason not to center it more?

[image]

maliberty, 2023-06-12:
"pins are outside die area" is an oddity of the OL flow but unrelated to your issue.

vijayank88, 2023-06-13:
@CristopherA96 
Are you using automatic macro placement or manual?
If manual can you share your `macro.cfg`?

CristopherA96, 2023-06-19:
Thanks @maliberty and @vijayank88. 

This issue was solved by changing the position of my macro, because I could not see the GUI, so I did not check if it was in the correct position.

I did a macro placement @vijayank88 .

closed 2023-06-19 by CristopherA96