The-OpenROAD-Project/OpenLane #1849: failed to generate  a PDN at floorplan stage

Pa1mantri, 2023-06-14:
### Description

In the RTL to GDS|| flow of RISV core with an customized sram module included as a macro, floor plan stage results in an error while generating PDN because of the configuration variable FD_PDN_MACRO_HOOKS.

### Expected Behavior

execution of floor-plan stage

### Environment report

```shell
Kernel: Linux v5.19.0-43-generic
Distribution: ubuntu 22.04
Python: v3.10.6 (OK)
Container Engine: docker v20.10.21 (OK)
OpenLane Git Version: b04797cb6c8c82403ffb14dd13da14a7c2379979
pip: INSTALLED
python-venv: INSTALLED
---
PDK Version Verification Status: FAILED
/home/pa1mantri/OpenLane/pdks/sky130A not found.
Traceback (most recent call last):
  File "/home/pa1mantri/OpenLane/dependencies/verify_versions.py", line 76, in verify_versions
    raise Exception(f"{pdk_dir} not found.")
Exception: /home/pa1mantri/OpenLane/pdks/sky130A not found.

Failed to verify sky130A.
---
Git Log (Last 3 Commits)

b04797cb 2023-06-13T14:29:11+03:00 Fix logging message of timing derate (#1839) - Kareem Farid -  (HEAD -> master, origin/master, origin/HEAD)
85e9dc67 2023-06-07T20:23:36+03:00 [BOT] Update openroad_app (#1829) - Openlane Bot -  (tag: 2023.06.08)
5ba7fa07 2023-06-06T15:30:19+03:00 Fix doctest (#1828) - Kareem Farid -  (tag: 2023.06.07)
---
Git Remotes

origin	https://github.com/The-OpenROAD-Project/OpenLane (fetch)
origin	https://github.com/The-OpenROAD-Project/OpenLane (push)
```

### Reproduction material

[issue.tar.gz](https://github.com/The-OpenROAD-Project/OpenLane/files/11744954/issue.tar.gz)

### Relevant log output

```shell
[ERROR]: during executing openroad script /openlane/scripts/openroad/pdn.tcl
[ERROR]: Log: designs/vsdmemsoc/runs/RUN_2023.06.13_06.08.48/logs/floorplan/7-pdn.log
[ERROR]: Last 10 lines:
OpenROAD ceae0ad175e8b0183c3a64af79ba18139dcda63a 
This program is licensed under the BSD-3 license. See the LICENSE file for details.
Components of this program may be licensed under more restrictive licenses which must be honored.
[INFO]: Reading ODB at '/openlane/designs/vsdmemsoc/runs/RUN_2023.06.13_06.08.48/tmp/floorplan/6-tapcell.odb'…
define_corners Typical
read_liberty -corner Typical /home/pa1mantri/.volare/sky130A/libs.ref/sky130_fd_sc_hd/lib/sky130_fd_sc_hd__tt_025C_1v80.lib
read_liberty -corner Typical /openlane/designs/vsdmemsoc/src/macros/sky130_sram_1kbyte_1rw1r_32x256_8_TT_1p8V_25C.lib
No regex match found for cntlr defined in FP_PDN_MACRO_HOOKS
child process exited abnormally

[ERROR]: Creating issue reproducible...[ERROR]: during executing openroad script /openlane/scripts/openroad/pdn.tcl
[ERROR]: Log: designs/vsdmemsoc/runs/RUN_2023.06.13_06.08.48/logs/floorplan/7-pdn.log
[ERROR]: Last 10 lines:
OpenROAD ceae0ad175e8b0183c3a64af79ba18139dcda63a 
This program is licensed under the BSD-3 license. See the LICENSE file for details.
Components of this program may be licensed under more restrictive licenses which must be honored.
[INFO]: Reading ODB at '/openlane/designs/vsdmemsoc/runs/RUN_2023.06.13_06.08.48/tmp/floorplan/6-tapcell.odb'…
define_corners Typical
read_liberty -corner Typical /home/pa1mantri/.volare/sky130A/libs.ref/sky130_fd_sc_hd/lib/sky130_fd_sc_hd__tt_025C_1v80.lib
read_liberty -corner Typical /openlane/designs/vsdmemsoc/src/macros/sky130_sram_1kbyte_1rw1r_32x256_8_TT_1p8V_25C.lib
No regex match found for cntlr defined in FP_PDN_MACRO_HOOKS
child process exited abnormally

[ERROR]: Creating issue reproducible...
```

vijayank88, 2023-06-14:
@donn @kareefardi 
Can you please check why with standalone test case run with `./run.sh` getting stopped with
```
OpenROAD 7134b2d295137b28b4b9f3f02acbdcb18391a6cf 
This program is licensed under the BSD-3 license. See the LICENSE file for details.
Components of this program may be licensed under more restrictive licenses which must be honored.
Error: pdn.tcl, 15 can't read "::env(SCRIPTS_DIR)": no such variable
```

vijayank88, 2023-06-14:
@Pa1mantri 
Is `issue_reproducible` generated by flow or you manually created?

Pa1mantri, 2023-06-14:
generated by the flow

kareefardi, 2023-06-14:
> @donn @kareefardi Can you please check why with standalone test case run with `./run.sh` getting stopped with
>
> ```
> OpenROAD 7134b2d295137b28b4b9f3f02acbdcb18391a6cf
> This program is licensed under [...]

There was a bug that should have been fixed. Let me check if it is still there.

@Pa1mantri can you upload your design folder meanwhile ?

Pa1mantri, 2023-06-14:
[src.tar.gz](https://github.com/The-OpenROAD-Project/OpenLane/files/11745235/src.tar.gz)

kareefardi, 2023-06-14:
> [src.tar.gz](https://github.com/The-OpenROAD-Project/OpenLane/files/11745235/src.tar.gz)

This is missing openlane config file.

Pa1mantri, 2023-06-14:
{

    "DESIGN_NAME": "vsdmemsoc",

    "VERILOG_INCLUDE_DIRS": "dir::src/include",

    "VERILOG_FILES": "dir::src/module/*.v",

    "EXTRA_LEFS":     "dir::src/macros/sky130_sram_1kbyte_1rw1r_32x256_8.lef",

    "EXTRA_GDS_FILES": "dir::src/macros/sky130_sram_1kbyte_1rw1r_32x256_8.gds",

    "EXTRA_LIBS":      "dir::src/macros/sky130_sram_1kbyte_1rw1r_32x256_8_TT_1p8V_25C.lib",

        

    "CLOCK_PORT": "CLK",

    "CLOCK_NET": "CLK",

    "CLOCK_PERIOD": 10,

        

    "VDD_NETS": "vccd1",

    "GND_NETS": "vssd1",

     

    "FP_SIZING": "absolute",

    "PL_TARGET_DENSITY": 0.45,

    "RT_MAX_LAYER": "met4",

    

    "FP_PDN_MACRO_HOOKS": "mem vccd1 vssd1 vccd1 vssd1, cntlr vccd1 vssd1 vccd1 vssd1",

    "DIE_AREA": "0 0 1100 700",

    "MACRO_PLACEMENT_CFG": "dir::/src/macros/macro_placement.cfg",

    "FP_PIN_ORDER_CFG": "dir::/src/macros/pin_order.cfg",

    "BASE_SDC_FILE": "dir::src/module/vsdmemsoc_synth.sdc",

    "SYNTH_SIZING": 1,

    "SYNTH_FLAT_TOP": 1,

    "SYNTH_STRATEGY": "AREA 0",

    

    "MAGIC_DRC_USE_GDS": false,

    "QUIT_ON_MAGIC_DRC": false,

    "RUN_KLAYOUT_XOR": false,

    "RUN_MAGIC_DRC" :false,

    "DESIGN_IS_CORE": true

}

would this be ok?, because uploading just the .json file is not supported. or should I include the file in the folder and zip it send it again?

kareefardi, 2023-06-14:
This should be fine

kareefardi, 2023-06-14:
@Pa1mantri You have `cntlr vccd1 vssd1 vccd1 vssd1` in `FP_PDN_MACRO_HOOKS` however instance `cntlr` is flattened in your design. In other to fix your issue, either remove `cntlr` from `FP_PDN_MACRO_HOOKS` or properly blackbox `cntlr` via `VERILOG_FILES_BLACKBOX` and `EXTRA_*` variables. 

On a different note, I am curios about the bad reproducible file generated. How are you running OpenLane? Are you using `make mount` followed by `flow.tcl ...` ?

Pa1mantri, 2023-06-14:
I am using make mount and then ./flow.tcl. Not quite sure about the bad reproducible file generated? where is it? Regarding the cntlr I will go through it by removing it from the MACRO_HOOKS and the other option you provided and then get back to you. Thank you for your time @kareefardi.

kareefardi, 2023-06-14:
> @kareefardi ok, do you mean issue_reproducible (bad reproducible)? if it is so, it is automatically generated by the flow .

Yes. Did you export `OPENLANE_IMAGE_NAME` by any chance?

Pa1mantri, 2023-06-14:
No

kareefardi, 2023-06-14:
Ok thank you

Pa1mantri, 2023-06-14:
@kareefardi , @vijayank88  Thank you it worked.

closed 2023-06-14 by kareefardi