The-OpenROAD-Project/OpenLane #1866: [ERROR PDN-0175] Pitch is too small for, must be at least

annnnna42, 2023-06-20:
### Description

I don't really know what to do at this point - I keep getting this error even though I simply copied the /spm example and changed it a bit to compile a minimal working example (`async`)

### Expected Behavior

Expected the same flow as when running `./flow.tcl -design spm` 

### Environment report

```shell
Kernel: Linux v5.19.0-43-generic
Distribution: ubuntu 22.10
Python: v3.10.7 (OK)
Container Engine: docker v24.0.2 (OK)
OpenLane Git Version: 6a056e37c1711fceabb20cef06323255211fb131
pip: INSTALLED
python-venv: INSTALLED
---
PDK Version Verification Status: OK
---
Git Log (Last 3 Commits)

6a056e37 2023-06-20T14:24:20+02:00 Merge branch 'master' of https://github.com/The-OpenROAD-Project/OpenLane - username -  (HEAD -> master)
4261cf68 2023-06-20T14:30:11+03:00 [BOT] Update PDK (#1834) - Openlane Bot -  (origin/master, origin/HEAD)
5dcbfb7f 2023-06-20T09:50:59+02:00 OpenLane initial local commit - username -  ()
---
Git Remotes

origin  https://github.com/The-OpenROAD-Project/OpenLane.git (fetch)
origin  https://gitlab.cc-asp.fraunhofer.de/hws/students/2023-fp-low-latency-masking-username/OpenLane (push)
```

### Reproduction material

[issue_reproducible.zip](https://github.com/The-OpenROAD-Project/OpenLane/files/11800713/issue_reproducible.zip)

### Relevant log output

```shell
OpenLane 6a056e37c1711fceabb20cef06323255211fb131
All rights reserved. (c) 2020-2022 Efabless Corporation and contributors.
Available under the Apache License, version 2.0. See the LICENSE file for more details.

[INFO]: Using configuration in 'designs/async/config.json'...
[INFO]: PDK Root: /home/gawehn/OpenLane
[INFO]: Process Design Kit: sky130A
[INFO]: Standard Cell Library: sky130_fd_sc_hd
[INFO]: Optimization Standard Cell Library: sky130_fd_sc_hd
[INFO]: Run Directory: /openlane/designs/async/runs/RUN_2023.06.20_13.02.23
[INFO]: Saving runtime environment...
[INFO]: Preparing LEF files for the nom corner...
[INFO]: Preparing LEF files for the min corner...
[INFO]: Preparing LEF files for the max corner...
[INFO]: Running linter (Verilator) (log: designs/async/runs/RUN_2023.06.20_13.02.23/logs/synthesis/linter.log)...
[INFO]: 0 errors found by linter
[INFO]: 0 warnings found by linter
[STEP 1]
[INFO]: Running Synthesis (log: designs/async/runs/RUN_2023.06.20_13.02.23/logs/synthesis/1-synthesis.log)...
[STEP 2]
[INFO]: Running Single-Corner Static Timing Analysis (log: designs/async/runs/RUN_2023.06.20_13.02.23/logs/synthesis/2-sta.log)...
[STEP 3]
[INFO]: Running Initial Floorplanning (log: designs/async/runs/RUN_2023.06.20_13.02.23/logs/floorplan/3-initial_fp.log)...
[WARNING]: Current core area is too small for the power grid settings chosen. The power grid will be scaled down.
[INFO]: Floorplanned with width 6.9 and height 5.44.
[STEP 4]
[INFO]: Running IO Placement...
[STEP 5]
[INFO]: Running Tap/Decap Insertion (log: designs/async/runs/RUN_2023.06.20_13.02.23/logs/floorplan/5-tap.log)...
[INFO]: Power planning with power {VPWR} and ground {VGND}...
[STEP 6]
[INFO]: Generating PDN (log: designs/async/runs/RUN_2023.06.20_13.02.23/logs/floorplan/6-pdn.log)...
[ERROR]: during executing openroad script /openlane/scripts/openroad/pdn.tcl
[ERROR]: Log: designs/async/runs/RUN_2023.06.20_13.02.23/logs/floorplan/6-pdn.log
[ERROR]: Last 10 lines:
OpenROAD 8d56294a54f42e0844ddf36cf7847f371bdb792a 
This program is licensed under the BSD-3 license. See the LICENSE file for details.
Components of this program may be licensed under more restrictive licenses which must be honored.
[INFO]: Reading ODB at '/openlane/designs/async/runs/RUN_2023.06.20_13.02.23/tmp/floorplan/5-tapcell.odb'…
define_corners Typical
read_liberty -corner Typical /home/gawehn/OpenLane/sky130A/libs.ref/sky130_fd_sc_hd/lib/sky130_fd_sc_hd__tt_025C_1v80.lib
[ERROR PDN-0175] Pitch 1.7250 is too small for, must be atleast 6.6000
Error: pdn_cfg.tcl, 92 PDN-0175
child process exited abnormally

[ERROR]: Creating issue reproducible...
[INFO]: Saving runtime environment...
OpenLane TCL Issue Packager

EFABLESS CORPORATION AND ALL AUTHORS OF THE OPENLANE PROJECT SHALL NOT BE HELD
LIABLE FOR ANY LEAKS THAT MAY OCCUR TO ANY PROPRIETARY DATA AS A RESULT OF USING
THIS SCRIPT. THIS SCRIPT IS PROVIDED ON AN "AS IS" BASIS, WITHOUT WARRANTIES OR
CONDITIONS OF ANY KIND.

BY USING THIS SCRIPT, YOU ACKNOWLEDGE THAT YOU FULLY UNDERSTAND THIS DISCLAIMER
AND ALL IT ENTAILS.

Parsing config file(s)…
Setting up /openlane/designs/async/runs/RUN_2023.06.20_13.02.23/issue_reproducible…
Done.
[INFO]: Reproducible packaged: Please tarball and upload 'designs/async/runs/RUN_2023.06.20_13.02.23/issue_reproducible' if you're going to submit an issue.
[ERROR]: Step 6 (floorplan) failed with error:
-code 1 -level 0 -errorcode NONE -errorinfo {
    while executing
"throw_error"
    (procedure "run_tcl_script" line 219)
    invoked from within
"run_tcl_script -tool openroad -no_consume {*}$args"
    (procedure "run_openroad_script" line 2)
    invoked from within
"run_openroad_script $::env(SCRIPTS_DIR)/openroad/pdn.tcl  -indexed_log [index_file $::env(floorplan_logs)/pdn.log]  -save "to=$::env(floorplan_results..."
    (procedure "gen_pdn" line 15)
    invoked from within
"gen_pdn"
    (procedure "run_power_grid_generation" line 65)
    invoked from within
"run_power_grid_generation"
    (procedure "run_floorplan" line 56)
    invoked from within
"run_floorplan"} -errorline 1
[INFO]: Saving current set of views in 'designs/async/runs/RUN_2023.06.20_13.02.23/results/final'...
[INFO]: Generating final set of reports...
[INFO]: Created manufacturability report at 'designs/async/runs/RUN_2023.06.20_13.02.23/reports/manufacturability.rpt'.
[INFO]: Created metrics report at 'designs/async/runs/RUN_2023.06.20_13.02.23/reports/metrics.csv'.
[INFO]: Saving runtime environment...
[ERROR]: Flow failed.
[INFO]: The failure may have been because of the following warnings:
[WARNING]: Current core area is too small for the power grid settings chosen. The power grid will be scaled down.
```

vijayank88, 2023-06-20:
share your `config.json`?
Try with absolute die area. Like 100x100

annnnna42, 2023-06-20:
`"FP_SIZING": "absolute"` did the trick! Thanks a lot for the hint :) Total openlane-newbie over here

vijayank88, 2023-06-20:
close?