The-OpenROAD-Project/OpenLane #1870: Error IR Drop in caravel_user_project with an OpenFPGA

CristopherA96, 2023-06-22:
Hi,

## Description
I have the next issue when running user_project_wrapper process.

I have on each instance of the fpga_top the declaration of vccd1 and vssd1, I consider that connections will be done with the user_project_wrapper, but it is still showing me those errors.

How can I treat those errors?
```
[STEP 23]
[INFO]: Creating IR Drop Report (log: ../my_path/caravel_user_project/openlane/user_project_wrapper/runs/23_06_22_14_31/logs/signoff/23-irdrop.log)...
[ERROR]: during executing openroad script /openlane/scripts/openroad/irdrop.tcl
[ERROR]: Log: ../my_path/caravel_user_project/openlane/user_project_wrapper/runs/23_06_22_14_31/logs/signoff/23-irdrop.log
[ERROR]: Last 10 lines:
[WARNING PSM-0038] Unconnected PDN node on net vccd1 at location (1582.520um, 2417.880um), layer: 5.
[WARNING PSM-0038] Unconnected PDN node on net vccd1 at location (1582.520um, 2417.880um), layer: 6.
[WARNING PSM-0038] Unconnected PDN node on net vccd1 at location (1582.520um, 2342.880um), layer: 5.
[WARNING PSM-0038] Unconnected PDN node on net vccd1 at location (1582.520um, 2342.880um), layer: 6.
[WARNING PSM-0038] Unconnected PDN node on net vccd1 at location (1582.520um, 2267.880um), layer: 5.
[WARNING PSM-0038] Unconnected PDN node on net vccd1 at location (1582.520um, 2267.880um), layer: 6.
[WARNING PSM-0038] message limit reached, this message will no longer print
[ERROR PSM-0078] IR drop setup failed.  Analysis can't proceed.
Error: irdrop.tcl, 37 PSM-0078
child process exited abnormally

[ERROR]: Creating issue reproducible...
```

Issue reproducible
This is the generated issue file.

[issue_reproducible.zip](https://github.com/The-OpenROAD-Project/OpenLane/files/11840417/issue_reproducible.zip)

## Setup
Linux Mint
RAM 32GB

Working with the latest version of caravel_user_project.

maliberty, 2023-06-22:
That message usually indicates some problem in the pdn generation.  Could you show a screen shot around the referenced area?  @gadfort for thoughts

vijayank88, 2023-06-23:
@CristopherA96 
In discussion #1862 itself i requested `pdn stage` test case. Steps to create test case: https://openlane.readthedocs.io/en/latest/for_developers/using_or_issue.html#manually

kareefardi, 2023-06-25:
> @CristopherA96 In discussion #1862 itself i requested `pdn stage` test case. Steps to create test case: https://openlane.readthedocs.io/en/latest/for_developers/using_or_issue.html#manually

Or use `CREATE_REPRODUCIBLE_FROM_SCRIPT` too

CristopherA96, 2023-06-27:
How can I add that variable if I am working into caravel user project repository, and as I know it is not a flow variable that can be set into config.json, and doing manually requires to be inside openlane container.

Thanks

kareefardi, 2023-06-28:
> How can I add that variable if I am working into caravel user project repository, and as I know it is not a flow variable that can be set into config.json, and doing manually requires to be inside openlane container.
> 
> Thanks

If you are using caravel user project Makefile, you should make a temporary edit here: https://github.com/efabless/caravel_user_project/blob/3a71fab0255a5eea4311f453e6ea95e7f16a992e/openlane/Makefile#L25. Replace `flow.tcl` with `CREATE_REPRODUCIBLE_FROM_SCRIPT=openroad/pdn.tcl flow.tcl`

CristopherA96, 2023-06-28:
This is the new issue, by following the instructons that you told me.

[issue_reproducible.zip](https://github.com/The-OpenROAD-Project/OpenLane/files/11895224/issue_reproducible.zip)

maliberty, 2023-06-28:
@gadfort this looks to be a pdn issue if you have time to take a look.

gadfort, 2023-06-28:
@maliberty I'm not sure what PDNGen could do in this case, there is a missing:
```
add_pdn_connect -grid stdcell_grid -layers "met4 met5"
```

vijayank88, 2023-06-30:
@gadfort 
Can you please enable `export FP_PDN_CHECK_NODES='1';` in `run.sh` and run the flow?
Log
```
[WARNING PSM-0038] Unconnected PDN node on net vccd1 at location (1862.520um, 1095.880um), layer: 5.
[WARNING PSM-0038] Unconnected PDN node on net vccd1 at location (1862.520um, 1095.880um), layer: 6.
[WARNING PSM-0038] Unconnected PDN node on net vccd1 at location (1862.520um, 1067.880um), layer: 5.
[WARNING PSM-0038] Unconnected PDN node on net vccd1 at location (1862.520um, 1067.880um), layer: 6.
[WARNING PSM-0038] message limit reached, this message will no longer print
[ERROR PSM-0069] Check connectivity failed.
Error: pdn.tcl, 39 PSM-0069
```
Where to define `add_pdn_connect` again?

gadfort, 2023-06-30:
It should be added: https://github.com/The-OpenROAD-Project/OpenLane/blob/3bc9d02d0b34ad032921553e512fbe4bebf1d833/scripts/openroad/common/pdn_cfg.tcl#L126

Without it there are no vias added to the macro

vijayank88, 2023-07-03:
@CristopherA96 
Please share your carvel repo link with design files.
Are you using latest OpenLane commit?

PaulinaRicoG, 2023-07-04:
Hi @vijayank88, I am Cristopher's partner.

Here is the link of the repository.

[OL_OFPGA_repo](https://github.com/PaulinaRicoG/ofpga_test)

maliberty, 2023-07-05:
@gadfort this sounds like a pdn issue

kareefardi, 2024-07-17:
`FP_PDN_ENABLE_RAILS` were disabled causing problems.

closed 2024-07-17 by kareefardi