The-OpenROAD-Project/OpenLane #1872: Global placement failed on small design

mattvenn, 2023-06-23:
### Description

`RePlAce divergence detected`

Tiny Tapeout Wokwi project: https://wokwi.com/projects/368241820443546625

Seems like a repeat of these issues:

* https://github.com/The-OpenROAD-Project/OpenLane/issues/1763
* https://github.com/The-OpenROAD-Project/OpenLane/issues/1559
* https://github.com/The-OpenROAD-Project/OpenLane/issues/1183

### Expected Behavior

OpenLane shouldn't fail to complete on small designs

### Environment report

```shell
Kernel: Linux v5.15.0-75-generic
Distribution: ubuntu 20.04
Python: v3.8.10 (OK)
Container Engine: docker v20.10.21 (OK)
OpenLane Git Version: 4cd0986b3ae550cdf7a6d0fba4e0657012f635d8
pip: INSTALLED
python-venv: INSTALLED
---
PDK Version Verification Status: OK
---
Git Log (Last 3 Commits)

4cd0986 2023-02-13T14:32:59+02:00 Bugfixes (#1658) - Kareem Farid -  (grafted, HEAD, tag: 2023.02.14)
---
Git Remotes

origin  https://github.com/The-OpenROAD-Project/OpenLane.git (fetch)
origin  https://github.com/The-OpenROAD-Project/OpenLane.git (push)
```

### Reproduction material

[issue_reproducible.tar.gz](https://github.com/The-OpenROAD-Project/OpenLane/files/11848386/issue_reproducible.tar.gz)

### Relevant log output

```shell
[ERROR]: Last 10 lines:
[NesterovSolve] Iter: 230 overflow: 0.403024 HPWL: 1971297
[NesterovSolve] Iter: 240 overflow: 0.400985 HPWL: 1969262
[NesterovSolve] Iter: 250 overflow: 0.403169 HPWL: 1971452
[NesterovSolve] Iter: 260 overflow: 0.387428 HPWL: 1964099
[NesterovSolve] Iter: 270 overflow: 0.403145 HPWL: 1971453
[NesterovSolve] Iter: 280 overflow: 0.380688 HPWL: 1949681
[NesterovSolve] Iter: 290 overflow: 0.373598 HPWL: 1949994
[ERROR GPL-0307] RePlAce divergence detected. Re-run with a smaller max_phi_cof value.
Error: gpl.tcl, 69 GPL-0307
child process exited abnormally
```

vijayank88, 2023-06-23:
Cell padding set to 0. If I use `set ::env(GPL_CELL_PADDING) {4};` it is passing. @mattvenn

vijayank88, 2023-06-23:
@maliberty fyi

mattvenn, 2023-06-23:
can you link to the GPL_CELL_PADDING documentation, I can't find it.

mattvenn, 2023-06-23:
Does that setting mean we'll get less utilisation?
I would rather not waste space just because it fails on small designs.

vijayank88, 2023-06-23:
> 

https://github.com/search?q=repo%3AThe-OpenROAD-Project%2FOpenLane%20GPL_CELL_PADDING&type=code

vijayank88, 2023-06-23:
> Does that setting mean we'll get less utilisation? I would rather not waste space just because it fails on small designs.

Whenever there is less number of pins or < 100 std cells this issue occurs randomly. Waiting for @maliberty reviews.

vijayank88, 2023-06-23:
> > 
> 
> https://github.com/search?q=repo%3AThe-OpenROAD-Project%2FOpenLane%20GPL_CELL_PADDING&type=code

Check this link, padding variable involved placement density calculation as well.

vijayank88, 2023-06-23:
[image]

mattvenn, 2023-06-23:
I just checked a larger design and I can see no difference in the final GDS. I guess if there is plenty of space then I wouldn't expect the design to be bigger?

mattvenn, 2023-06-23:
CELL_PADDING 4 fails with this: https://wokwi.com/projects/346662028519146068

I also tried setting it to 6 and 8 and it fails with all those.

maliberty, 2023-06-23:
I ran the attached issue with the head of OR (0ea657420682722c1867b4c4ac6817de05235bde) and got no error.

maliberty, 2023-06-23:
> > do you know what it does? like what does 4 actually mean?
> 
> 4 micron spacing around each standard cell.

Actually the cell padding is measuring in SITEs not microns.

kareefardi, 2023-06-25:
I will update OpenROAD as soon as I can. Currently, one of the dependencies website is down.

maliberty, 2023-06-25:
If you mean the lemon library then we have already fixed that by putting a copy in GH

kareefardi, 2023-06-25:
@maliberty https://github.com/The-OpenROAD-Project/lemon-graph is this what are you referring to ?

maliberty, 2023-06-25:
> @maliberty https://github.com/The-OpenROAD-Project/lemon-graph is this what you are referring to ?

Yes

kareefardi, 2023-06-26:
OpenROAD updated an testcase is passing now.

closed 2023-06-26 by kareefardi

mattvenn, 2023-06-27:
@kareefardi could you let me know what tag openlane should I try to check this resolves my issue?