The-OpenROAD-Project/OpenLane #1875: check for multiple drivers of a net

thesourcerer8, 2023-06-27:
### Description

We had the problem that we accidently had multiple drivers for a net, which would have resulted in a short-circuit on the ASIC. There were no warnings about it in the OpenLane flow, so we requested warnings from yosys, and got a result now: https://github.com/YosysHQ/yosys/issues/3793 

### Proposal

Please ensure to use the "check" command of yosys to detect such issues.

kareefardi, 2023-06-27:
`check` is called here https://github.com/The-OpenROAD-Project/OpenLane/blob/3bc9d02d0b34ad032921553e512fbe4bebf1d833/scripts/yosys/synth.tcl#L273. It seems that it is an issue that was recently fixed by yosys so we will look into updating yosys in OpenLane.

kareefardi, 2023-07-04:
@thesourcerer8 do you have a specific test case where you faced this issue?

kareefardi, 2023-07-04:
@thesourcerer8 Found it. Thanks

closed 2023-07-06 by kareefardi