The-OpenROAD-Project/OpenLane #1876: Global placement fails for single Rocket core with Sky130 PDK when running in OpenROAD flow scripts

hoanhe, 2023-06-29:
### Description

I set up a Single core Rocket, and run under the OpenROAD flow scripts with the platform of Nangate45.
But here is the error that happened:
OpenROAD  v2.0-8163-g9a5b2cd
Tool crash after I run global place:
[image]

### Expected Behavior

Multi-thread to run the OpenROAD process maybe?
Is the any way to fix global place fail? 

### Environment report

```shell
And this is my config file : 
`export DESIGN_NAME = ChipTop
export PLATFORM    = sky130hd

export VERILOG_FILES = ./designs/src/$(DESIGN_NICKNAME)/*.v
export SDC_FILE      = ./designs/$(PLATFORM)/$(DESIGN_NICKNAME)/ChipTop.sdc

export ADDER_MAP_FILE :=

export CORE_UTILIZATION = 30
export PLACE_DENISTY_LB_ADDON = 0.1
#export PLACE_DENSITY     = 0.51

export chameleon_DIR = ./designs/$(PLATFORM)/$(DESIGN_NICKNAME)

export ADDITIONAL_GDS  = $(chameleon_DIR)/gds/sramgen_sram_4096x8m8w8_replica_v1.gds \
                        $(chameleon_DIR)/gds/sramgen_sram_1024x32m8w8_replica_v1.gds \
                        $(chameleon_DIR)/gds/sramgen_sram_64x32m4w32_replica_v1.gds \
                        $(chameleon_DIR)/gds/sramgen_sram_4096x32m8w8_replica_v1.gds \
                        $(chameleon_DIR)/gds/sramgen_sram_512x32m4w8_replica_v1.gds \

export ADDITIONAL_LIBS = $(chameleon_DIR)/lib/sramgen_sram_4096x8m8w8_replica_v1_tt_025C_1v80.lib \
                        $(chameleon_DIR)/lib/sramgen_sram_1024x32m8w8_replica_v1_tt_025C_1v80.lib \
                        $(chameleon_DIR)/lib/sramgen_sram_64x32m4w32_replica_v1_tt_025C_1v80.lib \
                        $(chameleon_DIR)/lib/sramgen_sram_4096x32m8w8_replica_v1_tt_025C_1v80.lib \
                        $(chameleon_DIR)/lib/sramgen_sram_512x32m4w8_replica_v1_tt_025C_1v80.lib

export ADDITIONAL_LEFS  = $(chameleon_DIR)/lef/sramgen_sram_4096x8m8w8_replica_v1.lef \
			$(chameleon_DIR)/lef/sramgen_sram_1024x32m8w8_replica_v1.lef \
			$(chameleon_DIR)/lef/sramgen_sram_64x32m4w32_replica_v1.lef \
			$(chameleon_DIR)/lef/sramgen_sram_4096x32m8w8_replica_v1.lef \
			$(chameleon_DIR)/lef/sramgen_sram_512x32m4w8_replica_v1.lef \
                       $(chameleon_DIR)/lef/sky130_ef_io_core.lef \
                        $(chameleon_DIR)/lef/sky130_ef_io.lef \
                        $(chameleon_DIR)/lef/sky130_ef_sc_hd.lef \
                        $(chameleon_DIR)/lef/sky130_fd_io_core.lef \
                        $(chameleon_DIR)/lef/sky130_fd_io.lef \
                        $(chameleon_DIR)/lef/sky130_fd_pr.lef \
                        $(chameleon_DIR)/lef/sky130_fd_sc_hd.lef \
                        $(chameleon_DIR)/lef/sky130_fd_sc_hvl.lef 

export IO_CONSTRAINTS = ./designs/$(PLATFORM)/$(DESIGN_NICKNAME)/io.tcl
#export MACRO_PLACEMENT = $(chameleon_DIR)/chameleon.macro_placment.cfg
#export DIE_AREA    = 0.0 0.0 4000 2500
#export CORE_AREA   = 50 50 3950 2450

export MACRO_PLACE_CHANNEL  = 80 80
export MACRO_PLACE_HALO = 2 2

export FP_PDN_RAIL_WIDTH  = 0.48
export FP_PDN_RAIL_OFFSET = 0
export TNS_END_PERCENT   ?= 100

export MIN_ROUTING_LAYER = met1
export MAX_ROUTING_LAYER = met5

`
```

### Reproduction material

I using `make global_place_issue` and this is result: 

[https://drive.google.com/file/d/1YC7QjPH276VysOtDtmQuk12XeXiHPZB6/view?usp=sharing](url)

### Relevant log output

```shell
[NesterovSolve] Iter: 290 overflow: 0.8076 HPWL: 33919756904
[NesterovSolve] Iter: 300 overflow: 0.793513 HPWL: 34401390662
[NesterovSolve] Iter: 310 overflow: 0.773297 HPWL: 34351636433
[NesterovSolve] Iter: 320 overflow: 0.735001 HPWL: 36782446644
[NesterovSolve] Iter: 330 overflow: 0.703998 HPWL: 39799521938
[NesterovSolve] Iter: 340 overflow: 0.697883 HPWL: 41680711077
[NesterovSolve] Iter: 350 overflow: 0.686221 HPWL: 42825491214
[NesterovSolve] Iter: 360 overflow: 0.667444 HPWL: 39630777138
[NesterovSolve] Iter: 370 overflow: 0.623585 HPWL: 39527638141
[NesterovSolve] Snapshot saved at iter = 377
[NesterovSolve] Iter: 380 overflow: 0.597429 HPWL: 39916300010
[NesterovSolve] Iter: 390 overflow: 0.563805 HPWL: 40762514598
Elapsed time: 3:39:04[h:]min:sec. CPU time: user 10998.28 sys 610.90 (88%). Peak memory: 258640644KB.
make: *** [results/sky130hd/ChipTop/base/3_3_place_gp.odb] Error 9
```

vijayank88, 2023-06-29:
@hoanhe It is OpenLane flow. It is not relevant to discuss here.
Create your issue here: https://github.com/The-OpenROAD-Project/OpenROAD-flow-scripts/issues
Also your google drive attachment not accessible.

hoanhe, 2023-06-30:
Ok thank you!

closed 2023-06-30 by hoanhe