The-OpenROAD-Project/OpenLane #1877: example colab notebook fails. there's no env 'FP_PDN_LOWER_LAYER'

puppuccino, 2023-07-01:
### Description

Example Colab notebook fails at step 6. The log says 
> Error: pdn_cfg.tcl, 92 can't read "::env(FP_PDN_LOWER_LAYER)": no such variable

It seems this problem is related to [PR #1764 ](https://github.com/The-OpenROAD-Project/OpenLane/pull/1764)

### Expected Behavior

I just want it to run.

### Environment report

```shell
issue survey command didn't work in the notebook.
```

### Reproduction material

N.A.

### Relevant log output

```shell
env: PDK=sky130A
OpenLane 2023.04.12_0_g23e367a9-conda
All rights reserved. (c) 2020-2022 Efabless Corporation and contributors.
Available under the Apache License, version 2.0. See the LICENSE file for more details.

[INFO]: Using configuration in 'config.json'...
[INFO]: PDK Root: /content/conda-env/share/pdk
[INFO]: Process Design Kit: sky130A
[INFO]: Standard Cell Library: sky130_fd_sc_hd
[INFO]: Optimization Standard Cell Library: sky130_fd_sc_hd
[WARNING]: DIODE_INSERTION_STRATEGY is now deprecated; use GRT_REPAIR_ANTENNAS, DIODE_ON_PORTS and RUN_HEURISTIC_DIODE_INSERTION instead.
[INFO]: DIODE_INSERTION_STRATEGY set to 3. Setting GRT_REPAIR_ANTENNAS to 1
[INFO]: Run Directory: /content/runs/RUN_2023.07.01_15.04.25
[INFO]: Preparing LEF files for the nom corner...
[INFO]: Preparing LEF files for the min corner...
[INFO]: Preparing LEF files for the max corner...
[STEP 1]
[INFO]: Running Synthesis (log: runs/RUN_2023.07.01_15.04.25/logs/synthesis/1-synthesis.log)...
[STEP 2]
[INFO]: Running Single-Corner Static Timing Analysis (log: runs/RUN_2023.07.01_15.04.25/logs/synthesis/2-sta.log)...
[STEP 3]
[INFO]: Running Initial Floorplanning (log: runs/RUN_2023.07.01_15.04.25/logs/floorplan/3-initial_fp.log)...
[INFO]: Floorplanned with width 23.46 and height 35.36.
[STEP 4]
[INFO]: Running IO Placement...
[STEP 5]
[INFO]: Running Tap/Decap Insertion (log: runs/RUN_2023.07.01_15.04.25/logs/floorplan/5-tap.log)...
[INFO]: Power planning with power {VPWR} and ground {VGND}...
[STEP 6]
[INFO]: Generating PDN (log: runs/RUN_2023.07.01_15.04.25/logs/floorplan/6-pdn.log)...
[ERROR]: during executing openroad script /content/conda-env/share/openlane/scripts/openroad/pdn.tcl
[ERROR]: Log: runs/RUN_2023.07.01_15.04.25/logs/floorplan/6-pdn.log
[ERROR]: Last 10 lines:
OpenROAD 850238c2fb11cf83a9d33c6d065a121aae8d837b 
This program is licensed under the BSD-3 license. See the LICENSE file for details.
Components of this program may be licensed under more restrictive licenses which must be honored.
[INFO]: Reading ODB at '/content/runs/RUN_2023.07.01_15.04.25/tmp/floorplan/5-tapcell.odb'...
define_corners Typical
read_liberty -corner Typical /content/conda-env/share/pdk/sky130A/libs.ref/sky130_fd_sc_hd/lib/sky130_fd_sc_hd__tt_025C_1v80.lib
Error: pdn_cfg.tcl, 92 can't read "::env(FP_PDN_LOWER_LAYER)": no such variable
child process exited abnormally

[ERROR]: Creating issue reproducible...
[INFO]: Saving runtime environment...
OpenLane TCL Issue Packager

EFABLESS CORPORATION AND ALL AUTHORS OF THE OPENLANE PROJECT SHALL NOT BE HELD
LIABLE FOR ANY LEAKS THAT MAY OCCUR TO ANY PROPRIETARY DATA AS A RESULT OF USING
THIS SCRIPT. THIS SCRIPT IS PROVIDED ON AN "AS IS" BASIS, WITHOUT WARRANTIES OR
CONDITIONS OF ANY KIND.

BY USING THIS SCRIPT, YOU ACKNOWLEDGE THAT YOU FULLY UNDERSTAND THIS DISCLAIMER
AND ALL IT ENTAILS.

Parsing config file(s)…
Setting up /content/runs/RUN_2023.07.01_15.04.25/issue_reproducible…
Done.
[INFO]: Reproducible packaged: Please tarball and upload 'runs/RUN_2023.07.01_15.04.25/issue_reproducible' if you're going to submit an issue.
[ERROR]: Step(6:floorplan) failed with error:
-code 1 -level 0 -errorstack {INNER {invokeStk1 throw_error} CALL {run_tcl_script -tool openroad -no_consume /content/conda-env/share/openlane/scripts/openroad/pdn.tcl -indexed_log /content/runs/RUN_2023.07.01_15.04.25/logs/floorplan/6-pdn.log -save to=/content/runs/RUN_2023.07.01_15.04.25/results/floorplan,noindex,def,odb} CALL {run_openroad_script /content/conda-env/share/openlane/scripts/openroad/pdn.tcl -indexed_log /content/runs/RUN_2023.07.01_15.04.25/logs/floorplan/6-pdn.log -save to=/content/runs/RUN_2023.07.01_15.04.25/results/floorplan,noindex,def,odb} CALL gen_pdn CALL run_power_grid_generation CALL run_floorplan CALL {run_non_interactive_mode -design .}} -errorcode NONE -errorinfo {
    while executing
"throw_error"
    (procedure "run_tcl_script" line 219)
    invoked from within
"run_tcl_script -tool openroad -no_consume {*}$args"
    (procedure "run_openroad_script" line 2)
    invoked from within
"run_openroad_script $::env(SCRIPTS_DIR)/openroad/pdn.tcl  -indexed_log [index_file $::env(floorplan_logs)/pdn.log]  -save "to=$::env(floorplan_results..."
    (procedure "gen_pdn" line 15)
    invoked from within
"gen_pdn"
    (procedure "run_power_grid_generation" line 65)
    invoked from within
"run_power_grid_generation"
    (procedure "run_floorplan" line 56)
    invoked from within
"run_floorplan"} -errorline 1
[INFO]: Saving current set of views in 'runs/RUN_2023.07.01_15.04.25/results/final'...
[INFO]: Generating final set of reports...
[INFO]: Created manufacturability report at 'runs/RUN_2023.07.01_15.04.25/reports/manufacturability.rpt'.
[INFO]: Created metrics report at 'runs/RUN_2023.07.01_15.04.25/reports/metrics.csv'.
[INFO]: Saving runtime environment...
[ERROR]: Flow failed.
```

kareefardi, 2023-07-04:
The version in your log file `OpenLane 2023.04.12_0_g23e367a9-conda` is old. How did you get this version? When I tried the collab, the only thing that I had to replace was `CLOCK_TREE_SYNTH` in `Write Configuration` step.

puppuccino, 2023-07-15:
I clicked the link to the Colab notebook in the Quick-start Guide on the top page of this repository. When I tried it again today, it ran on a newer (the newest?) version. It's confusing, but the version problem is now resolved.
However, even after replacing `CLOCK_TREE_SYNTH` with `RUN_CTS`, it still doesn't work properly. Here is the log message.

```
env: PDK=sky130A
OpenLane 2023.06.26_0_g3bc9d02d-conda
All rights reserved. (c) 2020-2022 Efabless Corporation and contributors.
Available under the Apache License, version 2.0. See the LICENSE file for more details.

[INFO]: Using configuration in 'config.json'...
[INFO]: PDK Root: /content/conda-env/share/pdk
[INFO]: Process Design Kit: sky130A
[INFO]: Standard Cell Library: sky130_fd_sc_hd
[INFO]: Optimization Standard Cell Library: sky130_fd_sc_hd
[WARNING]: DIODE_INSERTION_STRATEGY is now deprecated; use GRT_REPAIR_ANTENNAS, DIODE_ON_PORTS and RUN_HEURISTIC_DIODE_INSERTION instead.
[INFO]: DIODE_INSERTION_STRATEGY set to 3. Setting GRT_REPAIR_ANTENNAS to 1
[INFO]: Run Directory: /content/runs/RUN_2023.07.15_13.39.42
[INFO]: Saving runtime environment...
[INFO]: Preparing LEF files for the nom corner...
[INFO]: Preparing LEF files for the min corner...
[INFO]: Preparing LEF files for the max corner...
[INFO]: Running linter (Verilator) (log: runs/RUN_2023.07.15_13.39.42/logs/synthesis/linter.log)...
[INFO]: 0 errors found by linter
[INFO]: 0 warnings found by linter
[STEP 1]
[INFO]: Running Synthesis (log: runs/RUN_2023.07.15_13.39.42/logs/synthesis/1-synthesis.log)...
[ERROR]: during executing yosys script /content/conda-env/share/openlane/scripts/yosys/synth.tcl
[ERROR]: Log: runs/RUN_2023.07.15_13.39.42/logs/synthesis/1-synthesis.log
[ERROR]: Last 10 lines:
ERROR: TCL interpreter returned an error: can't read "::env(SYNTH_CAP_LOAD)": no such variable
child process exited abnormally

[ERROR]: Creating issue reproducible...
[INFO]: Saving runtime environment...
OpenLane TCL Issue Packager

EFABLESS CORPORATION AND ALL AUTHORS OF THE OPENLANE PROJECT SHALL NOT BE HELD
LIABLE FOR ANY LEAKS THAT MAY OCCUR TO ANY PROPRIETARY DATA AS A RESULT OF USING
THIS SCRIPT. THIS SCRIPT IS PROVIDED ON AN "AS IS" BASIS, WITHOUT WARRANTIES OR
CONDITIONS OF ANY KIND.

BY USING THIS SCRIPT, YOU ACKNOWLEDGE THAT YOU FULLY UNDERSTAND THIS DISCLAIMER
AND ALL IT ENTAILS.

Parsing config file(s)…
Setting up /content/runs/RUN_2023.07.15_13.39.42/issue_reproducible…
[WRN] /content/runs/RUN_2023.07.15_13.39.42/tmp/synthesis/synthesis.sdc was not found, might be a product. Skipping
Done.
[INFO]: Reproducible packaged: Please tarball and upload 'runs/RUN_2023.07.15_13.39.42/issue_reproducible' if you're going to submit an issue.
[ERROR]: Step 1 (synthesis) failed with error:
-code 1 -level 0 -errorstack {INNER {invokeStk1 throw_error} CALL {run_tcl_script -tool yosys -no_consume /content/conda-env/share/openlane/scripts/yosys/synth.tcl -indexed_log /content/runs/RUN_2023.07.15_13.39.42/logs/synthesis/1-synthesis.log} CALL {run_yosys_script /content/conda-env/share/openlane/scripts/yosys/synth.tcl -indexed_log /content/runs/RUN_2023.07.15_13.39.42/logs/synthesis/1-synthesis.log} CALL {run_yosys -indexed_log /content/runs/RUN_2023.07.15_13.39.42/logs/synthesis/1-synthesis.log} CALL run_synthesis CALL {run_non_interactive_mode -design .}} -errorcode NONE -errorinfo {
    while executing
"throw_error"
    (procedure "run_tcl_script" line 219)
    invoked from within
"run_tcl_script -tool yosys -no_consume {*}$args"
    (procedure "run_yosys_script" line 2)
    invoked from within
"run_yosys_script $::env(SYNTH_SCRIPT) -indexed_log $arg_values(-indexed_log)"
    (procedure "run_yosys" line 44)
    invoked from within
"run_yosys -indexed_log $log"
    (procedure "run_synthesis" line 13)
    invoked from within
"run_synthesis"} -errorline 1
[INFO]: Saving current set of views in 'runs/RUN_2023.07.15_13.39.42/results/final'...
[INFO]: Generating final set of reports...
[INFO]: Created manufacturability report at 'runs/RUN_2023.07.15_13.39.42/reports/manufacturability.rpt'.
[INFO]: Created metrics report at 'runs/RUN_2023.07.15_13.39.42/reports/metrics.csv'.
[INFO]: Saving runtime environment...
[ERROR]: Flow failed.
```

proppy, 2023-07-24:
I believe this is now fixed with:
```
openlane            2023.06.26_0_g3bc9d02d  20230606_125334       litex-hub      12MB
```
from https://anaconda.org/LiteX-Hub/openlane/files (see https://github.com/hdl/conda-eda/pull/345 https://github.com/hdl/conda-eda/pull/348)

donn, 2025-09-15:
We're no longer recommending the Colab notebook for OpenLane and we recommend using [LibreLane](https://github.com/librelane/librelane) instead, which is a Python-native platform.