The-OpenROAD-Project/OpenLane #1884: [WARNING ODB-0176] ERROR: undefined layer (Metal4) referenced

akhilesh911, 2023-07-07:
### Description

Hello everyone, 
I am new to OpenLane and I am trying to read LEF and DEF files using OpenLane. After using command read_lef, the LEF file is not getting read by the tool. Please suggest any solutions on how to resolve this issue.
Thank you.

### Expected Behavior

Successfully read LEF file.

### Environment report

```shell
Kernel: Linux v5.4.0-148-generic
Distribution: ubuntu 18.04
Python: v3.6.9 (OK)
Container Engine: docker v20.10.24 (OK)
OpenLane Git Version: 90d369bcce5cf1133c28291bc3c8b6473a19ff62
pip: INSTALLED
python-venv: INSTALLED
---
PDK Version Verification Status: OK
---
Git Log (Last 3 Commits)

90d369b 2022-12-13T22:59:28+02:00 Iterate over `$libs` in read_libs (#1570) - Kareem Farid -  (HEAD -> master, tag: 2022.12.14, origin/master, origin/HEAD)
0d13487 2022-12-13T19:05:19+02:00 Add `report_parasitic_annotation` to sta (#1569) - Kareem Farid -  ()
d20c7b4 2022-12-13T12:32:13+02:00 Expose magic polygon subcells (#1566) - Kareem Farid -  ()
---
Git Remotes

origin	https://github.com/The-OpenROAD-Project/OpenLane.git (fetch)
origin	https://github.com/The-OpenROAD-Project/OpenLane.git (push)
```

### Reproduction material

No reproduction material produced.

### Relevant log output

```shell
[WARNING ODB-0220] WARNING (LEFPARS-2008): NOWIREEXTENSIONATPIN statement is obsolete in version 5.6 or later.
The NOWIREEXTENSIONATPIN statement will be ignored. See file designs/picosoc/macros/lef/timer.lef at line 2.

[WARNING ODB-0176] error: undefined layer (Metal4) referenced
[WARNING ODB-0176] error: undefined layer (Metal4) referenced
[WARNING ODB-0176] error: undefined layer (Metal4) referenced
[WARNING ODB-0176] error: undefined layer (Metal4) referenced
[WARNING ODB-0176] error: undefined layer (Metal4) referenced
[WARNING ODB-0176] error: undefined layer (Metal4) referenced
[WARNING ODB-0176] error: undefined layer (Metal4) referenced
[WARNING ODB-0176] error: undefined layer (Metal4) referenced
[WARNING ODB-0176] error: undefined layer (Metal2) referenced
[WARNING ODB-0176] error: undefined layer (Metal2) referenced
[WARNING ODB-0176] error: undefined layer (Metal3) referenced
[WARNING ODB-0176] error: undefined layer (Metal2) referenced
[WARNING ODB-0176] error: undefined layer (Metal3) referenced
[WARNING ODB-0176] error: undefined layer (Metal3) referenced
[WARNING ODB-0176] error: undefined layer (Metal2) referenced
[WARNING ODB-0176] error: undefined layer (Metal2) referenced
[WARNING ODB-0176] error: undefined layer (Metal2) referenced
[WARNING ODB-0176] error: undefined layer (Metal3) referenced
[WARNING ODB-0176] error: undefined layer (Metal3) referenced
[WARNING ODB-0176] error: undefined layer (Metal3) referenced
[WARNING ODB-0176] error: undefined layer (Metal3) referenced
[WARNING ODB-0176] error: undefined layer (Metal2) referenced
[WARNING ODB-0176] error: undefined layer (Metal2) referenced
[WARNING ODB-0176] error: undefined layer (Metal3) referenced
[WARNING ODB-0176] error: undefined layer (Metal3) referenced
[WARNING ODB-0176] error: undefined layer (Metal3) referenced
[WARNING ODB-0176] error: undefined layer (Metal2) referenced
[WARNING ODB-0176] error: undefined layer (Metal3) referenced
[WARNING ODB-0176] error: undefined layer (Metal3) referenced
[WARNING ODB-0176] error: undefined layer (Metal3) referenced
[WARNING ODB-0176] error: undefined layer (Metal2) referenced
[WARNING ODB-0176] error: undefined layer (Metal3) referenced
[WARNING ODB-0176] error: undefined layer (Metal3) referenced
[WARNING ODB-0176] error: undefined layer (Metal3) referenced
[WARNING ODB-0176] error: undefined layer (Metal2) referenced
[WARNING ODB-0176] error: undefined layer (Metal3) referenced
[WARNING ODB-0176] error: undefined layer (Metal2) referenced
[WARNING ODB-0176] error: undefined layer (Metal3) referenced
[WARNING ODB-0176] error: undefined layer (Metal3) referenced
[WARNING ODB-0176] error: undefined layer (Metal2) referenced
[WARNING ODB-0176] error: undefined layer (Metal2) referenced
[WARNING ODB-0176] error: undefined layer (Metal3) referenced
[WARNING ODB-0176] error: undefined layer (Metal2) referenced
[WARNING ODB-0176] error: undefined layer (Metal2) referenced
[WARNING ODB-0176] error: undefined layer (Metal3) referenced
[WARNING ODB-0176] error: undefined layer (Metal3) referenced
[WARNING ODB-0176] error: undefined layer (Metal2) referenced
[WARNING ODB-0176] error: undefined layer (Metal2) referenced
[WARNING ODB-0176] error: undefined layer (Metal2) referenced
[WARNING ODB-0176] error: undefined layer (Metal3) referenced
[WARNING ODB-0176] error: undefined layer (Metal3) referenced
[WARNING ODB-0176] error: undefined layer (Metal3) referenced
[WARNING ODB-0176] error: undefined layer (Metal2) referenced
[WARNING ODB-0176] error: undefined layer (Metal2) referenced
[WARNING ODB-0176] error: undefined layer (Metal3) referenced
[WARNING ODB-0176] error: undefined layer (Metal3) referenced
[WARNING ODB-0176] error: undefined layer (Metal3) referenced
[WARNING ODB-0176] error: undefined layer (Metal3) referenced
[WARNING ODB-0176] error: undefined layer (Metal3) referenced
[WARNING ODB-0176] error: undefined layer (Metal2) referenced
[WARNING ODB-0176] error: undefined layer (Metal3) referenced
[WARNING ODB-0176] error: undefined layer (Metal2) referenced
[WARNING ODB-0176] error: undefined layer (Metal2) referenced
[WARNING ODB-0176] error: undefined layer (Metal2) referenced
[WARNING ODB-0176] error: undefined layer (Metal2) referenced
[WARNING ODB-0176] error: undefined layer (Metal3) referenced
[WARNING ODB-0176] error: undefined layer (Metal3) referenced
[WARNING ODB-0176] error: undefined layer (Metal3) referenced
[WARNING ODB-0176] error: undefined layer (Metal3) referenced
[WARNING ODB-0176] error: undefined layer (Metal3) referenced
[WARNING ODB-0176] error: undefined layer (Metal2) referenced
[WARNING ODB-0176] error: undefined layer (Metal3) referenced
[WARNING ODB-0176] error: undefined layer (Metal3) referenced
[WARNING ODB-0176] error: undefined layer (Metal2) referenced
[WARNING ODB-0176] error: undefined layer (Metal3) referenced
[WARNING ODB-0176] error: undefined layer (Metal2) referenced
[WARNING ODB-0176] error: undefined layer (Metal3) referenced
[WARNING ODB-0176] error: undefined layer (Metal2) referenced
[WARNING ODB-0176] error: undefined layer (Metal2) referenced
[WARNING ODB-0176] error: undefined layer (Metal2) referenced
[WARNING ODB-0176] error: undefined layer (Metal3) referenced
[WARNING ODB-0176] error: undefined layer (Metal2) referenced
[WARNING ODB-0176] error: undefined layer (Metal3) referenced
[WARNING ODB-0176] error: undefined layer (Metal3) referenced
[WARNING ODB-0176] error: undefined layer (Metal3) referenced
[WARNING ODB-0176] error: undefined layer (Metal2) referenced
[WARNING ODB-0176] error: undefined layer (Metal2) referenced
[WARNING ODB-0176] error: undefined layer (Metal2) referenced
[WARNING ODB-0176] error: undefined layer (Metal2) referenced
[WARNING ODB-0176] error: undefined layer (Metal2) referenced
[WARNING ODB-0176] error: undefined layer (Metal3) referenced
[WARNING ODB-0176] error: undefined layer (Metal3) referenced
[WARNING ODB-0176] error: undefined layer (Metal3) referenced
[WARNING ODB-0176] error: undefined layer (Metal2) referenced
[WARNING ODB-0176] error: undefined layer (Metal2) referenced
[WARNING ODB-0176] error: undefined layer (Metal2) referenced
[WARNING ODB-0176] error: undefined layer (Metal3) referenced
[WARNING ODB-0176] error: undefined layer (Metal2) referenced
[WARNING ODB-0176] error: undefined layer (Metal2) referenced
[WARNING ODB-0176] error: undefined layer (Metal2) referenced
[WARNING ODB-0176] error: undefined layer (Metal3) referenced
[WARNING ODB-0176] error: undefined layer (Metal2) referenced
[WARNING ODB-0176] error: undefined layer (Metal2) referenced
[WARNING ODB-0176] error: undefined layer (Metal3) referenced
[WARNING ODB-0176] error: undefined layer (Metal3) referenced
[WARNING ODB-0176] error: undefined layer (Metal3) referenced
[WARNING ODB-0176] error: undefined layer (Metal1) referenced
```

vijayank88, 2023-07-07:
@akhilesh911 
Which PDK are you using? gf180?

akhilesh911, 2023-07-07:
@vijayank88 Yes gf180.

vijayank88, 2023-07-07:
Have you read technology lib/lib before reading your macro lef?

akhilesh911, 2023-07-07:
@vijayank88 Yes I have read liberty file using read_lib and it shows as follows
openroad> read_liberty pdks/gf180mcuC/libs.ref/gf180mcu_fd_sc_mcu7t5v0/liberty/gf180mcu_fd_sc_mcu7t5v0__tt_025C_5v00.lib 
1

vijayank88, 2023-07-07:
Technology lef file have you read or not?

akhilesh911, 2023-07-07:
@vijayank88 Technology lef is the file on this path, right? 
[pdks/gf180mcuC/libs.ref/gf180mcu_fd_sc_mcu7t5v0/lef/gf180mcu_fd_sc_mcu7t5v0.lef ]

vijayank88, 2023-07-07:
What is `SITE` name used in your macro.lef?

akhilesh911, 2023-07-07:
In my macro.lef, there is no SITE name.

vijayank88, 2023-07-07:
upload your lef file and steps you tried in script file reproduce the same.

vijayank88, 2023-07-07:
@akhilesh911 
You're using `d20c7b4 2022-12-13` 6 months old commit. Update to latest OpenLane and try again.

akhilesh911, 2023-07-07:
This is the lef file of macro.
[timer.zip](https://github.com/The-OpenROAD-Project/OpenLane/files/11979955/timer.zip)

I have followed this steps to read lef:
1) read_liberty pdks/gf180mcuC/libs.ref/gf180mcu_fd_sc_mcu7t5v0/liberty/gf180mcu_fd_sc_mcu7t5v0__tt_025C_5v00.lib
2) read_lef pdks/gf180mcuC/libs.ref/gf180mcu_fd_sc_mcu7t5v0/lef/gf180mcu_fd_sc_mcu7t5v0.lef
3) read_lef designs/macros/lef/timer.lef

akhilesh911, 2023-07-07:
@vijayank88 Thank you for your time. I updated openlane and the macro lef is read correctly.
But when i tried to read pdks/gf180mcuC/libs.ref/gf180mcu_fd_sc_mcu7t5v0/lef/gf180mcu_fd_sc_mcu7t5v0.lef this file, theere is unknown reference of site GF018hv5v_mcu_sc7.
Can you please tell me why is this happening?

kareefardi, 2023-07-08:
> @vijayank88 Thank you for your time. I updated openlane and the macro lef is read correctly. But when i tried to read pdks/gf180mcuC/libs.ref/gf180mcu_fd_sc_mcu7t5v0/lef/gf180mcu_fd_sc_mcu7t5v0.lef [...]

You need to read a techlef first `libs.ref/gf180mcu_fd_sc_mcu9t5v0/techlef/gf180mcu_fd_sc_mcu9t5v0__nom.tlef`

akhilesh911, 2023-07-10:
@kareefardi Thank you, I understood. Now the tool is reflecting syntax errors while read_verilog for any verilog file I try to read even when the flow has completed successfully. Is there any solution or is it a bug?
Please  let me know.

kareefardi, 2023-07-10:
@akhilesh911 can you share the error ?

akhilesh911, 2023-07-10:
openroad> read_verilog designs/picosoc/src/picosoc.v 
[ERROR STA-0164] designs/picosoc/src/picosoc.v line 80, syntax error, unexpected ID, expecting '='
STA-0164
while evaluating {read_verilog designs/picosoc/src/picosoc.v }

And the related line in the verilog code is:
parameter integer MEM_WORDS = 256;

kareefardi, 2023-07-10:
@akhilesh911 you can only read a gatelevel netlist (not the src verilog) through OpenROAD app. Can you open a new issue describing what you are trying to do in order to get proper help?

akhilesh911, 2023-07-11:
Thank you everyone for your help.

closed 2023-07-11 by akhilesh911