The-OpenROAD-Project/OpenLane #1895: [WARNING ORD-1012] Liberty cell has no LEF master.

akhilesh911, 2023-07-11:
### Description

Hello everyone,
I am trying to read verilog file (netlist) using openroad app. I have followed following steps:
1) read_liberty pdks/gf180mcuC/libs.ref/gf180mcu_fd_sc_mcu7t5v0/liberty/gf180mcu_fd_sc_mcu7t5v0__tt_025C_5v00.lib
2) read_lef pdks/gf180mcuC/libs.ref/gf180mcu_fd_sc_mcu7t5v0/techlef/gf180mcu_fd_sc_mcu7t5v0.tlef
3) read_verilog /path/to/results/synthesis/picosoc.v
4) link_design <top level module>
After the last step, I am getting error as:
Liberty cell has no LEF master 
instance LEF master not found.
I request you to suggest the solutions if any.
Thank you
[log_output.zip](https://github.com/The-OpenROAD-Project/OpenLane/files/12013974/log_output.zip)

### Expected Behavior

No expected errors.

### Environment report

```shell
Kernel: Linux v5.4.0-148-generic
Distribution: ubuntu 18.04
Python: v3.6.9 (OK)
Container Engine: docker v20.10.24 (OK)
OpenLane Git Version: 09cfff0dbb82d67405cddb65ec2089834e97198d
pip: INSTALLED
python-venv: INSTALLED
---
PDK Version Verification Status: FAILED
/home/signoff/openlane1/OpenLane/pdks/sky130A not found.
Traceback (most recent call last):
  File "/home/signoff/openlane1/OpenLane/dependencies/verify_versions.py", line 76, in verify_versions
    raise Exception(f"{pdk_dir} not found.")
Exception: /home/signoff/openlane1/OpenLane/pdks/sky130A not found.

Failed to verify sky130A.
---
Git Log (Last 3 Commits)

09cfff0d 2023-07-06T21:10:16+03:00 Fix lint errors in BM64 verilog sources (#1878) - Kareem Farid -  (HEAD -> master, tag: 2023.07.07, origin/master, origin/HEAD)
5998726e 2023-07-06T18:25:24+03:00 Always raise an exception when a design fails in the CI (#1879) - Kareem Farid -  ()
a0c0945d 2023-07-06T15:17:37+03:00 Make synth checks more aggressive (#1880) - Kareem Farid -  ()
---
Git Remotes

origin	https://github.com/The-OpenROAD-Project/OpenLane.git (fetch)
origin	https://github.com/The-OpenROAD-Project/OpenLane.git (push)
```

### Reproduction material

No reproduction material.

### Relevant log output

```shell
Log output is attached in log_output file.
```

kareefardi, 2023-07-11:
You need to read the standard cells lef after the techlef. The techlef defines technology related parameters, such as metal layers, vias, site and others while the standard cells lef contains the definitions of the cells themselves. That's why you got the error "no LEF master", because standard cells lef is not read and hence cells are not defined.

vijayank88, 2023-07-11:
@akhilesh911 
Before reading netlist, run
'read_lef gf180mcuC/libs.ref/gf180mcu_fd_sc_mcu7t5v0/lef/gf180mcu_fd_sc_mcu7t5v0.lef`

Otherwise in OpenLane flow `runs/RUN_2023..../tmp/merged.nom.lef` read that simply along with `.lib`

akhilesh911, 2023-07-13:
I understood. Thank you @kareefardi  @vijayank88

closed 2023-07-13 by akhilesh911