The-OpenROAD-Project/OpenLane #1896: [ERROR RSZ-0005] when making a blackbox from custom made circuit

jchin2, 2023-07-11:
### Description

Hi, I'm attempting to make test blackbox out of a custom circuit and ran into `[ERROR RSZ-0005] Run global_route before estimating parasitics for global routing.` I tried playing with the routing commands in config.json, however I'm lacking understanding to what the error log wants me to fix. I've turned `GRT_ESTIMATE_PARASITICS` on and off to no great effect. The -interactive console is currently not available to me as the moment of writing due to a tool bug.

### Expected Behavior

A generation of a blackbox with a "top wrapper circuit" around it. 

### Environment report

```shell
jchin2@short-simple:~/OpenLane$ python3 ./env.py issue-survey
Kernel: Linux v5.19.0-46-generic
Distribution: ubuntu 22.04
Python: v3.10.6 (OK)
Container Engine: docker v24.0.3 (OK)
OpenLane Git Version: 3bc9d02d0b34ad032921553e512fbe4bebf1d833
pip: INSTALLED
python-venv: INSTALLED
---
PDK Version Verification Status: FAILED
/home/jchin2/OpenLane/pdks/sky130A not found.
Traceback (most recent call last):
  File "/home/jchin2/OpenLane/dependencies/verify_versions.py", line 76, in verify_versions
    raise Exception(f"{pdk_dir} not found.")
Exception: /home/jchin2/OpenLane/pdks/sky130A not found.

Failed to verify sky130A.
---
Git Log (Last 3 Commits)

3bc9d02d 2023-06-26T07:48:24+03:00 Use OpenROAD mirror of `lemon-graph` (#1873) - Kareem Farid -  (HEAD -> master, tag: 2023.06.26, origin/master, origin/HEAD)
aeef4d05 2023-06-22T12:53:38+03:00 Enhance IR Drop Analysis (#1864) - Kareem Farid -  (tag: 2023.06.23)
064db850 2023-06-21T13:04:22+03:00 set MAGTYPE to mag for lef write as maglef is and abstract view and (#1859) - Kareem Farid -  (tag: 2023.06.22)
---
Git Remotes

origin	https://github.com/The-OpenROAD-Project/OpenLane.git (fetch)
origin	https://github.com/The-OpenROAD-Project/OpenLane.git (push)
```

### Reproduction material

[issue_reproducible.zip](https://github.com/The-OpenROAD-Project/OpenLane/files/12017722/issue_reproducible.zip)
[bb.zip](https://github.com/The-OpenROAD-Project/OpenLane/files/12017731/bb.zip)
[building_blocks.zip](https://github.com/The-OpenROAD-Project/OpenLane/files/12017732/building_blocks.zip)
[src.zip](https://github.com/The-OpenROAD-Project/OpenLane/files/12017738/src.zip)
I have included the config.json file within src.zip as well. 

### Relevant log output

```shell
OpenLane Container (3bc9d02):/openlane$ ./flow.tcl -design blackbox_test -tag test1 -overwrite
OpenLane 3bc9d02d0b34ad032921553e512fbe4bebf1d833
All rights reserved. (c) 2020-2022 Efabless Corporation and contributors.
Available under the Apache License, version 2.0. See the LICENSE file for more details.

[INFO]: Using configuration in 'designs/blackbox_test/config.json'...
[INFO]: PDK Root: /home/jchin2/.volare
[INFO]: Process Design Kit: sky130A
[INFO]: Standard Cell Library: sky130_fd_sc_hd
[INFO]: Optimization Standard Cell Library: sky130_fd_sc_hd
[INFO]: Run Directory: /openlane/designs/blackbox_test/runs/test1
[INFO]: Saving runtime environment...
[INFO]: Preparing LEF files for the nom corner...
[INFO]: Preparing LEF files for the min corner...
[INFO]: Preparing LEF files for the max corner...
[INFO]: Running linter (Verilator) (log: designs/blackbox_test/runs/test1/logs/synthesis/linter.log)...
[INFO]: 0 errors found by linter
[WARNING]: 8 warnings found by linter
[STEP 1]
[INFO]: Running Synthesis (log: designs/blackbox_test/runs/test1/logs/synthesis/1-synthesis.log)...
[STEP 2]
[INFO]: Running Single-Corner Static Timing Analysis (log: designs/blackbox_test/runs/test1/logs/synthesis/2-sta.log)...
[STEP 3]
[INFO]: Running Initial Floorplanning (log: designs/blackbox_test/runs/test1/logs/floorplan/3-initial_fp.log)...
[INFO]: Floorplanned with width 188.6 and height 176.8.
[STEP 4]
[INFO]: Running IO Placement...
[STEP 5]
[INFO]: Running Global Placement (log: designs/blackbox_test/runs/test1/logs/placement/5-global.log)...
[STEP 6]
[INFO]: Running Single-Corner Static Timing Analysis (log: designs/blackbox_test/runs/test1/logs/placement/6-gpl_sta.log)...
[STEP 7]
[INFO]: Running basic macro placement (log: designs/blackbox_test/runs/test1/logs/placement/7-basic_mp.log)...
[STEP 8]
[INFO]: Running Tap/Decap Insertion (log: designs/blackbox_test/runs/test1/logs/floorplan/8-tap.log)...
[INFO]: Power planning with power {VPWR} and ground {VGND}...
[STEP 9]
[INFO]: Generating PDN (log: designs/blackbox_test/runs/test1/logs/floorplan/9-pdn.log)...
[STEP 10]
[INFO]: Running Global Placement (log: designs/blackbox_test/runs/test1/logs/placement/10-global.log)...
[STEP 11]
[INFO]: Running Single-Corner Static Timing Analysis (log: designs/blackbox_test/runs/test1/logs/placement/11-gpl_sta.log)...
[STEP 12]
[INFO]: Running Placement Resizer Design Optimizations (log: designs/blackbox_test/runs/test1/logs/placement/12-resizer.log)...
[STEP 13]
[INFO]: Running Detailed Placement (log: designs/blackbox_test/runs/test1/logs/placement/13-detailed.log)...
[STEP 14]
[INFO]: Running Single-Corner Static Timing Analysis (log: designs/blackbox_test/runs/test1/logs/placement/14-dpl_sta.log)...
[STEP 15]
[INFO]: Running Clock Tree Synthesis (log: designs/blackbox_test/runs/test1/logs/cts/15-cts.log)...
[STEP 16]
[INFO]: Running Single-Corner Static Timing Analysis (log: designs/blackbox_test/runs/test1/logs/cts/16-cts_sta.log)...
[STEP 17]
[INFO]: Running Placement Resizer Timing Optimizations (log: designs/blackbox_test/runs/test1/logs/cts/17-resizer.log)...
[STEP 18]
[INFO]: Running Global Routing Resizer Design Optimizations (log: designs/blackbox_test/runs/test1/logs/routing/18-resizer_design.log)...
[ERROR]: during executing openroad script /openlane/scripts/openroad/resizer_routing_design.tcl
[ERROR]: Log: designs/blackbox_test/runs/test1/logs/routing/18-resizer_design.log
[ERROR]: Last 10 lines:

[INFO GRT-0018] Total wirelength: 0 um
[INFO GRT-0014] Routed nets: 0
[INFO]: Setting RC values...
[INFO GRT-0019] Found 0 clock nets.
[INFO GRT-0001] Minimum degree: 2147483647
[INFO GRT-0002] Maximum degree: 1
[ERROR RSZ-0005] Run global_route before estimating parasitics for global routing.
Error: resizer_routing_design.tcl, 52 RSZ-0005
child process exited abnormally

[ERROR]: Creating issue reproducible...
[INFO]: Saving runtime environment...
OpenLane TCL Issue Packager

EFABLESS CORPORATION AND ALL AUTHORS OF THE OPENLANE PROJECT SHALL NOT BE HELD
LIABLE FOR ANY LEAKS THAT MAY OCCUR TO ANY PROPRIETARY DATA AS A RESULT OF USING
THIS SCRIPT. THIS SCRIPT IS PROVIDED ON AN "AS IS" BASIS, WITHOUT WARRANTIES OR
CONDITIONS OF ANY KIND.

BY USING THIS SCRIPT, YOU ACKNOWLEDGE THAT YOU FULLY UNDERSTAND THIS DISCLAIMER
AND ALL IT ENTAILS.

Parsing config file(s)…
Setting up /openlane/designs/blackbox_test/runs/test1/issue_reproducible…
[WRN] /openlane/designs/blackbox_test/runs/test1/tmp/18-blackbox_test.sdc was not found, might be a product. Skipping
Done.
[INFO]: Reproducible packaged: Please tarball and upload 'designs/blackbox_test/runs/test1/issue_reproducible' if you're going to submit an issue.
[ERROR]: Step 18 (routing) failed with error:
-code 1 -level 0 -errorcode NONE -errorinfo {
    while executing
"throw_error"
    (procedure "run_tcl_script" line 219)
    invoked from within
"run_tcl_script -tool openroad -no_consume {*}$args"
    (procedure "run_openroad_script" line 2)
    invoked from within
"run_openroad_script $::env(SCRIPTS_DIR)/openroad/resizer_routing_design.tcl -indexed_log $log -save "dir=$::env(routing_tmpfiles),def,sdc,odb,netlist,..."
    (procedure "run_resizer_design_routing" line 9)
    invoked from within
"run_resizer_design_routing"
    (procedure "run_routing" line 8)
    invoked from within
"run_routing"
    (procedure "run_routing_step" line 7)
    invoked from within
"run_routing_step"} -errorline 1
[INFO]: Saving current set of views in 'designs/blackbox_test/runs/test1/results/final'...
[INFO]: Generating final set of reports...
[INFO]: Created manufacturability report at 'designs/blackbox_test/runs/test1/reports/manufacturability.rpt'.
[INFO]: Created metrics report at 'designs/blackbox_test/runs/test1/reports/metrics.csv'.
[INFO]: Saving runtime environment...
[ERROR]: Flow failed.
[INFO]: The failure may have been because of the following warnings:
[WARNING]: 8 warnings found by linter
```

kareefardi, 2023-07-12:
I think that's due to having no nets in the design. In the log there is `Routed nets: 0`. I think something should be done at OpenROAD's end to handle no nets when estimating parastics. Regardless, you should disable all optimization steps by disabling the following flags `PL_RESIZER_DESIGN_OPTIMIZATIONS`, `PL_RESIZER_TIMING_OPTIMIZATIONS`, `GLB_RESIZER_TIMING_OPTIMIZATIONS` and `GLB_RESIZER_DESIGN_OPTIMIZATIONS`

kareefardi, 2023-07-12:
Wait something else is wrong. Connection between the pins and the macro is established in the source file. There should be nets in the design. Looking into it ..

kareefardi, 2023-07-12:
I think this is happening because the pins in the LEF file of the custom made circuit and the verilog model of the same circuit don't have matching pins. These are the pins found in the LEF
```
  NOWIREEXTENSIONATPIN ON ;
  PIN CLK0
  PIN CLK1
  PIN CLK2
  PIN CLK3
  PIN Dis0
  PIN Dis1
  PIN Dis2
  PIN Dis3
  PIN GND
  PIN k0
  PIN k0_bar
  PIN k1
  PIN k1_bar
  PIN k2
  PIN k2_bar
  PIN k3
  PIN k3_bar
  PIN s0
  PIN s0_bar
  PIN s1
  PIN s1_bar
  PIN s2
  PIN s2_bar
  PIN s3
  PIN s3_bar
  PIN x0
  PIN x0_bar
  PIN x1
  PIN x1_bar
  PIN x2
  PIN x2_bar
  PIN x3
  PIN x3_bar
```
While these are the pins in the verilog model:
```
    input wire [BIT_SIZE-1:0] clk_in;
    input wire [BIT_SIZE-1:0] dis_in;
    input wire [BIT_SIZE-1:0] x_in;
    input wire [BIT_SIZE-1:0] x_in_bar;
    input wire [BIT_SIZE-1:0] k_in;
    input wire [BIT_SIZE-1:0] k_in_bar;
    
    output wire [BIT_SIZE-1:0] s_out;
    output wire [BIT_SIZE-1:0] s_out_bar;

```

jchin2, 2023-07-13:
I will try it out and make a report of the results.

jchin2, 2023-07-13:
I've reassigned the pins and increased the die area to get pass [Error GRT-0076] Net net1 not properly covered. Currently I'm having lvs error about VPWR and VGND pins? I have also tried disabling LVS_INSERT_POWER_PINS, but my lvs result is a mismatch in a netlist mismatch. 
[40-write_powered_def.log](https://github.com/The-OpenROAD-Project/OpenLane/files/12044519/40-write_powered_def.log)

jchin2, 2023-07-13:
I made a mistake in closing it...

kareefardi, 2023-07-30:
Looks like the macro is not connected to power/gnd. Somehow, magic and netgen are flagging this missing connection. I am still investigating.

kareefardi, 2023-07-30:
Looks like this is the issue https://github.com/RTimothyEdwards/magic/issues/260#issuecomment-1657118652. It is a design issue. The tools are functioning correctly.

jchin2, 2023-07-30:
Thank you for looking into the issue. It is good to know the tools are functioning as intended and I will explore the link you have provided. My question is do I have to make the pin names in the lef/gds to be verilog friendly format? 
Since the verilog model for the blackbox is something like this:
` input wire [BIT_SIZE-1:0] k_in;
  input wire [BIT_SIZE-1:0] k_in_bar;
  output wire [BIT_SIZE-1:0] s_out;
  output wire [BIT_SIZE-1:0] s_out_bar;`
would you recommend use of the following format:"s_out[0], s_out_bar[0], ... , s_out[31], s_out_bar[31]" for the layout pin names to make it easier for OpenLane to connect them?

kareefardi, 2023-08-03:
The format doesn't matter that much. It is just that they need to match each other. Whatever you chose be careful of escaping.

donn, 2024-09-12:
Kindly file a new issue if this persists.

closed 2024-09-12 by donn