The-OpenROAD-Project/OpenLane #1899: Debugging OpenLane Seg Fault

ShvetankPrakash, 2023-07-14:
Hi! I am trying the solution presented for debugging [here](https://github.com/The-OpenROAD-Project/OpenLane/issues/1718) in this issue! I have installed OpenROAD and lldb natively. However when I follow the steps I get the following issue:

$ pwd
~/OpenLane/designs/spm/runs/RUN_2023.07.13_10.43.23/issue_reproducible
$ lldb openroad -- ./run.tcl 
Current executable set to 'openroad' (x86_64).
(lldb) run
error: process launch failed: Child exec failed.
(lldb) 
Is `run` the correct command? Please let me know if you have any guidance for how to debug. I am getting a segfault at the moment so I tried this setup I found here in this past issue.

Thanks!

vijayank88, 2023-07-14:
@ShvetankPrakash 
Are you facing the same segfault while running OpenLane flow?

ShvetankPrakash, 2023-07-14:
@vijayank88 yes I am getting the segfault when running the OpenLane flow, so I am trying to setup debugging by using the lldb setup recommended by @donn in that previous issue I linked (with the native OpenROAD and lldb install + the issue_reproducible script)

vijayank88, 2023-07-14:
@ShvetankPrakash 
As per guidelines segfault should be reported to respective repo. So attach a `issue_reproducible` to fix the issue in upstream.

ShvetankPrakash, 2023-07-14:
Okay @vijayank88. Is there not a way for users to debug on their own? It seem like [this issue](https://github.com/The-OpenROAD-Project/OpenLane/issues/1718) was saying you could?

vijayank88, 2023-07-14:
Which stage of the flow, you're facing the segfault?
You need to edit `OpenROAD/src/<tool>/src/` , trace the log where it fails. Update the source code and check able to resolve segfault.

ShvetankPrakash, 2023-07-14:
Apologies for the confusion @vijayank88 from me! This is not an issue with any of the existing code/tools (so pulling the source from the repos is not the issue as everything is fine and working out of the box with the OpenLane flow). This is a segfault I believe I might have introduced playing around with the tool to try and modify it for my research purposes. 

Is there a way for me to debug myself, so I can see how my code changes may have broken things? I saw @donn setup so I thought there might be but when I tried that it crashed.