The-OpenROAD-Project/OpenLane #1911: Placement and routing flow is getting failed on diode insertion step after "repair_design"

AayushiRV, 2023-07-20:
### Description

I am running placement and routing for a design using OpenLane. The flow is getting completed but i am getting violations such as "max slew, max cap and max fanout". To reduce the violations I am trying to repair the design using "repair_design" command in openroad. After this I am again trying to run the placement and routing using openlane for the repaired design. The flow is getting terminated on "diode insertion" stage.   
[image]
[image]

### Expected Behavior

If design is being repaired by the tool, it should complete the placement and routing.

### Environment report

```shell
Kernel: Linux v5.15.0-76-generic
Distribution: ubuntu 20.04
Python: v3.8.10 (OK)
Container Engine: docker v24.0.2 (OK)
OpenLane Git Version: 90d369bcce5cf1133c28291bc3c8b6473a19ff62
pip: INSTALLED
python-venv: INSTALLED
---
PDK Version Verification Status: OK
---
Git Log (Last 3 Commits)

90d369b 2022-12-13T22:59:28+02:00 Iterate over `$libs` in read_libs (#1570) - Kareem Farid -  (HEAD -> master, tag: 2022.12.14, origin/master, origin/HEAD)
0d13487 2022-12-13T19:05:19+02:00 Add `report_parasitic_annotation` to sta (#1569) - Kareem Farid -  ()
d20c7b4 2022-12-13T12:32:13+02:00 Expose magic polygon subcells (#1566) - Kareem Farid -  ()
---
Git Remotes

origin	https://github.com/The-OpenROAD-Project/OpenLane.git (fetch)
origin	https://github.com/The-OpenROAD-Project/OpenLane.git (push)
```

### Reproduction material

No reproduction material

### Relevant log output

```shell
[STEP 12]
[INFO]: Running Diode Insertion (log: designs/subservient/runs/RUN_2023.07.20_05.16.01/logs/routing/12-diodes.log)...
[ERROR]: during executing: "openroad -exit -no_init -python /openlane/scripts/odbpy/diodes.py place --diode-cell gf180mcu_fd_sc_mcu7t5v0__antenna --diode-pin I --fake-diode-cell gf180mcu_fd_sc_mcu7t5v0__antenna --input-lef /openlane/designs/subservient/runs/RUN_2023.07.20_05.16.01/tmp/merged.nom.lef --output-def /openlane/designs/subservient/runs/RUN_2023.07.20_05.16.01/tmp/routing/12-diodes.def --output /openlane/designs/subservient/runs/RUN_2023.07.20_05.16.01/tmp/routing/12-diodes.odb /openlane/designs/subservient/runs/RUN_2023.07.20_05.16.01/tmp/11-subservient.odb |& tee /dev/null /openlane/designs/subservient/runs/RUN_2023.07.20_05.16.01/logs/routing/12-diodes.log"
[ERROR]: Exit code: 1
[ERROR]: Last 10 lines:
53# PyEval_EvalCodeEx in /lib64/libpython3.6m.so.1.0
54# PyEval_EvalCode in /lib64/libpython3.6m.so.1.0
55# 0x00007FC1879F453E in /lib64/libpython3.6m.so.1.0
56# PyRun_FileExFlags in /lib64/libpython3.6m.so.1.0
57# PyRun_SimpleFileExFlags in /lib64/libpython3.6m.so.1.0
58# Py_Main in /lib64/libpython3.6m.so.1.0
59# main in openroad
60# __libc_start_main in /lib64/libc.so.6
61# 0x0000000000CABE77 in openroad
child killed: segmentation violation
```

vijayank88, 2023-07-20:
@AayushiRV 
You're using last year commit `2022-12-13`.
Update your OpenLane to latest and try again.
If there an issue again, please attach a reproducible test case to debug the same.

AayushiRV, 2023-07-20:
I am trying to reinstall openlane for an updated version, the installation process is getting killed after throwing following error
 make[1]: Entering directory '/home/signoff/OpenLane'
d054702b2cce04761cc2bc598f6b95c9d8ca7c6c: Pulling from efabless/openlane
Digest: sha256:62bc52ee243242c73f9fc3e77d255c53d34bf9b937e0d586da5e5155cdbd21e4
Status: Image is up to date for efabless/openlane:d054702b2cce04761cc2bc598f6b95c9d8ca7c6c
docker.io/efabless/openlane:d054702b2cce04761cc2bc598f6b95c9d8ca7c6c
make[1]: Leaving directory '/home/signoff/OpenLane'
PYTHONPATH= ./venv/bin/python3 -m pip install --upgrade --no-cache-dir volare
Requirement already satisfied: volare in ./venv/lib/python3.8/site-packages (0.9.0)
Requirement already satisfied: click<9,>=8.0.0 in ./venv/lib/python3.8/site-packages (from volare) (8.1.6)
Requirement already satisfied: pyyaml<7,>=5 in ./venv/lib/python3.8/site-packages (from volare) (5.4.1)
Requirement already satisfied: rich<13,>=12 in ./venv/lib/python3.8/site-packages (from volare) (12.6.0)
Requirement already satisfied: requests<3,>=2.27.0 in ./venv/lib/python3.8/site-packages (from volare) (2.31.0)
Requirement already satisfied: pcpp<2,>=1.2 in ./venv/lib/python3.8/site-packages (from volare) (1.30)
Requirement already satisfied: zstandard<1,>=0.19.0 in ./venv/lib/python3.8/site-packages (from volare) (0.21.0)
Requirement already satisfied: charset-normalizer<4,>=2 in ./venv/lib/python3.8/site-packages (from requests<3,>=2.27.0->volare) (3.2.0)
Requirement already satisfied: idna<4,>=2.5 in ./venv/lib/python3.8/site-packages (from requests<3,>=2.27.0->volare) (3.4)
Requirement already satisfied: urllib3<3,>=1.21.1 in ./venv/lib/python3.8/site-packages (from requests<3,>=2.27.0->volare) (2.0.4)
Requirement already satisfied: certifi>=2017.4.17 in ./venv/lib/python3.8/site-packages (from requests<3,>=2.27.0->volare) (2023.5.7)
Requirement already satisfied: commonmark<0.10.0,>=0.9.0 in ./venv/lib/python3.8/site-packages (from rich<13,>=12->volare) (0.9.1)
Requirement already satisfied: pygments<3.0.0,>=2.6.0 in ./venv/lib/python3.8/site-packages (from rich<13,>=12->volare) (2.15.1)
Requirement already satisfied: typing-extensions<5.0,>=4.0.0 in ./venv/lib/python3.8/site-packages (from rich<13,>=12->volare) (4.7.1)
./venv/bin/volare enable --pdk sky130
[12:52:03] Found version 78b7bc32ddb4b6f14f76883c2e2dc5b5de9d1cbc in ./dependencies/tool_metadata.yml.                                                                                        common.py:221
enable() got an unexpected keyword argument 'console'
make: *** [Makefile:128: pdk] Error 255

vijayank88, 2023-07-20:
@AayushiRV 
Until it get fixed from upstream you may update https://github.com/The-OpenROAD-Project/OpenLane/blob/master/Makefile#L127 as follows and run `make` again
```
PYTHONPATH= ./venv/bin/$(PYTHON_BIN) -m pip install --upgrade --no-cache-dir volare==0.8.0
```
https://github.com/efabless/volare/issues/50

AayushiRV, 2023-07-20:
The openlane got installed with the above arrangement. Thanks

AayushiRV, 2023-07-21:
The placement and routing flow is getting completed for a design indicating proper installation. When I am running my design I am getting following error on synthesis step-

[STEP 1]
[INFO]: Running Synthesis (log: designs/subservient/runs/RUN_2023.07.21_09.53.52/logs/synthesis/1-synthesis.log)...
[ERROR]: Synthesis failed. Range select out of bounds on some signals. Search for 'out of bounds on signal' in /openlane/designs/subservient/runs/RUN_2023.07.21_09.53.52/logs/synthesis/1-synthesis.log
[ERROR]: Step 1 (synthesis) failed with error:
-code 1 -level 0 -errorcode NONE -errorinfo {
    while executing
"throw_error"
    (procedure "check_out_of_bound" line 7)
    invoked from within
"check_out_of_bound $log"
    (procedure "run_synthesis_checkers" line 3)
    invoked from within
"run_synthesis_checkers $log $pre_synth_report"
    (procedure "run_synthesis" line 19)
    invoked from within
"run_synthesis"} -errorline 1

The 1-synthesis.log file indicates following warning for one of the verilog file-

Warning: Range [5:-1] select out of bounds on signal `\i_raddr': Setting 5 LSB bits to undef.

vijayank88, 2023-07-21:
Is this your own RTL?
Try to fix the error by reading `1-synthesis.log`

AayushiRV, 2023-07-24:
RTL may not be the problem because the placement and routing flow was working properly in the previous version of openlane.

vijayank88, 2023-07-24:
`Lint` feature added recently. The version you worked previously not had that feature. So stick to flow and resolve synthesis error to proceed further.

AayushiRV, 2023-07-24:
The 1-synthesis.log file indicates following warning for one of the verilog file-

Warning: Range [5:-1] select out of bounds on signal `\i_raddr': Setting 5 LSB bits to undef.

vijayank88, 2023-07-24:
> The 1-synthesis.log file indicates following warning for one of the verilog file-
> 
> Warning: Range [5:-1] select out of bounds on signal `\i_raddr': Setting 5 LSB bits to undef.

so you should fix out of bounds for the signal `i_addr`

closed 2023-07-24 by donn

AayushiRV, 2023-07-25:
I am unable to rectify the problem, attaching the verilog file as well as log file.
[serv_rf_ram.zip](https://github.com/The-OpenROAD-Project/OpenLane/files/12157485/serv_rf_ram.zip)
[1-synthesis.zip](https://github.com/The-OpenROAD-Project/OpenLane/files/12157486/1-synthesis.zip)

vijayank88, 2023-08-02:
@AayushiRV 
As per https://github.com/The-OpenROAD-Project/OpenROAD/discussions/3732, just looking your files.

I am not verilog expert. But something to look into the following values,

```
  #(parameter width=0,
    parameter csr_regs=4,
    parameter depth=32*(32+csr_regs)/width)
```
What will be `depth` value here?

```
input wire [$clog2(depth)-1:0] i_waddr,
```
So linter error:
```
%Error: Internal Error: /openlane/designs/serv_rf_ram/src/serv_rf_ram.v:6:30: ../V3Number.cpp:905: toUInt with 4-state 32'bxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxx
                                                                            : ... In instance serv_rf_ram
    6 |     input wire [$clog2(depth)-1:0] i_waddr,
      |                              ^
                        ... See the manual at https://verilator.org/verilator_doc.html for more assistance.
```
So you should fix this and run the flow. If older version support mean, you should not depend on the same.