The-OpenROAD-Project/OpenLane #1913: documentation for starting klayout is broken

mattvenn, 2023-07-20:
### Description

running klayout inside docker according to instructions fails

https://openlane.readthedocs.io/en/latest/getting_started/installation/installation_ubuntu.html#optional-viewing-test-design-outputs

### Expected Behavior

should show design in klayout

### Environment report

```shell
Kernel: Linux v5.15.0-76-generic
Distribution: ubuntu 20.04
Python: v3.8.10 (OK)
Container Engine: docker v20.10.21 (OK)
OpenLane Git Version: 3bc9d02d0b34ad032921553e512fbe4bebf1d833
pip: INSTALLED
python-venv: INSTALLED
---
PDK Version Verification Status: OK
---
Git Log (Last 3 Commits)

3bc9d02 2023-06-26T07:48:24+03:00 Use OpenROAD mirror of `lemon-graph` (#1873) - Kareem Farid -  (grafted, HEAD, tag: 2023.06.26)
---
Git Remotes

origin  https://github.com/The-OpenROAD-Project/OpenLane.git (fetch)
origin  https://github.com/The-OpenROAD-Project/OpenLane.git (push)
```

### Reproduction material

I just used the spm example

### Relevant log output

```shell
OpenLane Container (3bc9d02):/openlane$ klayout -e -nn $PDK_ROOT/sky130A/libs.tech/klayout/tech/sky130A.lyt \
>    -l $PDK_ROOT/sky130A/libs.tech/klayout/tech/sky130A.lyp \
>    ./designs/spm/runs/openlane_test/results/final/gds/spm.gds
process 2306: The last reference on a connection was dropped without closing the connection. This is a bug in an application. See dbus_connection_unref() documentation for details.
Most likely, the application was supposed to call dbus_connection_close(), since this is a private connection.
  D-Bus not built with -rdynamic so unable to print a backtrace
```

vijayank88, 2023-07-21:
@mattvenn 
As per latest guide: https://openlane.readthedocs.io/en/latest/reference/gui.html
you have to use 
```
python3 gui.py --viewer klayout ./designs/spm/runs/openlane_test
```

mattvenn, 2023-07-21:
[image]

mattvenn, 2023-07-21:
also, can you give a command that would work for caravel?

vijayank88, 2023-07-21:
It's working fine with Centos7, getting error has above in Ubuntu20.04

mattvenn, 2023-07-21:
yes I'm on ubuntu 20

mattvenn, 2023-07-24:
also just to note that most people for the shuttle will be running inside caravel_user_project. They'll run make user_project_wrapper, and then what should they do to examine the resulting gds?

kareefardi, 2023-08-01:
> [image]

Does the workaround here fix the problem ? https://github.com/The-OpenROAD-Project/OpenLane/issues/1772

kareefardi, 2023-08-01:
> also just to note that most people for the shuttle will be running inside caravel_user_project. They'll run make user_project_wrapper, and then what should they do to examine the resulting gds?

OpenLane now mounts the home directory. In most cases, this guide (https://openlane.readthedocs.io/en/latest/reference/gui.html) should work.

wilfredkisku, 2023-10-20:
I too am unable to start klayout for the test generated gds. Anyone who knows a workaround?

OS: Ubuntu 20.04 LTS

donn, 2023-12-18:
Kindly find a new issue @wilfredkisku.

As for this issue, staled out.

closed 2023-12-18 by donn