The-OpenROAD-Project/OpenLane #1915: make error with something related to pip

Asma-Mohsin, 2023-07-21:
### Description

I have been trying to run make and this is the error I am getting. Have tried multiple things but dont know how to solve it

[image]

### Expected Behavior

make successful

### Environment report

```shell
Kernel: Linux v5.19.0-46-generic
Distribution: ubuntu 22.04
Python: v3.10.6 (OK)
Container Engine: docker v24.0.4 (OK)
OpenLane Git Version: d054702b2cce04761cc2bc598f6b95c9d8ca7c6c
pip: INSTALLED
python-venv: INSTALLED
---
PDK Version Verification Status: MISMATCH
The version of open_pdks used in building the PDK does not match the version OpenLane was tested on (installed: 3af133706e554a740cfe60f21e773d9eaa41838c, tested: 78b7bc32ddb4b6f14f76883c2e2dc5b5de9d1cbc)
This may introduce some issues. You may want to re-install the PDK by invoking `make pdk`.
---
Git Log (Last 3 Commits)

d054702b 2023-07-19T16:09:15+03:00 remove `unset_propagated_clock` (#1908) - passant5 -  (HEAD -> master, tag: 2023.07.19, origin/master, origin/HEAD)
32dc1852 2023-07-16T15:22:31+03:00 OR CI: consider that designs moved to submodule (#1900) - Vitor Bandeira -  (tag: 2023.07.17)
c43cd3cb 2023-07-13T12:39:52+03:00 Move CI designs to a separate repository  (#1867) - Kareem Farid -  (tag: 2023.07.14)
---
Git Remotes

origin	https://github.com/The-OpenROAD-Project/OpenLane (fetch)
origin	https://github.com/The-OpenROAD-Project/OpenLane (push)
```

### Reproduction material

cd openlane
make

### Relevant log output

```shell
none
```

vijayank88, 2023-07-21:
@Asma-Mohsin Are you doing 1st time installation?
Have you ran this?

On Ubuntu, you may also need to install venv: `apt-get install python3-venv`, and pip: `apt-get install python3-pip`

vijayank88, 2023-07-21:
Something problem with `wheel`

```
Getting requirements to build wheel ... error
  error: subprocess-exited-with-error
  
  × Getting requirements to build wheel did not run successfully.
  │ exit code: 1
  ╰─> [62 lines of output]
      /tmp/pip-build-env-su67glsq/overlay/lib/python3.10/site-packages/setuptools/config/setupcfg.py:293: _DeprecatedConfig: Deprecated config in `setup.cfg`
      !!
      
              ********************************************************************************
              The license_file parameter is deprecated, use license_files instead.
      
              By 2023-Oct-30, you need to update your project and remove deprecated calls
              or your builds will no longer be supported.
      
              See https://setuptools.pypa.io/en/latest/userguide/declarative_config.html for details.
              ********************************************************************************
      
      !!
        parsed = self.parsers.get(option_name, lambda x: x)(value)
      running egg_info
      writing lib3/PyYAML.egg-info/PKG-INFO
      writing dependency_links to lib3/PyYAML.egg-info/dependency_links.txt
      writing top-level names to lib3/PyYAML.egg-info/top_level.txt
      Traceback (most recent call last):
        File "/home/naveengdh/OLFS/OpenLane/venv/lib/python3.10/site-packages/pip/_vendor/pyproject_hooks/_in_process/_in_process.py", line 353, in <module>
          main()
        File "/home/naveengdh/OLFS/OpenLane/venv/lib/python3.10/site-packages/pip/_vendor/pyproject_hooks/_in_process/_in_process.py", line 335, in main
          json_out['return_val'] = hook(**hook_input['kwargs'])
        File "/home/naveengdh/OLFS/OpenLane/venv/lib/python3.10/site-packages/pip/_vendor/pyproject_hooks/_in_process/_in_process.py", line 118, in get_requires_for_build_wheel
          return hook(config_settings)
        File "/tmp/pip-build-env-su67glsq/overlay/lib/python3.10/site-packages/setuptools/build_meta.py", line 341, in get_requires_for_build_wheel
          return self._get_build_requires(config_settings, requirements=['wheel'])
        File "/tmp/pip-build-env-su67glsq/overlay/lib/python3.10/site-packages/setuptools/build_meta.py", line 323, in _get_build_requires
          self.run_setup()
        File "/tmp/pip-build-env-su67glsq/overlay/lib/python3.10/site-packages/setuptools/build_meta.py", line 338, in run_setup
          exec(code, locals())
        File "<string>", line 271, in <module>
        File "/tmp/pip-build-env-su67glsq/overlay/lib/python3.10/site-packages/setuptools/__init__.py", line 107, in setup
          return distutils.core.setup(**attrs)
        File "/tmp/pip-build-env-su67glsq/overlay/lib/python3.10/site-packages/setuptools/_distutils/core.py", line 185, in setup
          return run_commands(dist)
        File "/tmp/pip-build-env-su67glsq/overlay/lib/python3.10/site-packages/setuptools/_distutils/core.py", line 201, in run_commands
          dist.run_commands()
        File "/tmp/pip-build-env-su67glsq/overlay/lib/python3.10/site-packages/setuptools/_distutils/dist.py", line 969, in run_commands
          self.run_command(cmd)
        File "/tmp/pip-build-env-su67glsq/overlay/lib/python3.10/site-packages/setuptools/dist.py", line 1234, in run_command
          super().run_command(command)
        File "/tmp/pip-build-env-su67glsq/overlay/lib/python3.10/site-packages/setuptools/_distutils/dist.py", line 988, in run_command
          cmd_obj.run()
        File "/tmp/pip-build-env-su67glsq/overlay/lib/python3.10/site-packages/setuptools/command/egg_info.py", line 314, in run
          self.find_sources()
        File "/tmp/pip-build-env-su67glsq/overlay/lib/python3.10/site-packages/setuptools/command/egg_info.py", line 322, in find_sources
          mm.run()
        File "/tmp/pip-build-env-su67glsq/overlay/lib/python3.10/site-packages/setuptools/command/egg_info.py", line 551, in run
          self.add_defaults()
        File "/tmp/pip-build-env-su67glsq/overlay/lib/python3.10/site-packages/setuptools/command/egg_info.py", line 589, in add_defaults
          sdist.add_defaults(self)
        File "/tmp/pip-build-env-su67glsq/overlay/lib/python3.10/site-packages/setuptools/command/sdist.py", line 104, in add_defaults
          super().add_defaults()
        File "/tmp/pip-build-env-su67glsq/overlay/lib/python3.10/site-packages/setuptools/_distutils/command/sdist.py", line 251, in add_defaults
          self._add_defaults_ext()
        File "/tmp/pip-build-env-su67glsq/overlay/lib/python3.10/site-packages/setuptools/_distutils/command/sdist.py", line 336, in _add_defaults_ext
          self.filelist.extend(build_ext.get_source_files())
        File "<string>", line 201, in get_source_files
        File "/tmp/pip-build-env-su67glsq/overlay/lib/python3.10/site-packages/setuptools/_distutils/cmd.py", line 107, in __getattr__
          raise AttributeError(attr)
      AttributeError: cython_sources
      [end of output]
  
  note: This error originates from a subprocess, and is likely not a problem with pip.
error: subprocess-exited-with-error

× Getting requirements to build wheel did not run successfully.
│ exit code: 1
╰─> See above for output.

note: This error originates from a subprocess, and is likely not a problem with pip.
make: *** [Makefile:149: venv/created] Error 1
```

LDFranck, 2023-07-21:
Apparently, there is an issue with PyYAML version 5.4.0 and the latest release of Cython ([more info](https://github.com/yaml/pyyaml/issues/724)).

I've just made it work by removing `pyyaml` version from `OpenLane/dependencies/python/run_time.txt`.

I'm not sure if it is the right approach, but it works.

vijayank88, 2023-07-22:
@donn 
Can you please provide workaround to use for Ubuntu22.04, to resolve above issue?

Asma-Mohsin, 2023-07-22:
> @Asma-Mohsin Are you doing 1st time installation? Have you ran this?
> 
> On Ubuntu, you may also need to install venv: `apt-get install python3-venv`, and pip: `apt-get install python3-pip`

NO, I am not a beginner. someone was saying its the issue with python version. I should python 3.8 or 3.9 and it will work fine.

Asma-Mohsin, 2023-07-22:
> Apparently, there is an issue with PyYAML version 5.4.0 and the latest release of Cython ([more info](https://github.com/yaml/pyyaml/issues/724)).
>
> I've just made it work by removing `pyyaml` [...]

u mean like remove that line or change its version to something else?

Can you point out the exact changes we have to make

Asma-Mohsin, 2023-07-22:
> Apparently, there is an issue with PyYAML version 5.4.0 and the latest release of Cython ([more info](https://github.com/yaml/pyyaml/issues/724)).
>
> I've just made it work by removing `pyyaml` [...]

thankyou. I changed the version to 6.0.1 and its working for me.

vijayank88, 2023-07-22:
Yes updating https://github.com/The-OpenROAD-Project/OpenLane/blob/d054702b2cce04761cc2bc598f6b95c9d8ca7c6c/dependencies/python/run_time.txt#L2
```
pyyaml~=6.0.1
```
completes installation for Ubuntu22.04 successfully. Thanks @Asma-Mohsin @LDFranck

PR #1916 (2023-07-24): PIP Version Ranges
~ Update Pyyaml to 6
~ Add maximum versions for all dependencies as per semantic versioning

---

Resolves #1915
closed 2023-07-24 by donn