The-OpenROAD-Project/OpenLane #1924: How do I retain the actual names of registers ?

prathiksha97, 2023-08-01:
In our design there are D flipflops and fifo registers. After synthesing yosys , renames the flip flops
sky130_fd_sc_hd_dfrtp_2 _14258 (
.CLK(CLK),
.D(00157),
.Q(mv_output[20]),
.RESET_B(RST_N)
);

When synthesizing Verilog in openlane yosys generates a verilog rtl which consists of  6-digit codes for the registers. Thus most of the registers have unhelpful names such as _14258_.

How do I retain the actual names of the registers usded in the verilog rtl?

vijayank88, 2023-08-01:
@prathiksha97 
Add `"SYNTH_AUTONAME": 1,` part of `config.json` and try it.

prathiksha97, 2023-08-01:
I added this step in yosys tcl file

scripts/yosys/synth.tcl
@@ -335,6 +335,17 @@ proc run_strategy {output script strategy_name {postfix_with_strategy 0}} {
    tee -o "$::env(synth_report_prefix).$strategy_escaped.chk.rpt" check
    tee -o "$::env(synth_report_prefix).$strategy_escaped.stat.rpt" stat -top $::env(DESIGN_NAME) -liberty [lindex $::env(LIB_SYNTH_NO_PG) 0]

    if { [info exists ::env(SYNTH_AUTONAME)] && $::env(SYNTH_AUTONAME) } {
        # Generate public names for the various nets, resulting in very long names that include
        # the full heirarchy, which is preferable to the internal names that are simply
        # sequential numbers such as `000019`. Renamed net names can be very long, such as:
        #     manual_reset_gf180mcu_fd_sc_mcu7t5v0_dffq_1_Q_D_gf180mcu \
        #     fd_sc_mcu7t5v0_nor3_1_ZN_A1_gf180mcu_fd_sc_mcu7t5v0aoi21 \
        #     1_A2_A1_gf180mcu_fd_sc_mcu7t5v0_nand3_1_ZN_A3_gf180mcu_fd \
        #     sc_mcu7t5v0_and3_1_A3_Z_gf180mcu_fd_sc_mcu7t5v0_buf_1_I_Z
        autoname
    }

And I did add "SYNTH_AUTONAME": 1 in config.json but I got this error
[image]

kareefardi, 2023-08-01:
Only `SYNTH_AUTONAME` is needed. No need to modify `synth.tcl`. Do you get this error without `SYNTH_AUTONAME` on?

vijayank88, 2023-08-01:
Something issue with `src/mk_combo_mul.v` in line `112`

prathiksha97, 2023-08-01:
@kareefardi yess, I tried running without modifying the synth.tcl and with "SYNTH_AUTONAME" :1 in config.json 
I still got the error

Without adding "SYNTH_AUTONAME":1 in config.json the flow runs without any errors

vijayank88, 2023-08-01:
can you share OpenLane commit you're using?

prathiksha97, 2023-08-01:
@vijayank88 
This is the commit I am using
commit a0b19fa68b79eff3e29b674106b06d20aab637f1 (HEAD, tag: 2022.11.02)
Author: Mohamed Gaber <me@donn.website>
Date:   Tue Nov 1 19:09:25 2022 +0200

Yess I did add this manually in synth.tcl

if { [info exists ::env(SYNTH_AUTONAME)] && $::env(SYNTH_AUTONAME) } {
    # Generate public names for the various nets, resulting in very long names that include
    # the full heirarchy, which is preferable to the internal names that are simply
    # sequential numbers such as `000019`. Renamed net names can be very long, such as:
    #     manual_reset_gf180mcu_fd_sc_mcu7t5v0_dffq_1_Q_D_gf180mcu \
    #     fd_sc_mcu7t5v0_nor3_1_ZN_A1_gf180mcu_fd_sc_mcu7t5v0aoi21 \
    #     1_A2_A1_gf180mcu_fd_sc_mcu7t5v0_nand3_1_ZN_A3_gf180mcu_fd \
    #     sc_mcu7t5v0_and3_1_A3_Z_gf180mcu_fd_sc_mcu7t5v0_buf_1_I_Z
    autoname
}

vijayank88, 2023-08-01:
@prathiksha97 `2022.11.02` it is last year tag.
Better updated and run again.
```
git checkout master
git pull
make
```

prathiksha97, 2023-08-01:
@vijayank88 
This is the commit I am using

commit a0b19fa68b79eff3e29b674106b06d20aab637f1 (HEAD, tag: 2022.11.02)
Author: Mohamed Gaber <me@donn.website>
Date:   Tue Nov 1 19:09:25 2022 +0200

I added the below code in synth.tcl, this was done manually
if { [info exists ::env(SYNTH_AUTONAME)] && $::env(SYNTH_AUTONAME) } {
    # Generate public names for the various nets, resulting in very long names that include
    # the full heirarchy, which is preferable to the internal names that are simply
    # sequential numbers such as `000019`. Renamed net names can be very long, such as:
    #     manual_reset_gf180mcu_fd_sc_mcu7t5v0_dffq_1_Q_D_gf180mcu \
    #     fd_sc_mcu7t5v0_nor3_1_ZN_A1_gf180mcu_fd_sc_mcu7t5v0aoi21 \
    #     1_A2_A1_gf180mcu_fd_sc_mcu7t5v0_nand3_1_ZN_A3_gf180mcu_fd \
    #     sc_mcu7t5v0_and3_1_A3_Z_gf180mcu_fd_sc_mcu7t5v0_buf_1_I_Z
    autoname
}

kareefardi, 2023-08-01:
Can you remove the manual change, update OpenLane and try again?

prathiksha97, 2023-08-01:
@kareefardi 
When I removed the manual change the flow ran without any errors

But I am still getting the 6 digits instead of actual names
Also I updated Openlane

vijayank88, 2023-08-01:
> @prathiksha97 Add `"SYNTH_AUTONAME": 1,` part of `config.json` and try it.
> 
> Note: `Add a synthesis step to generate names for instances. This results in instance names that can be very long, but may be more useful than the internal names that are six digit numbers.`

@prathiksha97 now add this part of config.json and run again.

prathiksha97, 2023-08-01:
I'm sorry but previously my Openlane wasn't updated but when I updated and ran the design I am getting this error
[image]

The environment is not matching with the Openlane scripts

kareefardi, 2023-08-01:
Not sure how you updated OpenLane, but I would try a fresh clone and following the quick start guide.

prathiksha97, 2023-08-01:
@kareefardi  I ran these cmds
git checkout master
git pull
make

This was the commit after updating
commit 4c74d8b13c39ac20acfd412d90fdd689a959fafc (HEAD -> master, tag: 2023.08.01, origin/master, origin/HEAD)
Author: Kareem Farid <kareefardi@users.noreply.github.com>
Date:   Mon Jul 31 17:13:43 2023 +0000

kareefardi, 2023-08-01:
You need to unset that variable. This is causing the issue above.

prathiksha97, 2023-08-03:
@kareefardi  and @vijayank88  Thank You!!

I could get the desired results after adding the "SYNTH_AUTONAME": 1 in config json file and
changing the yosys synth.tcl file.