The-OpenROAD-Project/OpenLane #1927: Error while doing make in Openlane

aster3440, 2023-08-03:
I tried installing the newer version of Openlane but the pdks and most of the files don't get downloaded completely. I did follow the openlane documentation.

This is the version I was using prior a0b19fa68b79eff3e29b674106b06d20aab637f1

So I tried to commit the previous stable version which I was using by git checkout <version>

When I run make I am getting this error, How do I fix it?
[image]

vijayank88, 2023-08-03:
@aster3440 
Please provide the steps you used to update the OpenLane.

kareefardi, 2023-08-03:
I believe this is an old error. https://github.com/The-OpenROAD-Project/OpenLane/issues/1915. Can you post the issues that you are facing with the latest OpenLane?

donn, 2024-09-12:
Staled out.

closed 2024-09-12 by donn