The-OpenROAD-Project/OpenLane #1930: Resizer got stuck

vijayank88, 2023-08-04:
### Description

I am trying to harden this GFMPW design : https://github.com/gregdavill/gf180-mpw0-serv/tree/main/openlane/serv_0 with latest OpenLane, but resizer got struck and consume all RAM and get killed at the end.

### Expected Behavior

Complete resizer the stage.

### Environment report

```shell
Kernel: Linux v3.10.0-1160.90.1.el7.x86_64
Distribution: centos 7
Python: v3.6.8 (OK)
Container Engine: docker v24.0.2 (OK)
OpenLane Git Version: 4c74d8b13c39ac20acfd412d90fdd689a959fafc
pip: INSTALLED
python-venv: INSTALLED
---

---
Git Log (Last 3 Commits)

4c74d8b 2023-07-31T17:13:43+00:00 Rename `mca` folders (#1922) - Kareem Farid -  (grafted, HEAD, tag: 2023.08.01)
---
Git Remotes

origin	https://github.com/The-OpenROAD-Project/OpenLane (fetch)
origin	https://github.com/The-OpenROAD-Project/OpenLane (push)
```

### Reproduction material

[issue_reproducible.zip](https://github.com/The-OpenROAD-Project/OpenLane/files/12259682/issue_reproducible.zip)

### Relevant log output

```shell
OpenROAD 41a51eaf4ca2171c92ff38afb91eb37bbd3f36da 
This program is licensed under the BSD-3 license. See the LICENSE file for details.
Components of this program may be licensed under more restrictive licenses which must be honored.
[INFO]: Reading ODB at '/home/vijayan/CARAVEL_FLOW/03_08_OL_Caravel_Flow/caravel_user_project/openlane/serv_0/runs/23_08_04_07_44/tmp/placement/8-global.odb'…
define_corners Slowest Typical Fastest
read_liberty -corner Slowest /home/vijayan/CARAVEL_FLOW/03_08_OL_Caravel_Flow/caravel_user_project/dependencies/pdks/gf180mcuC/libs.ref/gf180mcu_fd_sc_mcu7t5v0/liberty/gf180mcu_fd_sc_mcu7t5v0__ss_125C_4v50.lib
read_liberty -corner Slowest /home/vijayan/CARAVEL_FLOW/03_08_OL_Caravel_Flow/caravel_user_project/openlane/serv_0/../../dependencies/pdks/gf180mcuC/libs.ref/gf180mcu_fd_ip_sram/liberty/gf180mcu_fd_ip_sram__sram256x8m8wm1__tt_025C_1v80.lib
read_liberty -corner Typical /home/vijayan/CARAVEL_FLOW/03_08_OL_Caravel_Flow/caravel_user_project/dependencies/pdks/gf180mcuC/libs.ref/gf180mcu_fd_sc_mcu7t5v0/liberty/gf180mcu_fd_sc_mcu7t5v0__tt_025C_5v00.lib
read_liberty -corner Typical /home/vijayan/CARAVEL_FLOW/03_08_OL_Caravel_Flow/caravel_user_project/openlane/serv_0/../../dependencies/pdks/gf180mcuC/libs.ref/gf180mcu_fd_ip_sram/liberty/gf180mcu_fd_ip_sram__sram256x8m8wm1__tt_025C_1v80.lib
[WARNING STA-0053] /home/vijayan/CARAVEL_FLOW/03_08_OL_Caravel_Flow/caravel_user_project/openlane/serv_0/../../dependencies/pdks/gf180mcuC/libs.ref/gf180mcu_fd_ip_sram/liberty/gf180mcu_fd_ip_sram__sram256x8m8wm1__tt_025C_1v80.lib line 29, library gf180mcu_fd_ip_sram__sram256x8m8wm1__tt_025C_1v80 already exists.
read_liberty -corner Fastest /home/vijayan/CARAVEL_FLOW/03_08_OL_Caravel_Flow/caravel_user_project/dependencies/pdks/gf180mcuC/libs.ref/gf180mcu_fd_sc_mcu7t5v0/liberty/gf180mcu_fd_sc_mcu7t5v0__ff_n40C_5v50.lib
read_liberty -corner Fastest /home/vijayan/CARAVEL_FLOW/03_08_OL_Caravel_Flow/caravel_user_project/openlane/serv_0/../../dependencies/pdks/gf180mcuC/libs.ref/gf180mcu_fd_ip_sram/liberty/gf180mcu_fd_ip_sram__sram256x8m8wm1__tt_025C_1v80.lib
[WARNING STA-0053] /home/vijayan/CARAVEL_FLOW/03_08_OL_Caravel_Flow/caravel_user_project/openlane/serv_0/../../dependencies/pdks/gf180mcuC/libs.ref/gf180mcu_fd_ip_sram/liberty/gf180mcu_fd_ip_sram__sram256x8m8wm1__tt_025C_1v80.lib line 29, library gf180mcu_fd_ip_sram__sram256x8m8wm1__tt_025C_1v80 already exists.
Using 1e-12 for capacitance...
Using 1e+00 for resistance...
Using 1e-09 for time...
Using 1e+00 for voltage...
Using 1e-03 for current...
Using 1e-06 for power...
Using 1e-06 for distance...
Reading design constraints file at '/home/vijayan/CARAVEL_FLOW/03_08_OL_Caravel_Flow/caravel_user_project/openlane/serv_0/runs/23_08_04_07_44/tmp/floorplan/3-initial_fp.sdc'…
[INFO]: Setting RC values...
[INFO RSZ-0027] Inserted 4 input buffers.
[INFO RSZ-0028] Inserted 2 output buffers.
[INFO RSZ-0058] Using max wire length 9189um.
```

maliberty, 2023-08-04:
Do you have any macros?  We have seen such cases when there is an unreasonably small max_transition

vijayank88, 2023-08-05:
Yes, sram integrated.

maliberty, 2023-08-05:
How did you generate the .lib for the sram?

vijayank88, 2023-08-05:
It is coming with part of gf180 pdk.

openroadie, 2023-08-10:
Same as:

https://github.com/The-OpenROAD-Project/OpenROAD/issues/3783

donn, 2023-08-15:
Careful with [keywords](https://docs.github.com/en/issues/tracking-your-work-with-issues/linking-a-pull-request-to-an-issue#linking-a-pull-request-to-an-issue-using-a-keyword) in PR descriptions: this is not yet fixed in OpenLane.

openroadie, 2023-08-18:
@donn @vijayank88 : What are the next steps needed for this to be fixed in OpenLane ? Do we need to update OpenLane to use the latest OR code?

maliberty, 2023-08-18:
The update is the OL team's responsibility (@donn )

closed 2023-09-05 by donn

scorbetta, 2023-11-22:
I got similar problem in my design. The "[STEP 15] Running Global Routing Resizer Design Optimizations" stage takes forever, while all previous steps take a bunch of minutes. Here's the tail of the log:

```
[INFO GRT-0096] Final congestion report:
Layer         Resource        Demand        Usage (%)    Max H / Max V / Total Overflow
---------------------------------------------------------------------------------------
Metal1               0             0            0.00%             0 /  0 /  0
Metal2         1144386        129375           11.31%             0 /  0 /  0
Metal3         1144522        151323           13.22%             0 /  0 /  0
Metal4         1119052         41853            3.74%             0 /  0 /  0
---------------------------------------------------------------------------------------
Total          3407960        322551            9.46%             0 /  0 /  0

[INFO GRT-0018] Total wirelength: 3697520 um
[INFO GRT-0014] Routed nets: 39950
[INFO]: Setting RC values...
[INFO RSZ-0058] Using max wire length 9189um.
```

I just updated OpenLane and everything. I am trying to harden my design to GF180MCUD PDK. The configuration file is:

```
{
    "DESIGN_NAME": "CORTEZ1",
    "DESIGN_IS_CORE": 0,
    "VERILOG_FILES": [
        "dir::../../verilog/rtl/all_modules.v",
        "dir::../../verilog/rtl/defines.v",
        "dir::../../verilog/rtl/CORTEZ1.v"
    ],
    "CLOCK_PERIOD": 20,
    "CLOCK_PORT": "wb_clk_i",
    "CLOCK_NET": "NETWORK_TOP_WRAPPER_i.CLK",
    "FP_SIZING": "absolute",
    "DIE_AREA": "0 0 3000 3000",
    "FP_PIN_ORDER_CFG": "dir::pin_order.cfg",
    "MAX_TRANSITION_CONSTRAINT": 10.0,
    "MAX_FANOUT_CONSTRAINT": 16,
    "PL_RESIZER_TIMING_OPTIMIZATIONS": 1,
    "PL_RESIZER_SETUP_SLACK_MARGIN": 0.5,
    "PL_RESIZER_HOLD_SLACK_MARGIN": 0.5,
    "PL_RESIZER_ALLOW_SETUP_VIOS": 1,
    "GLB_RESIZER_TIMING_OPTIMIZATIONS": 1,
    "GLB_RESIZER_SETUP_SLACK_MARGIN": 0.5,
    "GLB_RESIZER_HOLD_SLACK_MARGIN": 0.5,
    "GLB_RESIZER_ALLOW_SETUP_VIOS": 1,
    "MAGIC_DEF_LABELS": 0,
    "SYNTH_BUFFERING": 0,
    "RUN_HEURISTIC_DIODE_INSERTION": 1,
    "HEURISTIC_ANTENNA_THRESHOLD": 110,    
    "GRT_REPAIR_ANTENNAS": 1,
    "VDD_NETS": [
        "vccd1"
    ],
    "GND_NETS": [
        "vssd1"
    ],
    "IO_SYNC": 0,
    "BASE_SDC_FILE": "dir::base_cortez.sdc",
    "RUN_CVC": 1,
	"PDK": "gf180mcuD",
	"STD_CELL_LIBRARY": "gf180mcu_fd_sc_mcu7t5v0",
    "FP_CORE_UTIL": 40,
    "RT_MAX_LAYER": "Metal4",
    "PL_TARGET_DENSITY": 0.45,
    "SYNTH_STRATEGY": "DELAY 4"
}
```

Any idea?

donn, 2023-11-22:
If disabling it is out of the question, you may want file a new issue