The-OpenROAD-Project/OpenLane #1934: OpenLane quit at synthesis stage with no apparent error

heliboy2003, 2023-08-07:
### Description

I am pretty new to OpenLane, just started to try it out for only few days. I've successfully run the 'test' and one of my own design (very small) and able to generate GDS. But when I tried a larger design, OpenLane stop at Sythesis stage (Step 1) with no apparent error. 

### Expected Behavior

Be able to finish all the required steps and generate GDS

### Environment report

```shell
Kernel: Linux v5.15.0-78-generic
Distribution: ubuntu 20.04
Python: v3.8.10 (OK)
Container Engine: docker v24.0.5 (OK)
OpenLane Git Version: 4c74d8b13c39ac20acfd412d90fdd689a959fafc
NOT FOUND: Please install pip using your operating system's package manager.
---
PDK Version Verification Status: FAILED
/home/heliboy/Desktop/OpenLane/pdks/sky130A not found.
Traceback (most recent call last):
  File "/home/heliboy/Desktop/OpenLane/dependencies/verify_versions.py", line 76, in verify_versions
    raise Exception(f"{pdk_dir} not found.")
Exception: /home/heliboy/Desktop/OpenLane/pdks/sky130A not found.

Failed to verify sky130A.
---
Git Log (Last 3 Commits)

4c74d8b1 2023-07-31T17:13:43+00:00 Rename `mca` folders (#1922) - Kareem Farid -  (HEAD -> master, tag: 2023.08.01, origin/master, origin/HEAD)
01e67230 2023-07-25T13:53:21+03:00 [BOT] Update magic (#1914) - Openlane Bot -  (tag: 2023.07.26)
157a6505 2023-07-25T13:02:39+03:00 Update Documentation (#1917) - Mohamed Gaber -  ()
---
Git Remotes

origin	https://github.com/The-OpenROAD-Project/OpenLane.git (fetch)
origin	https://github.com/The-OpenROAD-Project/OpenLane.git (push)
```

### Reproduction material

[issue_reproducible.tar.gz](https://github.com/The-OpenROAD-Project/OpenLane/files/12274099/issue_reproducible.tar.gz)

### Relevant log output

```shell
No relevant log output which can indicate the reason for synthesis to quit.
```

heliboy2003, 2023-08-07:
Sorry, first time to use this, the following is the message output to console

root@Ubuntu20:/openlane# ./flow.tcl -design Proj
OpenLane 4c74d8b13c39ac20acfd412d90fdd689a959fafc
All rights reserved. (c) 2020-2022 Efabless Corporation and contributors.
Available under the Apache License, version 2.0. See the LICENSE file for more details.

[INFO]: Using configuration in 'Proj/config.json'...
[INFO]: PDK Root: /root/.volare
[INFO]: Process Design Kit: sky130A
[INFO]: Standard Cell Library: sky130_fd_sc_hd
[INFO]: Optimization Standard Cell Library: sky130_fd_sc_hd
[INFO]: Run Directory: /openlane/Proj/runs/RUN_2023.08.07_06.51.31
[INFO]: Saving runtime environment...
[INFO]: Preparing LEF files for the nom corner...
[INFO]: Preparing LEF files for the min corner...
[INFO]: Preparing LEF files for the max corner...
[INFO]: Running linter (Verilator) (log: Proj/runs/RUN_2023.08.07_06.51.31/logs/synthesis/linter.log)...
[INFO]: 0 errors found by linter
[WARNING]: 1149 warnings found by linter
[STEP 1]
[INFO]: Running Synthesis (log: Proj/runs/RUN_2023.08.07_06.51.31/logs/synthesis/1-synthesis.log)...
[ERROR]: during executing yosys script /openlane/scripts/yosys/synth.tcl
[ERROR]: Log: Proj/runs/RUN_2023.08.07_06.51.31/logs/synthesis/1-synthesis.log
[ERROR]: Last 10 lines:

[INFO]: USING STRATEGY AREA 0

64. Executing ABC pass (technology mapping using ABC).

64.1. Extracting gate netlist of module `\TxEncode' to `/tmp/yosys-abc-KdQOSV/input.blif'..
Extracted 182146 gates and 188238 wires to a netlist network with 6090 inputs and 4884 outputs.

64.1.1. Executing ABC.
child killed: kill signal

[ERROR]: Creating issue reproducible...

d-m-bailey, 2023-08-07:
Looks like an out-of-memory error. Can you watch your swap space while it's running?

kareefardi, 2023-08-07:
It might be a memory issue. Can you upload your project files so that I can test?

heliboy2003, 2023-08-07:
Hi,

You folks are correct, it is out-of-memory problem. I am using VM running Ubuntu, and the RAM seems too small (8GB) and after I increase that to 32 GB, it can execute up to Step 15, the Running Global Routing Resizer Design Optimizations then reports Routing congestion and quit. Since this is not related to this issue, so I close this issue.

Lesson learned, make sure memory is enough!!!

Thanks.

closed 2023-08-07 by heliboy2003