The-OpenROAD-Project/OpenLane #1935: TritonCTS crashes with multiple clocks

mo-hosni, 2023-08-07:
### Description

TritonCTS fails crashes with no specific error. 
Defining the clocks `dll_clk` and `dll_clk90` causes the crash. I tried removing the `.lib` files of the DLL macro that produces these clocks and it still crashes.

### Expected Behavior

CTS creating clock trees for `dll_clk` and `dll_clk90`.

### Environment report

```shell
Kernel: Linux v6.2.0-26-generic
Distribution: ubuntu 23.04
Python: v3.11.4 (OK)
Container Engine: docker v24.0.5 (OK)
OpenLane Git Version: 01e67230f35aa4cd08ef33a9f8df8a83f578b710
pip: INSTALLED
python-venv: INSTALLED
---
PDK Version Verification Status: OK
---
Git Log (Last 3 Commits)

01e67230 2023-07-25T13:53:21+03:00 [BOT] Update magic (#1914) - Openlane Bot -  (HEAD -> master, tag: 2023.07.26, origin/master, origin/HEAD)
157a6505 2023-07-25T13:02:39+03:00 Update Documentation (#1917) - Mohamed Gaber -  ()
f6e50abd 2023-07-24T15:20:03+03:00 PIP Version Ranges (#1916) - Mohamed Gaber -  (tag: 2023.07.25)
---
Git Remotes

origin  git@github.com:The-OpenROAD-Project/OpenLane.git (fetch)
origin  git@github.com:The-OpenROAD-Project/OpenLane.git (push)
```

### Reproduction material

[issue_reproducible.part1.zip](https://github.com/The-OpenROAD-Project/OpenLane/files/12280117/issue_reproducible.part1.zip)
[issue_reproducible.part2.zip](https://github.com/The-OpenROAD-Project/OpenLane/files/12280146/issue_reproducible.part2.zip)

### Relevant log output

```shell
[INFO CTS-0007] Net "gpio_in[38]" found for clock "clk".
[INFO CTS-0010]  Clock net "gpio_in[38]" has 3 sinks.
[WARNING CTS-0041] Net "_02778_" has 1 sinks. Skipping...
[INFO CTS-0007] Net "dll_clk" found for clock "dll_clk".
Signal 11 received
Stack trace:
 0# 0x0000000000D429A7 in openroad
 1# 0x00007FABFB640400 in /lib64/libc.so.6
 2# odb::dbBTerm::getConstName() in openroad
 3# cts::TritonCTS::initClock(odb::dbNet*, std::string const&, cts::TreeBuilder*) in openroad
 4# cts::TritonCTS::initOneClockTree(odb::dbNet*, std::string const&, cts::TreeBuilder*) in openroad
 5# cts::TritonCTS::populateTritonCTS() in openroad
 6# cts::TritonCTS::runTritonCts() in openroad
 7# 0x000000000123E23F in openroad
 8# 0x00007FAC00174EB2 in /lib64/libtcl8.5.so
 9# 0x00007FAC001B936C in /lib64/libtcl8.5.so
```

mo-hosni, 2023-08-07:
There are no virtual clocks, the clocks that make the issue are outputs of a macro and defined as follows:
```
create_clock -name dll_clk -period 6.666 [get_pins {dll/clockp[1]}]
create_clock -name dll_clk90 -period 6.666 [get_pins {dll/clockp[0]}]
```

precisionmoon, 2023-08-07:
It seems this zip file is corrupt and can't be uncompressed:
[issue_reproducible.part1.zip](https://github.com/The-OpenROAD-Project/OpenLane/files/12280117/issue_reproducible.part1.zip)

Also, can the testcase be provided without using proprietary software like PkZip? 

Thanks.

Cho

mo-hosni, 2023-08-08:
Apparently something went wrong while splitting the files, those should be working.

[issue_reproducible.partab.tar.gz](https://github.com/The-OpenROAD-Project/OpenLane/files/12290054/issue_reproducible.partab.tar.gz)
[issue_reproducible.partaa.tar.gz](https://github.com/The-OpenROAD-Project/OpenLane/files/12290053/issue_reproducible.partaa.tar.gz)

Thanks

donn, 2023-08-08:
> Also, can the testcase be provided without using proprietary software like PkZip?

the utility to extract zip files is present under a libre software license on every linux distribution under the name `unzip`

precisionmoon, 2023-12-11:
PR #4330 resolves this issue.

kareefardi, 2023-12-11:
@maliberty Please keep these kind of issues open in the future until OpenROAD is updated in OpenLane. This helps us track things. You can remove OpenROAD label if you would like to.

kareefardi, 2023-12-11:
@maliberty I am still facing the crash using OpenROAD commit `f758ec989211179258a31b8c19dba68b66370bec` and latest too.

maliberty, 2023-12-11:
> @maliberty I am still facing the crash using OpenROAD commit `f758ec989211179258a31b8c19dba68b66370bec` and latest too.

@precisionmoon please check if this works for you or not with the head of OR

maliberty, 2023-12-11:
> @maliberty Please keep these kind of issues open in the future until OpenROAD is updated in OpenLane. This helps us track things. You can remove OpenROAD label if you would like to.

Will do

precisionmoon, 2023-12-11:
Please re-assign this back to me.   Let me look into this.

precisionmoon, 2023-12-15:
My apologies for missing up this issue with another CTS fatal.   The enclosed testcases are corrupt and can't be uncompressed:

gunzip -t issue_reproducible.partaa.tar.gz 
gzip: issue_reproducible.partaa.tar.gz: unexpected end of file
gunzip -t issue_reproducible.partab.tar.gz 
gzip: issue_reproducible.partab.tar.gz: not in gzip format

vijayank88, 2023-12-15:
@precisionmoon 
Hope this is helpful to uncompress the parta/partb files.
```
cat issue_reproducible.parta* > issue_reproducible.tar.gz
tar zxvf issue_reproducible.tar.gz
```

precisionmoon, 2023-12-15:
The issue here is that a clock net is not connected to any output instance pin or input block terminal.  We can skip such nets for CTS.  CTS completes successfully with the testcase  with the following messages:

[INFO CTS-0122] Clock net "dll_clk" is skipped for CTS because it is not connected to any output instance pin or input block terminal.
[INFO CTS-0122] Clock net "dll_clk90" is skipped for CTS because it is not connected to any output instance pin or input block terminal.

This issue will be fixed when [PR 4390](https://github.com/The-OpenROAD-Project/OpenROAD/pull/4390) is merged.

Thanks for your patience.

kareefardi, 2023-12-17:
@precisionmoon Thank you. I can confirm that the crash has been fixed with the update.

precisionmoon, 2023-12-17:
Can we close this issue then?

kareefardi, 2023-12-18:
@precisionmoon it is linked against #2062. I will remove the assignment to you and OpenROAD label.

PR #2062 (2023-12-26): Update `OpenROAD`
~ openroad_app -> `75f2f32`

## CI
~ Always compare regression and create reproducibles
~ Always escape design name
+ Add tests for attached crashes by adding issues reproducibles in designs submodules and symlink them to interactive scripts under ./tests
+ Fix failing designs in extended test:
  * salsa20: setup violations
  * y_huff: routing congestion
  * aes_core: pin antenna violations benchmark mismatch

---
[ci ets]
Fixes #2065
Fixes #2060 
Fixes #1935
Propagates https://github.com/The-OpenROAD-Project/OpenROAD/issues/3982
https://github.com/efabless/openlane-ci-design [... cut]
closed 2023-12-26 by donn