The-OpenROAD-Project/OpenLane #1937: Routing Congestion and Openroad DRC viewer crash

heliboy2003, 2023-08-08:
### Description

I have a design which has been implemented successfully on Xilinx Artix-7 using Vivado. And I extract something from that design and tried it using OpenLane with success, I was able to get GDS and view it using klayout.

Then I make the extracted design a little bit larger, this time OpenLane stop at Step 15-Global Routing Resizer Design Optimizations, and reported routing congestion. And when I tried to open the .rpt file using OpenRoad's DRC Viewer, OpenRoad GUI crashed. 

I've been trying to find the solution by googling and looked at the issues reported here but could not get anything to solve my problem.

### Expected Behavior

1. Be able to finish routing
2. OpenRoad DRC Viewer does not crash

### Environment report

```shell
permission denied while trying to connect to the Docker daemon socket at unix:///var/run/docker.sock: Get "http://%2Fvar%2Frun%2Fdocker.sock/v1.24/info": dial unix /var/run/docker.sock: connect: permission denied
'NoneType' object is not iterable
Kernel: Linux v5.15.0-78-generic
Distribution: ubuntu 20.04
Python: v3.8.10 (OK)
Critical Alert: No Docker or Docker-compatible container engine was found.
OpenLane Git Version: 4c74d8b13c39ac20acfd412d90fdd689a959fafc
NOT FOUND: Please install pip using your operating system's package manager.
---
PDK Version Verification Status: FAILED
/home/heliboy/Desktop/OpenLane/pdks/sky130A not found.
Traceback (most recent call last):
  File "/home/heliboy/Desktop/OpenLane/dependencies/verify_versions.py", line 76, in verify_versions
    raise Exception(f"{pdk_dir} not found.")
Exception: /home/heliboy/Desktop/OpenLane/pdks/sky130A not found.

Failed to verify sky130A.
---
Git Log (Last 3 Commits)

4c74d8b1 2023-07-31T17:13:43+00:00 Rename `mca` folders (#1922) - Kareem Farid -  (HEAD -> master, tag: 2023.08.01, origin/master, origin/HEAD)
01e67230 2023-07-25T13:53:21+03:00 [BOT] Update magic (#1914) - Openlane Bot -  (tag: 2023.07.26)
157a6505 2023-07-25T13:02:39+03:00 Update Documentation (#1917) - Mohamed Gaber -  ()
---
Git Remotes

origin	https://github.com/The-OpenROAD-Project/OpenLane.git (fetch)
origin	https://github.com/The-OpenROAD-Project/OpenLane.git (push)
```

### Reproduction material

[RoutingCongestion1.tar.gz](https://github.com/The-OpenROAD-Project/OpenLane/files/12289644/RoutingCongestion1.tar.gz)
[RoutingCongestion2.tar.gz](https://github.com/The-OpenROAD-Project/OpenLane/files/12289648/RoutingCongestion2.tar.gz)

### Relevant log output

```shell
============= The following is the console output OpenLane generated at Step 15

[STEP 15]
[INFO]: Running Global Routing Resizer Design Optimizations (log: Proj/runs/RUN_2023.08.08_07.54.40/logs/routing/15-resizer_design.log)...
[ERROR]: during executing openroad script /openlane/scripts/openroad/resizer_routing_design.tcl
[ERROR]: Log: Proj/runs/RUN_2023.08.08_07.54.40/logs/routing/15-resizer_design.log
[ERROR]: Last 10 lines:
[INFO GRT-0101] Running extra iterations to remove overflow.
[INFO GRT-0103] Extra Run for hard benchmark.
[INFO GRT-0197] Via related to pin nodes: 400687
[INFO GRT-0198] Via related Steiner nodes: 27286
[INFO GRT-0199] Via filling finished.
[INFO GRT-0111] Final number of vias: 744723
[INFO GRT-0112] Final usage 3D: 3845943
[ERROR GRT-0119] Routing congestion too high. Check the congestion heatmap in the GUI and load /openlane/Proj/runs/RUN_2023.08.08_07.54.40/tmp/routing/resizer-routing-design-congestion.rpt in the DRC viewer.
Error: resizer_routing_design.tcl, 46 GRT-0119
child process exited abnormally

[ERROR]: Creating issue reproducible...
[INFO]: Saving runtime environment...

============== The following is the console output when trying to open  resizer-routing-design-congestion.rpt file using OpenRoad's DRC Viewer =================

heliboy@Ubuntu20:~$ openroad -gui
OpenROAD a5c01a88bd7ee0d0cb35332aa48886ff3fbccee6 
This program is licensed under the BSD-3 license. See the LICENSE file for details.
Components of this program may be licensed under more restrictive licenses which must be honored.
Signal 11 received
Stack trace:
 0# 0x00005629F5B130E5 in openroad
 1# 0x00007FBA1CEEA090 in /lib/x86_64-linux-gnu/libc.so.6
 2# odb::dbBlock::getDataBase() in openroad
 3# gui::DRCWidget::loadTRReport(QString const&) in openroad
 4# gui::DRCWidget::loadReport(QString const&) in openroad
 5# gui::DRCWidget::selectReport() in openroad
 6# 0x00005629F6008378 in openroad
 7# QMetaObject::activate(QObject*, int, int, void**) in /lib/x86_64-linux-gnu/libQt5Core.so.5
 8# 0x00007FBA1E766BD1 in /lib/x86_64-linux-gnu/libQt5Widgets.so.5
 9# 0x00007FBA1E767E5F in /lib/x86_64-linux-gnu/libQt5Widgets.so.5
10# QAbstractButton::mouseReleaseEvent(QMouseEvent*) in /lib/x86_64-linux-gnu/libQt5Widgets.so.5
11# QWidget::event(QEvent*) in /lib/x86_64-linux-gnu/libQt5Widgets.so.5
12# QApplicationPrivate::notify_helper(QObject*, QEvent*) in /lib/x86_64-linux-gnu/libQt5Widgets.so.5
13# QApplication::notify(QObject*, QEvent*) in /lib/x86_64-linux-gnu/libQt5Widgets.so.5
14# QCoreApplication::notifyInternal2(QObject*, QEvent*) in /lib/x86_64-linux-gnu/libQt5Core.so.5
15# QApplicationPrivate::sendMouseEvent(QWidget*, QMouseEvent*, QWidget*, QWidget*, QWidget**, QPointer<QWidget>&, bool, bool) in /lib/x86_64-linux-gnu/libQt5Widgets.so.5
16# 0x00007FBA1E6D035D in /lib/x86_64-linux-gnu/libQt5Widgets.so.5
17# 0x00007FBA1E6D31EC in /lib/x86_64-linux-gnu/libQt5Widgets.so.5
18# QApplicationPrivate::notify_helper(QObject*, QEvent*) in /lib/x86_64-linux-gnu/libQt5Widgets.so.5
19# QApplication::notify(QObject*, QEvent*) in /lib/x86_64-linux-gnu/libQt5Widgets.so.5
20# QCoreApplication::notifyInternal2(QObject*, QEvent*) in /lib/x86_64-linux-gnu/libQt5Core.so.5
21# QGuiApplicationPrivate::processMouseEvent(QWindowSystemInterfacePrivate::MouseEvent*) in /lib/x86_64-linux-gnu/libQt5Gui.so.5
22# QGuiApplicationPrivate::processWindowSystemEvent(QWindowSystemInterfacePrivate::WindowSystemEvent*) in /lib/x86_64-linux-gnu/libQt5Gui.so.5
23# QWindowSystemInterface::sendWindowSystemEvents(QFlags<QEventLoop::ProcessEventsFlag>) in /lib/x86_64-linux-gnu/libQt5Gui.so.5
24# 0x00007FBA193C932E in /lib/x86_64-linux-gnu/libQt5XcbQpa.so.5
25# g_main_context_dispatch in /lib/x86_64-linux-gnu/libglib-2.0.so.0
26# 0x00007FBA1C638400 in /lib/x86_64-linux-gnu/libglib-2.0.so.0
27# g_main_context_iteration in /lib/x86_64-linux-gnu/libglib-2.0.so.0
28# QEventDispatcherGlib::processEvents(QFlags<QEventLoop::ProcessEventsFlag>) in /lib/x86_64-linux-gnu/libQt5Core.so.5
29# QEventLoop::exec(QFlags<QEventLoop::ProcessEventsFlag>) in /lib/x86_64-linux-gnu/libQt5Core.so.5
30# QCoreApplication::exec() in /lib/x86_64-linux-gnu/libQt5Core.so.5
31# gui::startGui(int&, char**, Tcl_Interp*, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, bool) in openroad
32# ord::tclAppInit(Tcl_Interp*) in openroad
33# Tcl_MainEx in /lib/x86_64-linux-gnu/libtcl8.6.so
34# main in openroad
35# __libc_start_main in /lib/x86_64-linux-gnu/libc.so.6
36# _start in openroad
Segmentation fault (core dumped)
```

donn, 2023-08-09:
@heliboy2003 Would you re-run the environment survey outside the container?

kareefardi, 2023-08-09:
@heliboy2003 There is a folder missing in the issue reproducible. Either there is a bug in OpenLane or you didn't include all the files in issue_reproducible folder. Can you upload your design files ?

heliboy2003, 2023-08-10:
Hi @donn 

I think I did run the environment survey outside the container.

Since I am pretty new here, don't know if I did it correct. I use 

sudo make mount to execute flow.tcl. And if I run the environment survey , it gave me 

==================
root@Ubuntu20:/openlane# python3 ./env.py issue-survey
WARNING: issue-survey appears to be running inside the OpenLane
container.

This makes it difficult to rule out issues with your
environment.

Unless instructed specifically to do so, please run this command
outside the OpenLane container.
---

Kernel: Linux v5.15.0-78-generic
Distribution: centos 7
Python: v3.6.8 (OK)
OpenLane Git Version: 4c74d8b13c39ac20acfd412d90fdd689a959fafc
pip: INSTALLED
python-venv: INSTALLED
---
PDK Version Verification Status: OK
---
Git Log (Last 3 Commits)

4c74d8b1 2023-07-31T17:13:43+00:00 Rename `mca` folders (#1922) - Kareem Farid -  (HEAD -> master, tag: 2023.08.01, origin/master, origin/HEAD)
01e67230 2023-07-25T13:53:21+03:00 [BOT] Update magic (#1914) - Openlane Bot -  (tag: 2023.07.26)
157a6505 2023-07-25T13:02:39+03:00 Update Documentation (#1917) - Mohamed Gaber -  ()
---
Git Remotes

origin	https://github.com/The-OpenROAD-Project/OpenLane.git (fetch)
origin	https://github.com/The-OpenROAD-Project/OpenLane.git (push)
=====================

So I type din exit and ran the environment survey, it gave me

============================

heliboy@Ubuntu20:~/Desktop/OpenLane$ python3 ./env.py issue-survey
permission denied while trying to connect to the Docker daemon socket at unix:///var/run/docker.sock: Get "http://%2Fvar%2Frun%2Fdocker.sock/v1.24/info": dial unix /var/run/docker.sock: connect: permission denied
'NoneType' object is not iterable
Kernel: Linux v5.15.0-78-generic
Distribution: ubuntu 20.04
Python: v3.8.10 (OK)
Critical Alert: No Docker or Docker-compatible container engine was found.
OpenLane Git Version: 4c74d8b13c39ac20acfd412d90fdd689a959fafc
NOT FOUND: Please install pip using your operating system's package manager.
---
PDK Version Verification Status: FAILED
/home/heliboy/Desktop/OpenLane/pdks/sky130A not found.
Traceback (most recent call last):
  File "/home/heliboy/Desktop/OpenLane/dependencies/verify_versions.py", line 76, in verify_versions
    raise Exception(f"{pdk_dir} not found.")
Exception: /home/heliboy/Desktop/OpenLane/pdks/sky130A not found.

Failed to verify sky130A.
---
Git Log (Last 3 Commits)

4c74d8b1 2023-07-31T17:13:43+00:00 Rename `mca` folders (#1922) - Kareem Farid -  (HEAD -> master, tag: 2023.08.01, origin/master, origin/HEAD)
01e67230 2023-07-25T13:53:21+03:00 [BOT] Update magic (#1914) - Openlane Bot -  (tag: 2023.07.26)
157a6505 2023-07-25T13:02:39+03:00 Update Documentation (#1917) - Mohamed Gaber -  ()
---
Git Remotes

origin	https://github.com/The-OpenROAD-Project/OpenLane.git (fetch)
origin	https://github.com/The-OpenROAD-Project/OpenLane.git (push)

============

Not sure if I did it correctly?

heliboy2003, 2023-08-10:
Hi @kareefardi,

Under issue_reproducible folder, I see 3 folders

openlane
pdk
tmp

Anyway, the attached is the design files along with config.json. Appreciate your help

[Test.tar.gz](https://github.com/The-OpenROAD-Project/OpenLane/files/12309150/Test.tar.gz)

kareefardi, 2023-08-10:
> Hi @kareefardi,
> 
> Under issue_reproducible folder, I see 3 folders
> 
> openlane pdk tmp
> 
> Anyway, the attached is the design files along with config.json. Appreciate your help
> 
> [Test.tar.gz](https://github.com/The-OpenROAD-Project/OpenLane/files/12309150/Test.tar.gz)
> 
> [config.tar.gz](https://github.com/The-OpenROAD-Project/OpenLane/files/12309164/config.tar.gz)

Yeah the reproducible was missing tmp folder. Let me try to run the design and reproduce the error.

kareefardi, 2023-08-10:
> Hi @donn
>
> I think I did run the environment survey outside the container.
>
> Since I am pretty new here, don't know if I did it correct. I use
>
> sudo make mount to execute flow.tcl. And if I run the [...]

Do you need root to run docker ?

heliboy2003, 2023-08-16:
I update pip as the message show above, and run environment report again as follows

root@Ubuntu20:/home/heliboy/Desktop/OpenLane# python3 ./env.py issue-survey
Kernel: Linux v5.15.0-78-generic
Distribution: ubuntu 20.04
Python: v3.8.10 (OK)
Container Engine: docker v24.0.5 (OK)
OpenLane Git Version: 4c74d8b13c39ac20acfd412d90fdd689a959fafc
pip: INSTALLED
python-venv: INSTALLED
---
PDK Version Verification Status: FAILED
/home/heliboy/Desktop/OpenLane/pdks/sky130A not found.
Traceback (most recent call last):
  File "/home/heliboy/Desktop/OpenLane/dependencies/verify_versions.py", line 76, in verify_versions
    raise Exception(f"{pdk_dir} not found.")
Exception: /home/heliboy/Desktop/OpenLane/pdks/sky130A not found.

Failed to verify sky130A.
---
Git Log (Last 3 Commits)

4c74d8b1 2023-07-31T17:13:43+00:00 Rename `mca` folders (#1922) - Kareem Farid -  (HEAD -> master, tag: 2023.08.01, origin/master, origin/HEAD)
01e67230 2023-07-25T13:53:21+03:00 [BOT] Update magic (#1914) - Openlane Bot -  (tag: 2023.07.26)
157a6505 2023-07-25T13:02:39+03:00 Update Documentation (#1917) - Mohamed Gaber -  ()
---
Git Remotes

origin	https://github.com/The-OpenROAD-Project/OpenLane.git (fetch)
origin	https://github.com/The-OpenROAD-Project/OpenLane.git (push)

kareefardi, 2023-12-18:
@heliboy2003 for gui crash can you try the following workaround https://github.com/The-OpenROAD-Project/OpenLane/issues/1772#issuecomment-1523113799

donn, 2024-09-12:
Staled out.

closed 2024-09-12 by donn