The-OpenROAD-Project/OpenLane #1950: The module being blackboxed during STA (./flow.tcl)

mthudaa, 2023-08-22:
I have an issues when running `./flow.tcl -design <name_of _project>`. The layout result, any component being blackboxed.

`[WARNING]: Module sky130_fd_sc_hd__tapvpwrvgnd_1 blackboxed during sta
[WARNING]: Module sky130_ef_sc_hd__decap_12 blackboxed during sta
[WARNING]: Module sky130_fd_sc_hd__fill_1 blackboxed during sta
[WARNING]: Module sky130_fd_sc_hd__fill_2 blackboxed during sta
[WARNING]: VSRC_LOC_FILES is not defined. The IR drop analysis will run, but the values may be inaccurate.
`
I dont have idea to fix the modul not be blackboxed in the layout result. please anyone who can help my issues

donn, 2023-08-22:
1. We kindly ask you to use the provided issue form. It helps us solve problems way faster.
2. The black-boxed modules you listed here all have no effect on timing. `tapvpwrvgnd` are TAP cells, `fill` are fill cells and `decap` are decoupling-capacitance cells. i.e. they are to be expected- these warnings only matter if it's a functional cell or a submodule.
3. `VSRC_LOC_FILES` need to be defined by you for accurate IR drop analysis: See the documentation for more info. https://openlane.readthedocs.io/en/latest/reference/configuration.html#ir-drop-analysis

closed 2023-08-22 by donn

mthudaa, 2023-08-22:
i didn comfort with the warn about module in the blackbox, any solution if  i want see the transistor level of my layout?