The-OpenROAD-Project/OpenLane #1952: missleading default for irdrop configuration

proppy, 2023-08-23:
### Description

It looks like the default configuration for irdrop is documented as disable by default:
https://github.com/The-OpenROAD-Project/OpenLane/blob/00caae2b3186bb6b24a0b7db7ee1a6f056ad1702/docs/source/reference/configuration.md#L314

But the implementation seems different:
https://github.com/The-OpenROAD-Project/OpenLane/blob/00caae2b3186bb6b24a0b7db7ee1a6f056ad1702/configuration/general.tcl#L40

Additionally the canonical `spm` example doesn't include any configuration for `VSRC_LOC_FILES` causing the following warning:
```
2023-08-23T01:36:19.3228902Z [32;1m01:36:19[0m | [WARNING]: VSRC_LOC_FILES is not defined. The IR drop analysis will run, but the values may be inaccurate.
```
and ultimatly fails with the following error:
```
2023-08-23T01:36:19.3207819Z [32;1m01:36:19[0m | [ERROR PSM-0093] Voltage on net VDD is not set.
```

### Expected Behavior

The default example either produce a valid irdrop report or either disable the reporting according to the documented default.

### Environment report

```shell
The environment is a bit special here as I'm building openlane and its dependencies from `HEAD` as part of the https://github.com/hdl/conda-eda project.
```

### Reproduction material

N/A

### Relevant log output

```shell
https://github.com/hdl/conda-eda/actions/runs/5944971903/job/16125504708
```

proppy, 2023-08-23:
I wonder if it's related to #1882 ?

kareefardi, 2023-08-25:
> 2023-08-23T01:36:19.3207819Z �[32;1m01:36:19�[0m | [ERROR PSM-0093] Voltage on net VDD is not set.

This happens only with gf180. See https://github.com/The-OpenROAD-Project/OpenROAD/issues/3891. We are working on resolving this specific error.

Otherwise, the documentation needs to be updated. It is also a good idea to add a VSRC_LOC_FILE at least for the smoke-test example.

kareefardi, 2023-09-09:
Doc was updated here https://github.com/The-OpenROAD-Project/OpenLane/pull/1959
Voltage error is fixed here https://github.com/The-OpenROAD-Project/OpenLane/pull/1949 
Opening a new issue for an example for `VSRC_LOC_FILE`

closed 2023-09-09 by kareefardi