The-OpenROAD-Project/OpenLane #1955: addition of block of lines in synth.tcl is not reflecting in synthesis.log

Vinayakamk, 2023-08-26:
### Description

hello
i just want to add a SYNTH_RETIME as a config varible,or any if statemnt in synth.tcl of openlane,which ask user whether to do retime or not,so that i can do retiming just by passing true for the SYnTH_RETIME in config.json.i am imitating SYNTH_AUTONAME.
[image]

but synth retime is not reflecting in sythesis.log where in synth autoname happened.
though the commands are valid in yosys 

### Proposal

the synth.tcl  or the flow.tcl should allow the user defined tcl commands to execute normally(as in this case its not happening).
or atleast it should and excute userdefined .tcl  where ever we want during synthesis

vijayank88, 2023-08-26:
@Vinayakamk 
Share your tcl script and `config.json`.
Try with `spm` design or share a design which you're trying to implement.

Vinayakamk, 2023-08-26:
hello sir,im trying with openlanes picorv32 
either i want retiming should happen in synthesis stage like other config.json variable happen.
[synth.txt](https://github.com/The-OpenROAD-Project/OpenLane/files/12445075/synth.txt)  **//added retime if statement in line no 348**
[config.txt](https://github.com/The-OpenROAD-Project/OpenLane/files/12445072/config.txt) **//added true value for the synth_retime**

if it is not possible,flow.tcl should take my tcl files which does retiming,on true value set in config.json.

[user_env.txt](https://github.com/The-OpenROAD-Project/OpenLane/files/12445069/user_env.txt)
[automatc.txt](https://github.com/The-OpenROAD-Project/OpenLane/files/12445021/automatc.txt)
[retime.txt](https://github.com/The-OpenROAD-Project/OpenLane/files/12445024/retime.txt)

(i have sent my 3 tcl files in .txt)
i want to incorporate the retiming concept it flow.tcl in any of the above way

donn, 2023-08-26:
Please re-file this as an issue. We'll need a lot more information to figure out why this is happening as technically, modifying a script should modify the execution.

closed 2023-08-26 by donn

Vinayakamk, 2023-08-30:
sure thanks for the reply,,,