The-OpenROAD-Project/OpenLane #1958: [ERROR DRT-0218] Guide is not connected to design.

akhilesh911, 2023-08-29:
### Description

Good morning,
I am trying to run the OpenLANE flow on my design. There is an error in detailed routing step. There was no error when set::env(FP_PDN_CORE_RING) {0}. When I set ::env(FP_PDN_CORE_RING) {1}, the error occured.

[26-detailed.zip](https://github.com/The-OpenROAD-Project/OpenLane/files/12460881/26-detailed.zip)

### Expected Behavior

No error expected.

### Environment report

```shell
Kernel: Linux v5.4.0-150-generic
Distribution: ubuntu 18.04
Python: v3.9.16 (OK)
Container Engine: docker v20.10.24 (OK)
OpenLane Git Version: 90d369bcce5cf1133c28291bc3c8b6473a19ff62
NOT FOUND: Please install pip using your operating system's package manager.
---
PDK Version Verification Status: OK
---
Git Log (Last 3 Commits)

90d369b 2022-12-13T22:59:28+02:00 Iterate over `$libs` in read_libs (#1570) - Kareem Farid -  (HEAD -> master, tag: 2022.12.14, origin/master, origin/HEAD)
0d13487 2022-12-13T19:05:19+02:00 Add `report_parasitic_annotation` to sta (#1569) - Kareem Farid -  ()
d20c7b4 2022-12-13T12:32:13+02:00 Expose magic polygon subcells (#1566) - Kareem Farid -  ()
---
Git Remotes

origin	https://github.com/The-OpenROAD-Project/OpenLane.git (fetch)
origin	https://github.com/The-OpenROAD-Project/OpenLane.git (push)
```

### Reproduction material

[issue_reproducible.zip](https://github.com/The-OpenROAD-Project/OpenLane/files/12460852/issue_reproducible.zip)

### Relevant log output

```shell
[STEP 1]
[INFO]: Running Synthesis (log: designs/picosoc/runs/run42/logs/synthesis/1-synthesis.log)...
[STEP 2]
[INFO]: Running Single-Corner Static Timing Analysis (log: designs/picosoc/runs/run42/logs/synthesis/2-sta.log)...
[STEP 3]
[INFO]: Running Initial Floorplanning (log: designs/picosoc/runs/run42/logs/floorplan/3-initial_fp.log)...
[INFO]: Floorplanned with width 1933.68 and height 1932.56.
[STEP 4]
[INFO]: Running IO Placement...
[STEP 5]
[INFO]: Performing Manual Macro Placement (log: designs/picosoc/runs/run42/logs/placement/5-macro_placement.log)...
[STEP 6]
[INFO]: Running Tap/Decap Insertion (log: designs/picosoc/runs/run42/logs/floorplan/6-tap.log)...
[INFO]: Power planning with power {VDD} and ground {VSS}...
[STEP 7]
[INFO]: Generating PDN (log: designs/picosoc/runs/run42/logs/floorplan/7-pdn.log)...
[STEP 8]
[INFO]: Running Global Placement (log: designs/picosoc/runs/run42/logs/placement/8-global.log)...
[STEP 9]
[INFO]: Running Placement Resizer Design Optimizations (log: designs/picosoc/runs/run42/logs/placement/9-resizer.log)...
[STEP 10]
[INFO]: Running Detailed Placement (log: designs/picosoc/runs/run42/logs/placement/10-detailed.log)...
[STEP 11]
[INFO]: Running Clock Tree Synthesis (log: designs/picosoc/runs/run42/logs/cts/11-cts.log)...
[STEP 12]
[INFO]: Running Placement Resizer Timing Optimizations (log: designs/picosoc/runs/run42/logs/cts/12-resizer.log)...
[INFO]: Starting ECO flow...
[STEP 13]
[INFO]: Running Global Routing Resizer Timing Optimizations (log: designs/picosoc/runs/run42/logs/routing/eco_1/13-resizer.log)...
[STEP 14]
[INFO]: Running Detailed Placement (log: designs/picosoc/runs/run42/logs/routing/eco_1/14-diode_legalization.log)...
[STEP 15]
[INFO]: Running Global Routing (log: designs/picosoc/runs/run42/logs/routing/eco_1/15-global.log)...
[STEP 16]
[INFO]: Writing Verilog (log: designs/picosoc/runs/run42/logs/routing/eco_1/15-global_write_netlist.log)...
[STEP 17]
[INFO]: Running Detailed Routing (log: designs/picosoc/runs/run42/logs/routing/eco_1/17-detailed.log)...
[INFO]: No DRC violations after detailed routing.
[STEP 18]
[INFO]: Checking Wire Lengths (log: designs/picosoc/runs/run42/logs/routing/eco_1/18-wire_lengths.log)...
[STEP 19]
[INFO]: Running SPEF Extraction at the nom process corner (log: designs/picosoc/runs/run42/logs/signoff/eco_1/19-parasitics_extraction.nom.log)...
[STEP 20]
[INFO]: Running Multi-Corner Static Timing Analysis at the nom process corner (log: designs/picosoc/runs/run42/logs/signoff/eco_1/20-rcx_mcsta.nom.log)...
[STEP 21]
[INFO]: Running Single-Corner Static Timing Analysis at the nom process corner (log: designs/picosoc/runs/run42/logs/signoff/eco_1/21-rcx_sta.log)...
[INFO]: [ECO: 1] Checking output...
[INFO]: [ECO: 1] Generating buffer insertion script...
[INFO]: [ECO: 1] Timing violations found, performing another ECO iteration...
[INFO]: [ECO: 2] Starting iteration...
[INFO]: [ECO: 2] Applying fixes...
[STEP 22]
[INFO]: Running Global Routing Resizer Timing Optimizations (log: designs/picosoc/runs/run42/logs/routing/eco_2/22-resizer.log)...
[STEP 23]
[INFO]: Running Detailed Placement (log: designs/picosoc/runs/run42/logs/routing/eco_2/23-diode_legalization.log)...
[STEP 24]
[INFO]: Running Global Routing (log: designs/picosoc/runs/run42/logs/routing/eco_2/24-global.log)...
[STEP 25]
[INFO]: Writing Verilog (log: designs/picosoc/runs/run42/logs/routing/eco_2/24-global_write_netlist.log)...
[STEP 26]
[INFO]: Running Detailed Routing (log: designs/picosoc/runs/run42/logs/routing/eco_2/26-detailed.log)...
[ERROR]: during executing openroad script /openlane/scripts/openroad/droute.tcl
[ERROR]: Log: designs/picosoc/runs/run42/logs/routing/eco_2/26-detailed.log
[ERROR]: Last 10 lines:
  complete 30000 nets.
[INFO DRT-1000] Pin _56443_/CLK not in any guide. Attempting to patch guides to cover (at least part of) the pin.
[WARNING DRT-1001] No guide in the pin neighborhood
[WARNING DRT-0215] Pin _56443_/CLK not covered by guide.
[WARNING DRT-0215] Pin _56443_/CLK not covered by guide.
[WARNING DRT-0225] clknet_leaf_642_clk 1 pin not visited, fall back to feedthrough mode.
[WARNING DRT-0224] clknet_leaf_642_clk 1 pin not visited, number of guides = 14.
[ERROR DRT-0218] Guide is not connected to design.
Error: droute.tcl, 38 DRT-0218
child process exited abnormally
```

vijayank88, 2023-08-29:
@akhilesh911 
With the current test case missing library list
```
set ::env(EXTRA_LIBS) { /openlane/designs/picosoc/../../pdks/gf180mcuC/libs.ref/gf180mcu_fd_ip_sram/liberty/gf180mcu_fd_ip_sram__sram256x8m8wm1__tt_025C_5v00.lib  /openlane/designs/picosoc/../timer/runs/run2/results/final/lib/timer.lib  /openlane/designs/picosoc/../i2c/runs/run1/results/final/lib/i2c_master_top.lib  /openlane/designs/picosoc/../gpio_top/runs/run1/results/final/lib/gpio_top.lib  /openlane/designs/picosoc/../watchdog/runs/run2/results/final/lib/watchdog.lib};
```
ECO flow currently not supported.

You're using `2022-12-13` commit, update to latest commit and run the flow. If there is any issue again, attach latest test case generated by the flow.

akhilesh911, 2023-08-29:
@vijayank88 This is the latest test case generated. Any solution with respect to 2022-12-13 commit? Kindly suggest.

vijayank88, 2023-08-29:
I am not able to reproduce the error as missing libraries path.

akhilesh911, 2023-08-29:
Should I provide the missing library files? Sorry I did not understand.

vijayank88, 2023-08-29:
> Should I provide the missing library files? Sorry I did not understand.

yes. attach all libraries as per `EXTRA_LIBS` variables.

akhilesh911, 2023-08-29:
[lib.zip](https://github.com/The-OpenROAD-Project/OpenLane/files/12461545/lib.zip)

vijayank88, 2023-08-29:
@akhilesh911 
Similar error reported by another user and it got resolved. https://github.com/The-OpenROAD-Project/OpenROAD/issues/1197
So I recommend you to update your OpenLane and try again.

akhilesh911, 2023-08-29:
@vijayank88  Actually in updated openlane, there is error in yosys check. That's why I am using old version. Can you please suggest any solution with respect to old version only?

vijayank88, 2023-08-29:
Consider if the error got fixed in OpenROAD upstream, still you need to update OpenLane to get it fixed.

To save your time and others, check the synthesis log and try to resolve by updating your RTL file to proceed further.

kareefardi, 2023-08-29:
@akhilesh911 can you open another issue for synth checks ?

closed 2023-08-29 by kareefardi