The-OpenROAD-Project/OpenLane #1960: addition of block of lines in synth.tcl is not reflecting in synthesis.log

Vinayakamk, 2023-08-30:
### Description

hello
i just want to add a SYNTH_RETIME[new variable not in configuration documentation of openlane] as a config varible,or any **if statemnt** in synth.tcl of openlane,which ask user whether to do retime or not,so that i can do retiming just by passing true for the SYnTH_RETIME in config.json.i am imitating SYNTH_AUTONAME.
[image]

The issue is, though i tried setting synth_retime as true .its is not excuting ,as like any other synth variable do.
so i tried my own tcl script,,
either i want the variable to work or flow.tcl should exceute my retime.tcl on if true condion.

### Expected Behavior

synth_retime : 1, in config.json should reflect 
it should execute retiming.{abc -dff -D 1}

### Environment report

```shell
{below i pasted which i expected in synthesis.log to have after setting synth_retime :1,note the below snippent is portion,which is too long to paste 

5. Executing ABC pass (technology mapping using ABC).

5.1. Summary of detected clock domains:
  2673 cells in clk={ }, en={ }, arst={ }, srst={ }

5.2. Extracting gate netlist of module `\mkinst_pwmapb' to `<abc-temp-dir>/input.blif'..
No matching clock domain found. Not extracting any FF cells.
Extracted 0 gates and 0 wires to a netlist network with 0 inputs and 0 outputs.
Don't call ABC as there is nothing to map.
Removing temp directory.

5.3. Summary of detected clock domains:
  6 cells in clk={ }, en={ }, arst={ }, srst={ }

5.4. Extracting gate netlist of module `\sky130_fd_sc_hd__sdfxtp_4' to `<abc-temp-dir>/input.blif'..
No matching clock domain found. Not extracting any FF cells.
Extracted 5 gates and 9 wires to a netlist network with 4 inputs and 1 outputs.

5.4.1. Executing ABC.
Running ABC command: "<yosys-exe-dir>/yosys-abc" -s -f <abc-temp-dir>/abc.script 2>&1
ABC: ABC command line: "source <abc-temp-dir>/abc.script".
ABC: 
ABC: + read_blif <abc-temp-dir>/input.blif 
ABC: + read_library <abc-temp-dir>/stdcells.genlib 
ABC: Entered genlib library with 13 gates from file "<abc-temp-dir>/stdcells.genlib".
ABC: + strash 
ABC: + &get -n 
ABC: + &fraig -x 
ABC: + &put 
ABC: + scorr 
ABC: Warning: The network is combinational (run "fraig" or "fraig_sweep").
ABC: + dc2 
ABC: + dretime 
ABC: + retime -o -D 1 
ABC: + strash 
ABC: + &get -n 
ABC: + &dch -f 
ABC: + &nf -D 1 
ABC: + &put 
ABC: + write_blif <abc-temp-dir>/output.blif 

5.4.2. Re-integrating ABC results.
ABC RESULTS:               MUX cells:        1
ABC RESULTS:        internal signals:        4
ABC RESULTS:           input signals:        4
ABC RESULTS:          output signals:        1
Removing temp directory.

5.5. Summary of detected clock domains:
  6 cells in clk={ }, en={ }, arst={ }, srst={ }
[... 211 lines cut ...]
ABC: + &nf -D 1 
ABC: + &put 
ABC: + write_blif <abc-temp-dir>/output.blif 

5.16.2. Re-integrating ABC results.
ABC RESULTS:               MUX cells:        1
ABC RESULTS:        internal signals:        5
ABC RESULTS:           input signals:        5
ABC RESULTS:          output signals:        1
Removing temp directory.

5.17. Summary of detected clock domains:
  7 cells in clk={ }, en={ }, arst={ }, srst={ }
```

### Reproduction material

this is my script ,,,which should be called id synth_variable is not happeninig on true condition
[automatc.txt](https://github.com/The-OpenROAD-Project/OpenLane/files/12471441/automatc.txt)

### Relevant log output

```shell
no special log file ,as it is not reflecting in log file
```

donn, 2023-09-03:
You pasted the log in the environment report section then said there was no log output

Vinayakamk, 2023-09-03:
hello,
the  log output that i pasted is the output that i got when i ran my automatc.tcl in console,
my problem is i want that output to be in synthesis.log,on true condtion of **if statement** that i added in synth.tcl[SYNTH_RETIME].which is not happening during synthesis.

 now plz let me know the way i added the block of tcl code related to synth_retime is right or wrong?or any where should i change so that if i pass synth_retime :1 in config.json, it should do retiming and log output should be in synthesis.log
 
 thank you

closed 2023-09-06 by Vinayakamk