The-OpenROAD-Project/OpenLane #1970: OpenROAD memory crash in antenna repaire in global routing

mo-hosni, 2023-09-03:
### Description

Memory utilization reaches 100% (67.1 GB) then the OR crashes in antenna repair in global routing. I faced this on 2 different macros since 30/August/2023. Changing some variables such as `FP_CORE_UTIL` or `PL_TARGET_DENSITY` gets the flow running but I think this should be investigated. 

Attached a screenshot of the memory utilization behavior, the utilization increases linearly since the start of global routing till it reaches 100% utilization.

[image]

### Expected Behavior

OpenROAD shouldn't need 67GB of RAM to do antenna repair.

### Environment report

```shell
Kernel: Linux v6.2.0-31-generic
Distribution: ubuntu 23.04
Python: v3.11.4 (OK)
Container Engine: docker v24.0.5 (OK)
OpenLane Git Version: 2264b1240bacca90f8dfd74b8f5d62b7d618038e
pip: INSTALLED
python-venv: INSTALLED
---
PDK Version Verification Status: OK
---
Git Log (Last 3 Commits)

2264b124 2023-09-03T13:16:28+00:00 bump open_pdks version (#1969) - Kareem Farid -  (HEAD -> master, origin/master, origin/HEAD)
6a293792 2023-09-03T14:02:52+03:00 Add Banner for OpenLane 2 (#1966) - Mohamed Gaber -  ()
698c0fac 2023-09-03T14:02:00+03:00 Partially Revert "Illegal Overlap Check Fix" (#1967) - Mohamed Gaber -  ()
---
Git Remotes

origin	git@github.com:The-OpenROAD-Project/OpenLane.git (fetch)
origin	git@github.com:The-OpenROAD-Project/OpenLane.git (push)
```

### Reproduction material

[issue_reproducible.zip](https://github.com/The-OpenROAD-Project/OpenLane/files/12506385/issue_reproducible.zip)

### Relevant log output

```shell
[ERROR]: during executing openroad script /openlane/scripts/openroad/groute.tcl
[ERROR]: Log: ../home/hosni/ML-SOC/caravel-ML-SoC/openlane/SW_SHA256_AHBL/runs/23_09_03_17_57/logs/routing/23-global.log
[ERROR]: Last 10 lines:
met3             53990         13712           25.40%             0 /  0 /  0
met4             29572          5140           17.38%             0 /  0 /  0
---------------------------------------------------------------------------------------
Total           236746         91689           38.73%             0 /  0 /  0

[INFO GRT-0018] Total wirelength: 906839 um
[INFO GRT-0014] Routed nets: 13280
[INFO GRT-0006] Repairing antennas, iteration 1.
[INFO GRT-0043] No OR_DEFAULT vias defined.
child killed: kill signal
```

donn, 2023-09-03:
We're working on getting OR updated.

donn, 2023-09-06:
OR should be updated now..?

kareefardi, 2023-09-09:
Still facing the same issue using OpenROAD `6dacde2b1f4ddc58ea64e6d889bc8665762da095`

kareefardi, 2023-12-18:
No longer facing a crash or high memory consumption with OpenLane `66e938bcb87d7f5b9a0487c018e6a0083b289d5d`

closed 2023-12-18 by kareefardi