The-OpenROAD-Project/OpenLane #1972: Error DRT-0218 Guide is not connected to design

jchin2, 2023-09-04:
### Description

Making blackbox of layout that has dnwell and nwell guard rings. Goal to have openlane recognize, route and connect to the blackbox. Openlane seemingly can't find the pins. Initially started with pins within the prboundary and then moved them outside plus elongating the pins.  Omitted `FP_PDN_HOOK` to see that it doesn't make a difference. 
Relevant files is included in the current_state_09042023.zip folder. 

### Expected Behavior

Openlane to recognize, route and connect the pins to the blackboxed macro layout. 

### Environment report

```shell
Kernel: Linux v6.2.0-31-generic
Distribution: ubuntu 22.04
Python: v3.11.3 (OK)
Container Engine: docker v24.0.5 (OK)
OpenLane Git Version: d054702b2cce04761cc2bc598f6b95c9d8ca7c6c
pip: INSTALLED
python-venv: INSTALLED
---
PDK Version Verification Status: FAILED
/home/jchin2/sky130/openlane_src/pdks/sky130A not found.
Traceback (most recent call last):
  File "/home/jchin2/sky130/openlane_src/dependencies/verify_versions.py", line 76, in verify_versions
    raise Exception(f"{pdk_dir} not found.")
Exception: /home/jchin2/sky130/openlane_src/pdks/sky130A not found.

Failed to verify sky130A.
---
Git Log (Last 3 Commits)

d054702 2023-07-19T16:09:15+03:00 remove `unset_propagated_clock` (#1908) - passant5 -  (grafted, HEAD, tag: 2023.07.19)
---
Git Remotes

origin	https://github.com/The-OpenROAD-Project/OpenLane (fetch)
origin	https://github.com/The-OpenROAD-Project/OpenLane (push)
```

### Reproduction material

[issue_reproducible.zip](https://github.com/The-OpenROAD-Project/OpenLane/files/12514187/issue_reproducible.zip)
[current_state_09042023.zip](https://github.com/The-OpenROAD-Project/OpenLane/files/12514191/current_state_09042023.zip)

### Relevant log output

```shell
(base) jchin2@short-simple:~/sky130/caravel_user_project_analog$ make blackbox_test_3
cd openlane && make blackbox_test_3
make[1]: Entering directory '/home/jchin2/sky130/caravel_user_project_analog/openlane'
# blackbox_test_3
mkdir -p ./blackbox_test_3/runs/23_09_04_07_10 
rm -rf ./blackbox_test_3/runs/blackbox_test_3
ln -s $(realpath ./blackbox_test_3/runs/23_09_04_07_10) ./blackbox_test_3/runs/blackbox_test_3
docker run -it -v $(realpath /home/jchin2/sky130/caravel_user_project_analog/openlane/..):$(realpath /home/jchin2/sky130/caravel_user_project_analog/openlane/..) -v /home/jchin2/sky130/pdks:/pdk -v /home/jchin2/sky130/caravel:/home/jchin2/sky130/caravel -v /home/jchin2/sky130/openlane_src:/openlane -e PDK_ROOT=/pdk -e PDK=sky130A -e MISMATCHES_OK=1 -e CARAVEL_ROOT=/home/jchin2/sky130/caravel -e OPENLANE_RUN_TAG=23_09_04_07_10 -u 1000:1000 \
	efabless/current-local-amd64 sh -c "flow.tcl -design $(realpath ./blackbox_test_3) -save_path $(realpath ..) -save -tag 23_09_04_07_10 -overwrite -ignore_mismatches"
OpenLane d054702b2cce04761cc2bc598f6b95c9d8ca7c6c
All rights reserved. (c) 2020-2022 Efabless Corporation and contributors.
Available under the Apache License, version 2.0. See the LICENSE file for more details.

[INFO]: Using configuration in '../home/jchin2/sky130/caravel_user_project_analog/openlane/blackbox_test_3/config.json'...
[INFO]: PDK Root: /pdk
[INFO]: Process Design Kit: sky130A
[INFO]: Standard Cell Library: sky130_fd_sc_hd
[INFO]: Optimization Standard Cell Library: sky130_fd_sc_hd
[INFO]: Run Directory: /home/jchin2/sky130/caravel_user_project_analog/openlane/blackbox_test_3/runs/23_09_04_07_10
[INFO]: Saving runtime environment...
[INFO]: Preparing LEF files for the nom corner...
[INFO]: Preparing LEF files for the min corner...
[INFO]: Preparing LEF files for the max corner...
[... 94 lines cut ...]
    (procedure "detailed_routing_tritonroute" line 14)
    invoked from within
"detailed_routing_tritonroute {*}$args"
    (procedure "detailed_routing" line 2)
    invoked from within
"detailed_routing"
    (procedure "run_routing" line 32)
    invoked from within
"run_routing"
    (procedure "run_routing_step" line 7)
    invoked from within
"run_routing_step"} -errorline 1
[INFO]: Saving current set of views in '../home/jchin2/sky130/caravel_user_project_analog/openlane/blackbox_test_3/runs/23_09_04_07_10/results/final'...
[INFO]: Generating final set of reports...
[INFO]: Created manufacturability report at '../home/jchin2/sky130/caravel_user_project_analog/openlane/blackbox_test_3/runs/23_09_04_07_10/reports/manufacturability.rpt'.
[INFO]: Created metrics report at '../home/jchin2/sky130/caravel_user_project_analog/openlane/blackbox_test_3/runs/23_09_04_07_10/reports/metrics.csv'.
[INFO]: Saving runtime environment...
[ERROR]: Flow failed.
[INFO]: The failure may have been because of the following warnings:
[WARNING]: 9 warnings found by linter

make[1]: *** [Makefile:73: blackbox_test_3] Error 255
make[1]: Leaving directory '/home/jchin2/sky130/caravel_user_project_analog/openlane'
make: *** [Makefile:72: blackbox_test_3] Error 2
```

kareefardi, 2023-09-04:
I used an old script in OpenLane to shift change the origin of the macro to 0,0 and shift shapes accordingly in the lef view of the marco and I was able to pass Pin not covered by guide error. I ran into another error but that's a different issue.
 
I am not entirely sure whether there is an issue with the lef or OpenROAD expects shapes in the lef in a certain way. The above is a temporary workaround because you would need both the gds and lef shifted in the same way. I believe that you are using magic and there should be a way to shift the geometries. I can't recall how it is done but I will look for it.

jchin2, 2023-09-04:
The way I generated the gds file is in klayout. I don't really know how to change the origin of the cell in klayout... the image shows one of the cells' origin is not centered but i did not think too much about it. 
[image]
I guess what I can do at the moment is to try to make the origin of the cell centered in the middle?

jchin2, 2023-09-04:
Changing the top cell and subcell origin, regenerating the lef made the flow work! I made sure to uss FP_PDN_MACROHOOKS to conform to the power pin and gnd pin in top verilog. I have included the updated files in the following .zip folder.
[current_state_09042023_1526pm.zip](https://github.com/The-OpenROAD-Project/OpenLane/files/12516261/current_state_09042023_1526pm.zip)
Leaving steps here for adjusting origins: 
In klayout I set the top cell origin via Edit > Cell > Adjust Origin. 
As for the subcells, they are set by show as new top > highlight all the cells > Edit > Selection > Make Cell > choose origin and name to replace.    
Current environment exports:
export OPENLANE_ROOT=$PWD/openlane_src
echo $OPENLANE_ROOT
export PDK_ROOT=$PWD/pdks
echo $PDK_ROOT
export CARAVEL_ROOT=$PWD/caravel
echo $CARAVEL_ROOT
export SKYWATER_COMMIT=f70d8ca46961ff92719d8870a18a076370b85f6c
echo $SKYWATER_COMMIT
export OPEN_PDKS_COMMIT=78b7bc32ddb4b6f14f76883c2e2dc5b5de9d1cbc
echo $OPEN_PDKS_COMMIT
export OPENLANE_TAG=2023.07.19
echo $OPENLANE_TAG
export OPENLANE_IMAGE_NAME=efabless/current-local-amd64

kareefardi, 2023-09-09:
@jchin2 I noticed there is activity relating to this in slack. Please add the resolution here if any.

donn, 2023-12-18:
Staled out.

closed 2023-12-18 by donn