The-OpenROAD-Project/OpenLane #1973: IO Pads generation in OpenLANE

akhilesh911, 2023-09-05:
Good morning all,
I am trying to create IO pads frame for my design following https://github.com/efabless/OpenLane/tree/master/scripts/topModuleGen
I have written json files required by topModuleGen.py 
It gives an error
[image]
kindly let me know how to use this script.

kareefardi, 2023-09-09:
There is a method that can be used in OpenLane interactive mode `padframe_gen_batch`. There is an example of using it here https://github.com/efabless/caravel/blob/93596a3651699b91f461954c6d7d3140c3aca0e0/openlane/chip_io/interactive.tcl#L36. There are bunch of configuration variables needed here https://github.com/efabless/caravel/blob/chip_io_config_updates/openlane/chip_io/config.tcl. Ultimately the script that's being used is https://github.com/The-OpenROAD-Project/OpenLane/blob/master/scripts/odbpy/padringer.py.

The documentation needs to be updated to reflect that.

Note:
You shouldn't be using https://github.com/The-OpenROAD-Project/OpenLane not https://github.com/efabless/OpenLane

akhilesh911, 2023-09-11:
@kareefardi is there any way to create padframe using https://github.com/efabless/OpenLane ??
Kindly suggest.

donn, 2024-09-12:
There is no padframe generation in OpenLane.

closed 2024-09-12 by donn