The-OpenROAD-Project/OpenLane #1977: requirement.txt not found during make pdk

78yashn, 2023-09-07:
### Description

I am trying to install OPENLANE with a script(the script is attached), when I give the command ''make pdk'', following error occurs
[image]

I'm absolute beginner in openlane and ubuntu, hence can't understand the error.
script file: - 
[openlane_setup.zip](https://github.com/The-OpenROAD-Project/OpenLane/files/12548086/openlane_setup.zip)

Im following the below playlist: https://youtube.com/playlist?list=PL6lEfVxcwjW3tOnmmU_4WAqRYppDvGLHM&si=8n1bkk7bmQmUELyv 

### Expected Behavior

Installation should have been completed

### Environment report

```shell
python3: can't open file './env.py': [Errno 2] No such file or directory
```

### Reproduction material

#!/bin/bash

Log_File=./installation.log

CommandLog () {
author='APR-Training'
startdate=$(date)
scriptName='installation.log'

echo "${author}"
echo "[$startdate]:[INSTALLATION]: This is ${scriptName}, which shows the loging of install_openLane.sh:"

set -ex
echo "=============================================="
echo "----KICK OFF Basic Toolschain INSTALATTION----"
echo "=============================================="
echo

sudo apt update
sudo apt install build-essential
sudo apt-get install lsb-core
sudo apt install git-all
sudo apt install ufw
sudo apt-get install cmake
sudo apt-get update -y
sudo apt-get install -y device-tree-compiler
sudo apt-get install -y make
sudo apt-get install apt-transport-https ca-certificates curl software-properties-common
sudo apt install containerd
sudo apt-get install python-pip
sudo apt-get install python3-pip
sudo pip install --upgrade pip

echo "=================================="
echo "---KICK OFF DOCKER INSTALLATION---"
echo "=================================="

sudo apt install docker.io

sudo systemctl start docker

sudo systemctl enable docker

echo "=================================="
echo "----------DOCKER  ENABLED---------"
echo "=================================="

docker --version

sudo apt install klayout

# making tools folder

mkdir -p apr; cd apr

echo "=================================="
echo "--------MAGIC INSTALLATION--------"
echo "=================================="

# pre-req for magic
sudo apt-get install m4
sudo apt-get install csh
sudo apt-get install libx11-dev
sudo apt-get install libncurses-dev
sudo apt-get install tcl-dev tk-dev
sudo apt-get install blt-dev
sudo apt-get install freeglut3
sudo apt-get install libgl1-mesa-dev
sudo apt-get install libglu1-mesa-dev

git clone https://github.com/RTimothyEdwards/magic.git
cd magic
./configure
sudo make
sudo make install
magic --version
cd ../

echo "======================================"
echo "----KICK OFF OpenLane INSTALATTION----"
echo "======================================"
echo

git clone https://github.com/The-OpenROAD-Project/OpenLane.git
cd OpenLane/
git checkout v0.23
# Default PDK_ROOT is $(pwd)/pdks. If you want to install the PDK at a differnt location, uncomment the next line.
#export PDK_ROOT=<absolute path to where skywater-pdk and open_pdks will reside>
make openlane
make pdk
make test # This is to test that the flow and the pdk were properly inst
} 

CommandLog | tee "${Log_File}"
#ErrorLog "Found error in logic to logging for investigation" | tee "${Log_File}"

### Relevant log output

```shell
yashn@yashn-HP-Laptop-15s-gr0xxx:~/Desktop/pr3/11dbef9b83075b06b9ec90fddb8dc96f/apr/OpenLane$ sudo make pdk
[sudo] password for yashn: 
cd /home/yashn/Desktop/pr3/11dbef9b83075b06b9ec90fddb8dc96f/apr/OpenLane/pdks/skywater-pdk && \
	git checkout main && git submodule init && git pull --no-recurse-submodules && \
	git checkout -qf 00bdbcf4a3aa922cc1f4a0d0cd8b80dbd73149d3
M	third_party/make-env
Previous HEAD position was 00bdbcf Merge pull request #323 from mithro/fix-make-submodule
Switched to branch 'main'
Your branch is up to date with 'origin/main'.
Already up to date.
cd /home/yashn/Desktop/pr3/11dbef9b83075b06b9ec90fddb8dc96f/apr/OpenLane/pdks/skywater-pdk && \
	git submodule update --init libraries/sky130_fd_sc_hd/latest && \
	git submodule update --init libraries/sky130_fd_io/latest && \
	git submodule update --init libraries/sky130_fd_sc_hvl/latest && \
	make timing
make[1]: Entering directory '/home/yashn/Desktop/pr3/11dbef9b83075b06b9ec90fddb8dc96f/apr/OpenLane/pdks/skywater-pdk'
cd /home/yashn/Desktop/pr3/11dbef9b83075b06b9ec90fddb8dc96f/apr/OpenLane/pdks/skywater-pdk; git submodule update --init third_party/make-env
source /home/yashn/Desktop/pr3/11dbef9b83075b06b9ec90fddb8dc96f/apr/OpenLane/pdks/skywater-pdk/env/conda/bin/activate && conda env update --name skywater-pdk-scripts --file environment.yml
Retrieving notices: ...working... done
Collecting package metadata (repodata.json): done
Solving environment: done
Installing pip dependencies: / Ran pip subprocess with arguments:
['/home/yashn/Desktop/pr3/11dbef9b83075b06b9ec90fddb8dc96f/apr/OpenLane/pdks/skywater-pdk/env/conda/envs/skywater-pdk-scripts/bin/python', '-m', 'pip', 'install', '-U', '-r', '/home/yashn/Desktop/pr3/11dbef9b83075b06b9ec90fddb8dc96f/apr/OpenLane/pdks/skywater-pdk/condaenv.8xpkz01_.requirements.txt', '--exists-action=b']
Pip subprocess output:

Pip subprocess error:
ERROR: 404 Client Error: FileNotFoundError for url: file:///requirements.txt

failed

CondaEnvException: Pip failed

make[1]: *** [/home/yashn/Desktop/pr3/11dbef9b83075b06b9ec90fddb8dc96f/apr/OpenLane/pdks/skywater-pdk/third_party/make-env/conda.mk:145: /home/yashn/Desktop/pr3/11dbef9b83075b06b9ec90fddb8dc96f/apr/OpenLane/pdks/skywater-pdk/env/conda/envs/skywater-pdk-scripts/bin/python] Error 1
```

vijayank88, 2023-09-07:
@78yashn 
Just run following steps will setup everything automatically. No need to run `make pdk` separately.\
Steps to follow: https://github.com/The-OpenROAD-Project/OpenLane#installation-the-short-version

donn, 2023-09-07:
OpenLane v0.23 is no longer supported. We kindly ask you to follow the official documentation here: https://openlane.readthedocs.io/en/latest/getting_started/installation/installation_ubuntu.html

I'm curious as to which instructions you were following.

closed 2023-09-07 by donn