The-OpenROAD-Project/OpenLane #1982: Placement Resizer optimization error with recently version of OpenLane

Baungarten-CINVESTAV, 2023-09-11:
### Description

Hi all,

I create this issues due to i'm working on a project with OpenRAM, i create a module that combine two SRAM macros to work as one, the problem comes when i update OpenLane last week an the same project now doesn't work.

I get the following error message:
[image]

- The project works with the OpenLane version:  6ca12d8eab5aaeef8a62431e798bfd06c5495a0e
- The project does not work with the OpenLane version: 456838bffcd80291282e4485d276ada28680447e

### Expected Behavior

[image]

### Environment report

```shell
open_pdks 1341f54f5ce0c4955326297f235e4ace1eb6d419
WARNING: issue-survey appears to be running inside the OpenLane
container.

This makes it difficult to rule out issues with your
environment.

Unless instructed specifically to do so, please run this command
outside the OpenLane container.
---

Kernel: Linux v5.15.0-79-generic
Distribution: centos 7
Python: v3.6.8 (OK)
OpenLane Git Version: 456838bffcd80291282e4485d276ada28680447e
pip: INSTALLED
python-venv: INSTALLED
---
PDK Version Verification Status: OK
---
Git Log (Last 3 Commits)

456838b 2023-09-06T14:48:32+03:00 Remove CVC (#1976) - Mohamed Gaber -  (grafted, HEAD -> master, tag: 2023.09.07, origin/master, origin/HEAD)
---
Git Remotes

origin	https://github.com/The-OpenROAD-Project/OpenLane.git (fetch)
origin	https://github.com/The-OpenROAD-Project/OpenLane.git (push)
```

### Reproduction material

I attach the project:
[SRAM_32_1024_2_row.zip](https://github.com/The-OpenROAD-Project/OpenLane/files/12577880/SRAM_32_1024_2_row.zip)

### Relevant log output

```shell
[STEP 15]
[INFO]: Running Single-Corner Static Timing Analysis (log: designs/SRAM_32_1024_2_row/runs/RUN_2023.09.11_15.29.28/logs/cts/15-cts_sta.log)...
[STEP 16]
[INFO]: Running Placement Resizer Timing Optimizations (log: designs/SRAM_32_1024_2_row/runs/RUN_2023.09.11_15.29.28/logs/cts/16-resizer.log)...
[STEP 17]
[INFO]: Running Global Routing Resizer Design Optimizations (log: designs/SRAM_32_1024_2_row/runs/RUN_2023.09.11_15.29.28/logs/routing/17-resizer_design.log)...
[ERROR]: during executing openroad script /openlane/scripts/openroad/resizer_routing_design.tcl
[ERROR]: Log: designs/SRAM_32_1024_2_row/runs/RUN_2023.09.11_15.29.28/logs/routing/17-resizer_design.log
[ERROR]: Last 10 lines:
---------------------------------------------------------------------------------------
Total           253846         25434           10.02%             0 /  0 /  0

[INFO GRT-0018] Total wirelength: 197084 um
[INFO GRT-0014] Routed nets: 659
[INFO]: Setting RC values...
[INFO RSZ-0058] Using max wire length 4508um.
[ERROR RSZ-0090] Max transition time from SDC is 0.040ns. Best achievable transition time is 0.043ns with a load of 0.01pF
Error: resizer_routing_design.tcl, 63 RSZ-0090
child process exited abnormally

[ERROR]: Creating issue reproducible...
[INFO]: Saving runtime environment...
OpenLane TCL Issue Packager

EFABLESS CORPORATION AND ALL AUTHORS OF THE OPENLANE PROJECT SHALL NOT BE HELD
LIABLE FOR ANY LEAKS THAT MAY OCCUR TO ANY PROPRIETARY DATA AS A RESULT OF USING
THIS SCRIPT. THIS SCRIPT IS PROVIDED ON AN "AS IS" BASIS, WITHOUT WARRANTIES OR
CONDITIONS OF ANY KIND.

BY USING THIS SCRIPT, YOU ACKNOWLEDGE THAT YOU FULLY UNDERSTAND THIS DISCLAIMER
AND ALL IT ENTAILS.

Parsing config file(s)…
Setting up /openlane/designs/SRAM_32_1024_2_row/runs/RUN_2023.09.11_15.29.28/issue_reproducible…
[WRN] /openlane/designs/SRAM_32_1024_2_row/runs/RUN_2023.09.11_15.29.28/tmp/17-memory_generator_sky130.sdc was n
```

vijayank88, 2023-09-12:
@openroadie 
Can you please look into this error message?

openroadie, 2023-09-13:
This is a correct error message to prevent the resizer/optimizer from inserting buffers in perpetuity to meet the max transition time requirement. The way to avoid this issue is to set the max transition time to be a more reasonable value.

NSampathIIITB, 2023-09-13:
[image]
 
How to rectify this error

openroadie, 2023-09-13:
https://github.com/The-OpenROAD-Project/OpenLane/issues/1982#issuecomment-1716951805

"The way to avoid this issue is to set the max transition time to be a more reasonable value".

NSampathIIITB, 2023-09-13:
> [#1982 (comment)](https://github.com/The-OpenROAD-Project/OpenLane/issues/1982#issuecomment-1716951805)
> 
> "The way to avoid this issue is to set the max transition time to be a more reasonable value".

where should i change it

openroadie, 2023-09-13:
Looks like it is coming from the .lib file(s). Look for "max_transition". You may want to figure out how these files were generated and why this transition time was consider appropriate before making modifications.

vijayank88, 2023-09-13:
Check sram .lib max_transition value

Baungarten-CINVESTAV, 2023-09-13:
I compared the constraints of the previous version and the newest version and I noticed that the newest version didn't add automatically the constraint "MAX_TRANSITION_CONSTRAINT", the previous version added "MAX_TRANSITION_CONSTRAINT" with a value of 0.75 so I add that to my constraints file (JSON):

"MAX_TRANSITION_CONSTRAINT": 0.75

But the result was the same:
[image]

I attached the .lib file of the SRAM and checked for the "max_transition" There are 5 match that takes values from 0.5 to 0.04. Honestly, I don't know what those values mean, and I don't know if I can modify them.

[sky130_sram_2kbyte_1rw1r_32x512_8_TT_1p8V_25C.zip](https://github.com/The-OpenROAD-Project/OpenLane/files/12600203/sky130_sram_2kbyte_1rw1r_32x512_8_TT_1p8V_25C.zip)

Dolu1990, 2023-10-27:
I just hit the same issue

@Baungarten-CINVESTAV 
I will try after doing a  
```
cd sram
sed -i 's/max_transition       : 0.04/max_transition       : 0.75/g' */*.lib
```

Baungarten-CINVESTAV, 2023-10-27:
If I remember correctly I solved the error by removing the "EXTRA_LIBS", this is used for time analysis and the OpenLane documentation indicates that it is optional.

kareefardi, 2023-12-21:
@NSampathIIITB SDC constraints should overwrite the constraints set by modules in `EXTRA_LIBS`. The problem here is that `CLOCK_PORT` is not defined correctly. If you take a look at the log file of the failing step, you will notice the following: 
```
[WARNING STA-0337] port 'clk' not found.
[INFO]: Setting output delay to: 2.0
[INFO]: Setting input delay to: 2.0
[INFO]: Setting max fanout to: 10
[WARNING STA-0337] port 'clk' not found.
[INFO]: Setting load to: 0.033442
[INFO]: Setting clock uncertainty to: 0.25
[INFO]: Setting clock transition to: 0.15
[WARNING STA-0559] transition time can not be specified for virtual clocks.
```
Hence the transition time set in the constraints wasn't applied. If you change `CLOCK_PORT` to `clk0`, you will not face the error above.