The-OpenROAD-Project/OpenLane #1983: ERROR: Unimplemented compiler directive

akhilesh911, 2023-09-13:
### Description

Hello,
While running synthesis on my design, there is an error in synthesis stage.
ERROR: Unimplemented compiler directive or undefined macro `suppress_faults.
The reproducible package is attached. Kindly guide.

### Expected Behavior

Synthesis should be completed.

### Environment report

```shell
Kernel: Linux v5.4.0-150-generic
Distribution: ubuntu 18.04
Python: v3.9.16 (OK)
Container Engine: docker v20.10.24 (OK)
OpenLane Git Version: 09cfff0dbb82d67405cddb65ec2089834e97198d
NOT FOUND: Please install pip using your operating system's package manager.
---
PDK Version Verification Status: MISMATCH
The version of open_pdks used in building the PDK does not match the version OpenLane was tested on (installed: 0059588eebfc704681dc2368bd1d33d96281d10f, tested: 35c7265f51749ad8d9fdbb575af22c7c8fab974e)
This may introduce some issues. You may want to re-install the PDK by invoking `make pdk`.
---
Git Log (Last 3 Commits)

09cfff0d 2023-07-06T21:10:16+03:00 Fix lint errors in BM64 verilog sources (#1878) - Kareem Farid -  (HEAD -> master, tag: 2023.07.07, origin/master, origin/HEAD)
5998726e 2023-07-06T18:25:24+03:00 Always raise an exception when a design fails in the CI (#1879) - Kareem Farid -  ()
a0c0945d 2023-07-06T15:17:37+03:00 Make synth checks more aggressive (#1880) - Kareem Farid -  ()
---
Git Remotes

origin	https://github.com/The-OpenROAD-Project/OpenLane.git (fetch)
origin	https://github.com/The-OpenROAD-Project/OpenLane.git (push)
```

### Reproduction material

[issue_reproducible.zip](https://github.com/The-OpenROAD-Project/OpenLane/files/12596065/issue_reproducible.zip)

### Relevant log output

```shell
[STEP 1]
[INFO]: Running Synthesis (log: designs/top/runs/RUN_2023.09.13_09.06.30/logs/synthesis/1-synthesis.log)...
[ERROR]: during executing yosys script /openlane/scripts/yosys/synth.tcl
[ERROR]: Log: designs/top/runs/RUN_2023.09.13_09.06.30/logs/synthesis/1-synthesis.log
[ERROR]: Last 10 lines:
Parsing SystemVerilog input from `/openlane/designs/top/../picosoc/src/picosoc.v' to AST representation.
Replacing existing blackbox module `\picosoc' at /openlane/designs/top/../picosoc/src/picosoc.v:36.1-348.10.
Generating RTLIL representation for module `\picosoc'.
Generating RTLIL representation for module `\picosoc_regs'.
Generating RTLIL representation for module `\picosoc_mem'.
Successfully finished Verilog frontend.

3. Executing Verilog-2005 frontend: /openlane/designs/top/../../pdks/gf180mcuC/libs.ref/gf180mcu_fd_io/verilog/gf180mcu_fd_io.v
/openlane/designs/top/../../pdks/gf180mcuC/libs.ref/gf180mcu_fd_io/verilog/gf180mcu_fd_io.v:17: ERROR: Unimplemented compiler directive or undefined macro `suppress_faults.
child process exited abnormally
```

kareefardi, 2023-09-13:
The reproducible isn't working for some reason, looking into it. Meanwhile, can you upload your source files and config?

donn, 2023-09-13:
``` `suppress_faults ``` is a non-standard Verilog compiler directive, and will likely never be supported by our synthesis tool, Yosys.

You more likely than not have to just remove it from your design.

kareefardi, 2023-09-13:
The source of the issue is including `gf180mcu_fd_io.v` in the design input verilog sources files. By uploading the source files, I wanted to know whether this was absolutely needed and offer a workaround.

donn, 2024-09-12:
Staled out.

closed 2024-09-12 by donn