The-OpenROAD-Project/OpenLane #1994: ERROR: Re-definition of module `\para_ser_01'!

yslim2002, 2023-09-18:
### Description

env: PDK=sky130A
OpenLane 2023.07.25_2_g01e67230-conda
All rights reserved. (c) 2020-2022 Efabless Corporation and contributors.
Available under the Apache License, version 2.0. See the LICENSE file for more details.

[INFO]: Using configuration in 'config.json'...
[INFO]: PDK Root: /home/yslim/Seacas_Improve/conda-env/share/pdk
[INFO]: Process Design Kit: sky130A
[INFO]: Standard Cell Library: sky130_fd_sc_hd
[INFO]: Optimization Standard Cell Library: sky130_fd_sc_hd
[INFO]: Run Directory: /home/yslim/Seacas_Improve/runs/RUN_2023.09.19_02.19.39
[INFO]: Saving runtime environment...
[INFO]: Preparing LEF files for the nom corner...
[INFO]: Preparing LEF files for the min corner...
[INFO]: Preparing LEF files for the max corner...
[INFO]: Running linter (Verilator) (log: runs/RUN_2023.09.19_02.19.39/logs/synthesis/linter.log)...
[INFO]: 0 errors found by linter
[INFO]: 0 warnings found by linter
[STEP 1]
[INFO]: Running Synthesis (log: runs/RUN_2023.09.19_02.19.39/logs/synthesis/1-synthesis.log)...
[ERROR]: during executing yosys script /home/yslim/Seacas_Improve/conda-env/share/openlane/scripts/yosys/synth.tcl
[ERROR]: Log: runs/RUN_2023.09.19_02.19.39/logs/synthesis/1-synthesis.log
[ERROR]: Last 10 lines:
Generating RTLIL representation for module `\multip_016'.
Note: Assuming pure combinatorial block at multip_016.v:56.3-68.6 in
compliance with IEC 62142(E):2005 / IEEE Std. 1364.1(E):2002. Recommending
use of @* instead of @(...) for better match of synthesis and simulation.
Generating RTLIL representation for module `\fft'.
Successfully finished Verilog frontend.

3. Executing Verilog-2005 frontend: /home/yslim/Seacas_Improve/para_ser_01.v
/home/yslim/Seacas_Improve/para_ser_01.v:51: ERROR: Re-definition of module `\para_ser_01'!
child process exited abnormally

[ERROR]: Creating issue reproducible...
[INFO]: Saving runtime environment...
OpenLane TCL Issue Packager

EFABLESS CORPORATION AND ALL AUTHORS OF THE OPENLANE PROJECT SHALL NOT BE HELD
LIABLE FOR ANY LEAKS THAT MAY OCCUR TO ANY PROPRIETARY DATA AS A RESULT OF USING
THIS SCRIPT. THIS SCRIPT IS PROVIDED ON AN "AS IS" BASIS, WITHOUT WARRANTIES OR
CONDITIONS OF ANY KIND.

BY USING THIS SCRIPT, YOU ACKNOWLEDGE THAT YOU FULLY UNDERSTAND THIS DISCLAIMER
AND ALL IT ENTAILS.

Parsing config file(s)…
Setting up /home/yslim/Seacas_Improve/runs/RUN_2023.09.19_02.19.39/issue_reproducible…
Done.
[INFO]: Reproducible packaged: Please tarball and upload 'runs/RUN_2023.09.19_02.19.39/issue_reproducible' if you're going to submit an issue.
[ERROR]: Step 1 (synthesis) failed with error:
-code 1 -level 0 -errorstack {INNER {invokeStk1 throw_error} CALL {run_tcl_script -tool yosys -no_consume /home/yslim/Seacas_Improve/conda-env/share/openlane/scripts/yosys/synth.tcl -indexed_log /home/yslim/Seacas_Improve/runs/RUN_2023.09.19_02.19.39/logs/synthesis/1-synthesis.log} CALL {run_yosys_script /home/yslim/Seacas_Improve/conda-env/share/openlane/scripts/yosys/synth.tcl -indexed_log /home/yslim/Seacas_Improve/runs/RUN_2023.09.19_02.19.39/logs/synthesis/1-synthesis.log} CALL {run_yosys -indexed_log /home/yslim/Seacas_Improve/runs/RUN_2023.09.19_02.19.39/logs/synthesis/1-synthesis.log} CALL run_synthesis CALL {run_non_interactive_mode -design .}} -errorcode NONE -errorinfo {
    while executing
"throw_error"
    (procedure "run_tcl_script" line 219)
    invoked from within
"run_tcl_script -tool yosys -no_consume {*}$args"
    (procedure "run_yosys_script" line 2)
    invoked from within
"run_yosys_script $::env(SYNTH_SCRIPT) -indexed_log $arg_values(-indexed_log)"
    (procedure "run_yosys" line 44)
    invoked from within
"run_yosys -indexed_log $log"
    (procedure "run_synthesis" line 13)
    invoked from within
"run_synthesis"} -errorline 1
[INFO]: Saving current set of views in 'runs/RUN_2023.09.19_02.19.39/results/final'...
[INFO]: Generating final set of reports...
[INFO]: Created manufacturability report at 'runs/RUN_2023.09.19_02.19.39/reports/manufacturability.rpt'.
[INFO]: Created metrics report at 'runs/RUN_2023.09.19_02.19.39/reports/metrics.csv'.
[INFO]: Saving runtime environment...
[ERROR]: Flow failed.

### Expected Behavior

I would like to pass the Synthesis process. However, the error keeps poping up. May I know what shall I do? I have tried to change the name of the module.

### Environment report

```shell
python3: can't open file '/mnt/c/Users/limyi/./env.py': [Errno 2] No such file or directory
```

### Reproduction material

https://github.com/isha1210/SoC-Physical-design-using-Sky130

### Relevant log output

```shell
env: PDK=sky130A
OpenLane 2023.07.25_2_g01e67230-conda
All rights reserved. (c) 2020-2022 Efabless Corporation and contributors.
Available under the Apache License, version 2.0. See the LICENSE file for more details.

[INFO]: Using configuration in 'config.json'...
[INFO]: PDK Root: /home/yslim/Seacas_Improve/conda-env/share/pdk
[INFO]: Process Design Kit: sky130A
[INFO]: Standard Cell Library: sky130_fd_sc_hd
[INFO]: Optimization Standard Cell Library: sky130_fd_sc_hd
[INFO]: Run Directory: /home/yslim/Seacas_Improve/runs/RUN_2023.09.19_02.19.39
[INFO]: Saving runtime environment...
[INFO]: Preparing LEF files for the nom corner...
[INFO]: Preparing LEF files for the min corner...
[INFO]: Preparing LEF files for the max corner...
[INFO]: Running linter (Verilator) (log: runs/RUN_2023.09.19_02.19.39/logs/synthesis/linter.log)...
[INFO]: 0 errors found by linter
[INFO]: 0 warnings found by linter
[STEP 1]
[INFO]: Running Synthesis (log: runs/RUN_2023.09.19_02.19.39/logs/synthesis/1-synthesis.log)...
[ERROR]: during executing yosys script /home/yslim/Seacas_Improve/conda-env/share/openlane/scripts/yosys/synth.tcl
[ERROR]: Log: runs/RUN_2023.09.19_02.19.39/logs/synthesis/1-synthesis.log
[ERROR]: Last 10 lines:
Generating RTLIL representation for module `\multip_016'.
Note: Assuming pure combinatorial block at multip_016.v:56.3-68.6 in
compliance with IEC 62142(E):2005 / IEEE Std. 1364.1(E):2002. Recommending
use of @* instead of @(...) for better match of synthesis and simulation.
Generating RTLIL representation for module `\fft'.
Successfully finished Verilog frontend.

3. Executing Verilog-2005 frontend: /home/yslim/Seacas_Improve/para_ser_01.v
/home/yslim/Seacas_Improve/para_ser_01.v:51: ERROR: Re-definition of module `\para_ser_01'!
child process exited abnormally

[ERROR]: Creating issue reproducible...
[INFO]: Saving runtime environment...
OpenLane TCL Issue Packager

EFABLESS CORPORATION AND ALL AUTHORS OF THE OPENLANE PROJECT SHALL NOT BE HELD
LIABLE FOR ANY LEAKS THAT MAY OCCUR TO ANY PROPRIETARY DATA AS A RESULT OF USING
THIS SCRIPT. THIS SCRIPT IS PROVIDED ON AN "AS IS" BASIS, WITHOUT WARRANTIES OR
CONDITIONS OF ANY KIND.

BY USING THIS SCRIPT, YOU ACKNOWLEDGE THAT YOU FULLY UNDERSTAND THIS DISCLAIMER
AND ALL IT ENTAILS.

Parsing config file(s)…
Setting up /home/yslim/Seacas_Improve/runs/RUN_2023.09.19_02.19.39/issue_reproducible…
Done.
[INFO]: Reproducible packaged: Please tarball and upload 'runs/RUN_2023.09.19_02.19.39/issue_reproducible' if you're going to submit an issue.
[ERROR]: Step 1 (synthesis) failed with error:
-code 1 -level 0 -errorstack {INNER {invokeStk1 throw_error} CALL {run_tcl_script -tool yosys -no_consume /home/yslim/Seacas_Improve/conda-env/share/openlane/scripts/yosys/synth.tcl -indexed_log /home/yslim/Seacas_Improve/runs/RUN_2023.09.19_02.19.39/logs/synthesis/1-synthesis.log} CALL {run_yosys_script /home/yslim/Seacas_Improve/conda-env/share/openlane/scripts/yosys/synth.tcl -indexed_log /home/yslim/Seacas_Improve/runs/RUN_2023.09.19_02.19.39/logs/synthesis/1-synthesis.log} CALL {run_yosys -indexed_log /home/yslim/Seacas_Improve/runs/RUN_2023.09.19_02.19.39/logs/synthesis/1-synthesis.log} CALL run_synthesis CALL {run_non_interactive_mode -design .}} -errorcode NONE -errorinfo {
    while executing
"throw_error"
    (procedure "run_tcl_script" line 219)
    invoked from within
"run_tcl_script -tool yosys -no_consume {*}$args"
    (procedure "run_yosys_script" line 2)
    invoked from within
"run_yosys_script $::env(SYNTH_SCRIPT) -indexed_log $arg_values(-indexed_log)"
    (procedure "run_yosys" line 44)
    invoked from within
"run_yosys -indexed_log $log"
    (procedure "run_synthesis" line 13)
    invoked from within
"run_synthesis"} -errorline 1
[INFO]: Saving current set of views in 'runs/RUN_2023.09.19_02.19.39/results/final'...
[INFO]: Generating final set of reports...
[INFO]: Created manufacturability report at 'runs/RUN_2023.09.19_02.19.39/reports/manufacturability.rpt'.
[INFO]: Created metrics report at 'runs/RUN_2023.09.19_02.19.39/reports/metrics.csv'.
[INFO]: Saving runtime environment...
[ERROR]: Flow failed.
```

donn, 2023-09-18:
1. This guide is fairly out-of-date and does not have the "Seacas" design you're implementing. Not sure where you got it.
2. You appear to be using Conda. I'm not sure why.

We can't do much to help if you don't post the actual design that was synthesized and the configuration file.

yslim2002, 2023-09-19:
[](url)[](url)

yslim2002, 2023-09-19:
[](url)
[SEACAS_2023.zip](https://github.com/The-OpenROAD-Project/OpenLane/files/12654568/SEACAS_2023.zip)
here is the google colab notebook that I have exported

yslim2002, 2023-09-19:
> 1. This guide is fairly out-of-date and does not have the "Seacas" design you're implementing. Not sure where you got it.
> 2. You appear to be using Conda. I'm not sure why.
> 3. We can't do much to help if you don't post the actual design that was synthesized and the configuration file.
> 
> Could you kindly post the design?

kareefardi, 2023-09-24:
The google colab in the zip file clones this repo https://github.com/VenciFreeman/FFT_ChipDesign. I am assuming this contains the source verilog files. In the section of the log file that you posted in the first message, there are references to some verilog files `multip_016.v` and `para_ser_01.v`:
```
Note: Assuming pure combinatorial block at multip_016.v:56.3-68.6 in

3. Executing Verilog-2005 frontend: /home/yslim/Seacas_Improve/para_ser_01.v
```
These files are not found inside `https://github.com/VenciFreeman/FFT_ChipDesign`. Can you again upload the design files and openlane configuration files that you used? (No need to upload the whole collab file)

donn, 2024-09-12:
Staled out.

closed 2024-09-12 by donn