The-OpenROAD-Project/OpenLane #1996: Error: Flow failed. Multiplexers

DmitryZlobec, 2023-09-19:
### Description

I have got an error, when try to syntheze some examples of multiplexers in verilog.
```
[ERROR]: Step 1 (synthesis) failed with error:
-code 1 -level 0 -errorcode NONE -errorinfo {
    while executing
"throw_error"
    (procedure "run_tcl_script" line 219)
    invoked from within
"run_tcl_script -tool yosys -no_consume {*}$args"
    (procedure "run_yosys_script" line 2)
    invoked from within
"run_yosys_script $::env(SYNTH_SCRIPT) -indexed_log $arg_values(-indexed_log)"
    (procedure "run_yosys" line 44)
    invoked from within
"run_yosys -indexed_log $log"
    (procedure "run_synthesis" line 13)
    invoked from within
"run_synthesis"} -errorline 1

```

### Expected Behavior

Synthezis should be ok.

### Environment report

```shell
WARNING: issue-survey appears to be running inside the OpenLane
container.

This makes it difficult to rule out issues with your
environment.

Unless instructed specifically to do so, please run this command
outside the OpenLane container.
---

Kernel: Linux v5.15.0-83-generic
Distribution: centos 7
Python: v3.6.8 (OK)
OpenLane Git Version: 925d631c3e0328841536667c7f06b69040238e98
pip: INSTALLED
python-venv: INSTALLED
---
PDK Version Verification Status: OK
---
Git Log (Last 3 Commits)

925d631c 2023-08-03T13:30:36+00:00 OpenROAD python `read_def` api updates (#1926) - Kareem Farid -  (HEAD -> master, tag: 2023.08.04, origin/master, origin/HEAD)
4c74d8b1 2023-07-31T17:13:43+00:00 Rename `mca` folders (#1922) - Kareem Farid -  (tag: 2023.08.01)
01e67230 2023-07-25T13:53:21+03:00 [BOT] Update magic (#1914) - Openlane Bot -  (tag: 2023.07.26)
---
Git Remotes

origin	https://github.com/The-OpenROAD-Project/OpenLane (fetch)
origin	https://github.com/The-OpenROAD-Project/OpenLane (push)
```

### Reproduction material

](url)[
[issue_reproducible.tar.gz](https://github.com/The-OpenROAD-Project/OpenLane/files/12656223/issue_reproducible.tar.gz)

### Relevant log output

```shell
cd /home/zlobec/OpenLane && \
	docker run --rm -v /home/zlobec/OpenLane:/openlane -v /home/zlobec/OpenLane/designs:/openlane/install -v /home/zlobec:/home/zlobec -v /home/zlobec/.volare:/home/zlobec/.volare -e PDK_ROOT=/home/zlobec/.volare -e PDK=sky130A  --user 1000:1000 -e DISPLAY=:0 -v /tmp/.X11-unix:/tmp/.X11-unix -v /home/zlobec/.Xauthority:/.Xauthority --network host --security-opt seccomp=unconfined efabless/openlane:925d631c3e0328841536667c7f06b69040238e98-amd64 sh -c "./flow.tcl -design 02_mux"
OpenLane 925d631c3e0328841536667c7f06b69040238e98
All rights reserved. (c) 2020-2022 Efabless Corporation and contributors.
Available under the Apache License, version 2.0. See the LICENSE file for more details.

[INFO]: Using configuration in 'designs/02_mux/config.tcl'...
[INFO]: PDK Root: /home/zlobec/.volare
[INFO]: Process Design Kit: sky130A
[INFO]: Standard Cell Library: sky130_fd_sc_hd
[INFO]: Optimization Standard Cell Library: sky130_fd_sc_hd
[WARNING]: SYNTH_MAX_FANOUT is now deprecated; use MAX_FANOUT_CONSTRAINT instead.
[INFO]: Run Directory: /openlane/designs/02_mux/runs/RUN_2023.09.19_05.37.14
[INFO]: Saving runtime environment...
[INFO]: Preparing LEF files for the nom corner...
[INFO]: Preparing LEF files for the min corner...
[INFO]: Preparing LEF files for the max corner...
[INFO]: Running linter (Verilator) (log: designs/02_mux/runs/RUN_2023.09.19_05.37.14/logs/synthesis/linter.log)...
[INFO]: 0 errors found by linter
[WARNING]: 20 warnings found by linter
[STEP 1]
[INFO]: Running Synthesis (log: designs/02_mux/runs/RUN_2023.09.19_05.37.14/logs/synthesis/1-synthesis.log)...
[ERROR]: during executing yosys script /openlane/scripts/yosys/synth.tcl
[ERROR]: Log: designs/02_mux/runs/RUN_2023.09.19_05.37.14/logs/synthesis/1-synthesis.log
[ERROR]: Last 10 lines:
Successfully finished Verilog frontend.

4. Executing Verilog-2005 frontend: /openlane/designs/02_mux/src/slow_clk_gen.sv
Parsing SystemVerilog input from `/openlane/designs/02_mux/src/slow_clk_gen.sv' to AST representation.
Generating RTLIL representation for module `\slow_clk_gen'.
Successfully finished Verilog frontend.

5. Executing Verilog-2005 frontend: /openlane/designs/02_mux/src/top.sv
/openlane/designs/02_mux/src/top.sv:146: ERROR: syntax error, unexpected '[', expecting TOK_ID or '#'
child process exited abnormally

[ERROR]: Creating issue reproducible...
[INFO]: Saving runtime environment...
OpenLane TCL Issue Packager

EFABLESS CORPORATION AND ALL AUTHORS OF THE OPENLANE PROJECT SHALL NOT BE HELD
LIABLE FOR ANY LEAKS THAT MAY OCCUR TO ANY PROPRIETARY DATA AS A RESULT OF USING
THIS SCRIPT. THIS SCRIPT IS PROVIDED ON AN "AS IS" BASIS, WITHOUT WARRANTIES OR
CONDITIONS OF ANY KIND.

BY USING THIS SCRIPT, YOU ACKNOWLEDGE THAT YOU FULLY UNDERSTAND THIS DISCLAIMER
AND ALL IT ENTAILS.

Parsing config file(s)…
Setting up /openlane/designs/02_mux/runs/RUN_2023.09.19_05.37.14/issue_reproducible…
Done.
[INFO]: Reproducible packaged: Please tarball and upload 'designs/02_mux/runs/RUN_2023.09.19_05.37.14/issue_reproducible' if you're going to submit an issue.
[ERROR]: Step 1 (synthesis) failed with error:
-code 1 -level 0 -errorcode NONE -errorinfo {
    while executing
"throw_error"
    (procedure "run_tcl_script" line 219)
    invoked from within
"run_tcl_script -tool yosys -no_consume {*}$args"
    (procedure "run_yosys_script" line 2)
    invoked from within
"run_yosys_script $::env(SYNTH_SCRIPT) -indexed_log $arg_values(-indexed_log)"
    (procedure "run_yosys" line 44)
    invoked from within
"run_yosys -indexed_log $log"
    (procedure "run_synthesis" line 13)
    invoked from within
"run_synthesis"} -errorline 1
[INFO]: Saving current set of views in 'designs/02_mux/runs/RUN_2023.09.19_05.37.14/results/final'...
[INFO]: Generating final set of reports...
[INFO]: Created manufacturability report at 'designs/02_mux/runs/RUN_2023.09.19_05.37.14/reports/manufacturability.rpt'.
[INFO]: Created metrics report at 'designs/02_mux/runs/RUN_2023.09.19_05.37.14/reports/metrics.csv'.
[INFO]: Saving runtime environment...
[ERROR]: Flow failed.
[INFO]: The failure may have been because of the following warnings:
[WARNING]: 20 warnings found by linter
```

vijayank88, 2023-09-19:
There was some limitation to use sv. Check this GH https://github.com/The-OpenROAD-Project/OpenLane/issues/492

DmitryZlobec, 2023-09-19:
Does https://github.com/chipsalliance/synlig plugin can help for my case?

DmitryZlobec, 2023-09-19:
> There was some limitation to use sv. Check this GH #492

Does https://github.com/chipsalliance/synlig plugin can help for my case?

vijayank88, 2023-09-20:
https://github.com/zachjs/sv2v try this repo for SV to verilog conversion. Verify the code before proceeding to flow.

closed 2023-09-20 by DmitryZlobec