The-OpenROAD-Project/OpenLane #2003: Incorrect number of repair_antennas iterations - remove repair_antennas call from grt.tcl

jacobhaehn, 2023-09-22:
### Description

I have been trying to remedy antenna issues in a large design with many macros. I have been having issues similar to #1970, where the process gets killed after running out of memory. I can normally get through about 7-9 total repair_antenna iterations before the crash (looking at global.log), reducing antenna violations from 1431 to about 30, but always gets stuck on the last few (or maybe it's keeping previous iterations in memory and eventually runs out?). 

I have been trying to avoid a crash by reducing GRT_ANT_ITERS. However, since the repair_antenna flow was reworked in #1968, repair_antennas.tcl calls repair_antennas 3 times, each with GRT_ANT_ITERS: once in grt.tcl, once within itself, and then again in grt.tcl. For example, if I have GRT_ANT_ITERS=3, then it actually ends up running 9 total iterations (3, 3, 2) and the process is killed because I always run out of memory around the 9th total iteration. If I have GRT_ANT_ITERS=5, it still runs 9 total iterations, but 5 iterations of the first pass, then 3 on the second pass before crashing on the 9th total iteration. 

I suggest removing the repair_antennas call from grt.tcl so that repair_antennas is only called once from repair_antennas.tcl itself, making the true number of antenna repair iterations equal to GRT_ANT_ITERS. Additionally, let me know if you have any suggestions or fixes to prevent the crash in the first place.

### Expected Behavior

GRT_ANT_ITERS should be the true total number of repair_antenna iterations.

### Environment report

```shell
Kernel: Linux v6.2.0-26-generic
Distribution: ubuntu 22.04
Python: v3.10.12 (OK)
Container Engine: docker v24.0.4 (OK)
OpenLane Git Version: 5215ea7cc01e5311086e995905492f0e8fe1ceb6
pip: INSTALLED
python-venv: INSTALLED
---
PDK Version Verification Status: OK
---
Git Log (Last 3 Commits)

5215ea7c 2023-09-20T15:53:26+03:00 Update `open_pdks` (#1992) - Mohamed Gaber -  (HEAD -> master, origin/master, origin/HEAD)
7e5a2e9f 2023-09-10T19:06:55+03:00 fix synthesis `chparam` issue (#1981) - Kareem Farid -  (tag: 2023.09.11)
0e8827eb 2023-09-10T19:04:45+03:00 `run_sta`: fix `-blackbox_check` not running when `-multi_corner` is not set (#1980) - Kareem Farid -  ()
---
Git Remotes

origin	https://github.com/The-OpenROAD-Project/OpenLane (fetch)
origin	https://github.com/The-OpenROAD-Project/OpenLane (push)
```

### Reproduction material

[issue_reproducible.zip](https://github.com/The-OpenROAD-Project/OpenLane/files/12701696/issue_reproducible.zip)

### Relevant log output

```shell
21-global.log for GRT_ANT_ITERS=3: (**NOTE: I have removed warnings about nets requiring more than 10 diodes per gate for clarity**)

OpenROAD 0a6d0fd469bc674417036342994520ee2e0a2727 
This program is licensed under the BSD-3 license. See the LICENSE file for details.
Components of this program may be licensed under more restrictive licenses which must be honored.
[INFO]: Reading ODB at '/home/george/Documents/batfive/designs/batfive/openlane_src/runs/RUN_2023.09.21_21.51.24/tmp/routing/./19-diodes_legalized.odb'…
define_corners Typical
read_liberty -corner Typical /home/george/.volare/sky130A/libs.ref/sky130_fd_sc_hd/lib/sky130_fd_sc_hd__tt_025C_1v80.lib
Using 1e-12 for capacitance...
Using 1e+03 for resistance...
Using 1e-09 for time...
Using 1e+00 for voltage...
Using 1e-03 for current...
Using 1e-09 for power...
Using 1e-06 for distance...
Reading design constraints file at '/openlane/scripts/base.sdc'…
[INFO]: Setting output delay to: 15.0
[INFO]: Setting input delay to: 15.0
[INFO]: Setting load to: 0.033442
[INFO]: Setting clock uncertainty to: 0.25
[INFO]: Setting clock transition to: 0.15
[INFO]: Setting timing derate to: 5.0 %
[INFO]: Setting signal min routing layer to: met1 and clock min routing layer to met1. 
[INFO]: Setting signal max routing layer to: met5 and clock max routing layer to met5. 
-congestion_iterations 50 -verbose -congestion_report_file /home/george/Documents/batfive/designs/batfive/openlane_src/runs/RUN_2023.09.21_21.51.24/tmp/routing/groute-congestion.rpt
[INFO GRT-0020] Min routing layer: met1
[... 160 lines cut ...]

[INFO GRT-0018] Total wirelength: 48008840 um
[INFO GRT-0014] Routed nets: 315818
[INFO GRT-0006] Repairing antennas, iteration 1.
[INFO GRT-0043] No OR_DEFAULT vias defined.
[INFO GRT-0012] Found 417 antenna violations.
[INFO GRT-0015] Inserted 3760 diodes.
[INFO GRT-0054] Using detailed placer to place 1230 diodes.
[INFO GRT-0009] rerouting 218481 nets.
[INFO GRT-0001] Minimum degree: 2
[INFO GRT-0002] Maximum degree: 105
[INFO GRT-0006] Repairing antennas, iteration 2.
[INFO GRT-0043] No OR_DEFAULT vias defined.
[INFO GRT-0012] Found 221 antenna violations.
[INFO GRT-0015] Inserted 1429 diodes.
[INFO GRT-0054] Using detailed placer to place 401 diodes.
[INFO GRT-0009] rerouting 2023 nets.
[INFO GRT-0001] Minimum degree: 2
[INFO GRT-0002] Maximum degree: 105
[INFO GRT-0006] Repairing antennas, iteration 3.
[INFO GRT-0043] No OR_DEFAULT vias defined.
child killed: kill signal

[ERROR]: Creating issue reproducible...
```

vijayank88, 2023-09-23:
@jacobhaehn 
Is the attached test case produced by the flow?
Test case contains only the scripts.

kareefardi, 2023-09-24:
@jacobhaehn `repair_antennas` in `grt.tcl` should have been removed and will be fixed. As for the crash itself the reproducible is missing source files as highlighted by @vijayank88.

jacobhaehn, 2023-09-25:
> @jacobhaehn Is the attached test case produced by the flow? Test case contains only the scripts.

Thank you for looking into this! Yes this is the test case automatically produced by the flow. I was also a little confused that it didn't contain source files. I was running in interactive mode (skipped synthesis and loaded synthesized netlist from previous run), so not sure if that's a bug with interactive mode not including source files, or just how interactive mode is supposed to work. 

I'll go ahead and run the entire flow and attach the test case with the correct source files when it completes in a few hours.

jacobhaehn, 2023-09-26:
The design is quite large, so I had to split into a few tarballs. For reference, the files can be extracted with the following command:

`cat routing_crash_issue_reproducible_updated.* | tar -xzvf -`

I tried this out and it worked for me, but if it doesn't work, let me know and I can try to find a solution.

[routing_crash_issue_reproducible_updated.partaa.tar.gz](https://github.com/The-OpenROAD-Project/OpenLane/files/12729517/routing_crash_issue_reproducible_updated.partaa.tar.gz)
[routing_crash_issue_reproducible_updated.partab.tar.gz](https://github.com/The-OpenROAD-Project/OpenLane/files/12729519/routing_crash_issue_reproducible_updated.partab.tar.gz)
[routing_crash_issue_reproducible_updated.partac.tar.gz](https://github.com/The-OpenROAD-Project/OpenLane/files/12729520/routing_crash_issue_reproducible_updated.partac.tar.gz)
[routing_crash_issue_reproducible_updated.partad.tar.gz](https://github.com/The-OpenROAD-Project/OpenLane/files/12729513/routing_crash_issue_reproducible_updated.partad.tar.gz)
[routing_crash_issue_reproducible_updated.partae.tar.gz](https://github.com/The-OpenROAD-Project/OpenLane/files/12729514/routing_crash_issue_reproducible_updated.partae.tar.gz)
[routing_crash_issue_reproducible_updated.partaf.tar.gz](https://github.com/The-OpenROAD-Project/OpenLane/files/12729515/routing_crash_issue_reproducible_updated.partaf.tar.gz)

I am running on a device with 32 GB RAM, but today I will be testing with 128 GB RAM.

closed 2023-10-01 by donn

kareefardi, 2023-10-01:
@jacobhaehn After manually applying #2004 in the issue reproducible, I didn't get a crash. Peak memory consumption was 12 GB in a 64 GB machine.

jacobhaehn, 2023-10-02:
Thank you for resolving this! 

For anyone also struggling with crashes, with this fix, I found I still got crashes with 32 GB RAM when increasing GRT_ANT_ITERS to 15. Increasing RAM solved all of these issues, as detailed routing peaked at 30 GB memory consumption on my machine, so I just needed a few more GB than available to the process.

PR #2004 (2023-10-01): Fix error with repair antennas rework 
- remove repair_antennas from common/grt.tcl
~ properly execute GRT_REPAIR_ANTENNAS flow

---
Fixes #2003