The-OpenROAD-Project/OpenLane #2007: run_designs is broken, not documented

mattvenn, 2023-09-28:
### Description

I'm unable to get run_designs to work. Either by calling direct or using the makefile's `test_design_list`

Also, there doesn't seem to be any documentation, and the built in help isn't very helpful. I don't know what these do:

* matrix
* configuration_parameters
* append_configurations

I want to be able to use run_designs to iterate through a set of parameters. How do I do that?

### Expected Behavior

not to immediately crash

### Environment report

```shell
:python3 ./env.py issue-survey
Kernel: Linux v6.2.0-33-generic
Distribution: ubuntu 22.04
Python: v3.10.12 (OK)
Container Engine: docker v24.0.5 (OK)
OpenLane Git Version: d054702b2cce04761cc2bc598f6b95c9d8ca7c6c
pip: INSTALLED
python-venv: INSTALLED
---
PDK Version Verification Status: OK
---
Git Log (Last 3 Commits)

d054702 2023-07-19T16:09:15+03:00 remove `unset_propagated_clock` (#1908) - passant5 -  (grafted, HEAD, tag: 2023.07.19)
---
Git Remotes

origin	https://github.com/The-OpenROAD-Project/OpenLane (fetch)
origin	https://github.com/The-OpenROAD-Project/OpenLane (push)
```

### Reproduction material

you can just run ./run_designs.py with no arguments and it will fail in the same way.

### Relevant log output

```shell
:make test_design_list
cd /home/zerotoasic/asic_tools/openlane && \
	docker run --rm -v /home/zerotoasic/asic_tools/openlane:/openlane -v /home/zerotoasic/asic_tools/openlane/designs:/openlane/install -v /home/zerotoasic:/home/zerotoasic -v /home/zerotoasic/asic_tools/pdk:/home/zerotoasic/asic_tools/pdk -e PDK_ROOT=/home/zerotoasic/asic_tools/pdk -e PDK=sky130A  --user 1000:1000 -e DISPLAY=:0 -v /tmp/.X11-unix:/tmp/.X11-unix -v /home/zerotoasic/.Xauthority:/.Xauthority --network host --security-opt seccomp=unconfined efabless/openlane:d054702b2cce04761cc2bc598f6b95c9d8ca7c6c-amd64 sh -c "\
		python3 run_designs.py\
		--tag custom_design_List\
		--threads 1\
		--print_rem 0\
		--benchmark regression_results/benchmark_results/SW_HD.csv\
		--config_file config\
		spm\
	"
Traceback (most recent call last):
  File "run_designs.py", line 506, in <module>
    cli()
  File "/usr/local/lib/python3.6/site-packages/click/core.py", line 1128, in __call__
    return self.main(*args, **kwargs)
  File "/usr/local/lib/python3.6/site-packages/click/core.py", line 1053, in main
    rv = self.invoke(ctx)
  File "/usr/local/lib/python3.6/site-packages/click/core.py", line 1395, in invoke
    return ctx.invoke(self.callback, **ctx.params)
  File "/usr/local/lib/python3.6/site-packages/click/core.py", line 754, in invoke
    return __callback(*args, **kwargs)
  File "run_designs.py", line 203, in cli
    handler1 = logging.FileHandler(f"{report_file_name}.log", "w")
  File "/usr/lib64/python3.6/logging/__init__.py", line 1032, in __init__
    StreamHandler.__init__(self, self._open())
  File "/usr/lib64/python3.6/logging/__init__.py", line 1061, in _open
    return open(self.baseFilename, self.mode, encoding=self.encoding)
FileNotFoundError: [Errno 2] No such file or directory: '/openlane/regression_results/custom_design_List_28_09_2023_08_52/custom_design_List_28_09_2023_08_52.log'
make: *** [Makefile:161: test_design_list] Error 1
```

donn, 2023-09-28:
https://openlane.readthedocs.io/en/latest/usage/exploration_script.html

With regards to the bug, I'll look at it when I'm back.

kareefardi, 2023-10-01:
@mattvenn I can't replicate this issue with `run_designs.py`. The benchmark file is wrong but I also tried that. Perhaps this (eaa6680c62a666cb3b8ef45369f920726efc502c) fixed your issue. Can you update and try again?

closed 2023-10-04 by mattvenn