The-OpenROAD-Project/OpenLane #2010: documentation for run designs

mattvenn, 2023-10-04:
https://github.com/The-OpenROAD-Project/OpenLane/blob/5771836bf47dc77996b35471f2bf40a6acdb5983/docs/source/usage/exploration_script.md?plain=1#L73

the command line parameters are unclear

there are a few examples, I guess that --matrix and a json file replaces the previous config.txt.

there is no matrix.json file example

kareefardi, 2023-10-04:
https://github.com/efabless/openlane-ci-designs/blob/d9ba6ea51cdd50589ad64842f85c5aa238cf003a/wbqspiflash/matrix.json

closed 2023-10-04 by donn

mattvenn, 2023-10-04:
for your reference, here are the options I can't work out what they do from the help message:

--configuration_parameters

--append_configurations

mattvenn, 2023-10-04:
If you can explain what they do I will make  PR on the docs

kareefardi, 2023-10-04:
@mattvenn Looking a the source code, `run_designs.py` generates a csv report file that contains a set of variables. `--configuration_parameters` points to a file that contains a comma separated list of variable names. Depending on `--append_configurations`, these variables either are added to or replace the set of variables in the csv report file. Note that this hasn't been tested in a while and might be broken. I will test them as soon as I get a chance.

mattvenn, 2023-10-04:
ok. I think the addition of the matrix file and a link to the online docs would be helpful