The-OpenROAD-Project/OpenLane #2012: Error at synthesis stage.

Pa1mantri, 2023-10-06:
### Description

Error at the synthesis stage. Error is shown in a code generated by sandpiper-saas tool which converts the TL-verilog into verilog.

### Expected Behavior

Successful completion of synthesis stage.

### Environment report

```shell
% python3 ./env.py issue-survey

WARNING: issue-survey appears to be running inside the OpenLane

container.

This makes it difficult to rule out issues with your

environment.

Unless instructed specifically to do so, please run this command

outside the OpenLane container.

---

Kernel: Linux v6.2.0-32-generic

Distribution: centos 7

Python: v3.6.8 (OK)

OpenLane Git Version: 00caae2b3186bb6b24a0b7db7ee1a6f056ad1702

pip: INSTALLED

python-venv: INSTALLED

---

PDK Version Verification Status: OK

---

Git Log (Last 3 Commits)

00caae2b 2023-08-22T19:35:09+03:00 Update PDK + Restore `RC_LAYER` variables (#1932) - Kareem Farid -  (HEAD -> master, tag: 2023.08.23, origin/master, origin/HEAD)

eaa6680c 2023-08-22T19:12:10+03:00 `run_designs.py`: Fix `mkdirp` not being called in some scenarios (#1945) - UMW -  ()

17559fa5 2023-08-22T18:54:25+03:00 CI Tweaks (#1951) - Mohamed Gaber -  ()

---

Git Remotes

origin	https://github.com/The-OpenROAD-Project/OpenLane (fetch)

origin	https://github.com/The-OpenROAD-Project/OpenLane (push)

%
```

### Reproduction material

[issue_synth.tar.gz](https://github.com/The-OpenROAD-Project/OpenLane/files/12830381/issue_synth.tar.gz)

### Relevant log output

```shell
/----------------------------------------------------------------------------\

 |                                                                            |

 |  yosys -- Yosys Open SYnthesis Suite                                       |

 |                                                                            |

 |  Copyright (C) 2012 - 2020  Claire Xenia Wolf <claire@yosyshq.com>         |

 |                                                                            |

 |  Permission to use, copy, modify, and/or distribute this software for any  |

 |  purpose with or without fee is hereby granted, provided that the above    |

 |  copyright notice and this permission notice appear in all copies.         |

 |                                                                            |

 |  THE SOFTWARE IS PROVIDED "AS IS" AND THE AUTHOR DISCLAIMS ALL WARRANTIES  |

 |  WITH REGARD TO THIS SOFTWARE INCLUDING ALL IMPLIED WARRANTIES OF          |

 |  MERCHANTABILITY AND FITNESS. IN NO EVENT SHALL THE AUTHOR BE LIABLE FOR   |

 |  ANY SPECIAL, DIRECT, INDIRECT, OR CONSEQUENTIAL DAMAGES OR ANY DAMAGES    |

 |  WHATSOEVER RESULTING FROM LOSS OF USE, DATA OR PROFITS, WHETHER IN AN     |

 |  ACTION OF CONTRACT, NEGLIGENCE OR OTHER TORTIOUS ACTION, ARISING OUT OF   |

 |  OR IN CONNECTION WITH THE USE OR PERFORMANCE OF THIS SOFTWARE.            |

 |                                                                            |

 \----------------------------------------------------------------------------/

 Yosys 0.30+48 (git sha1 14d50a176d5, gcc 8.3.1 -fPIC -Os)

[TCL: yosys -import] Command name collision: found pre-existing command `cd' -> skip.

[TCL: yosys -import] Command name collision: found pre-existing command `eval' -> skip.

[TCL: yosys -import] Command name collision: found pre-existing command `exec' -> skip.

[TCL: yosys -import] Command name collision: found pre-existing command `read' -> skip.

[TCL: yosys -import] Command name collision: found pre-existing command `trace' -> skip.

1. Executing Liberty frontend: /openlane/designs/vsdmemsoc/src/macros/sky130_sram_1kbyte_1rw1r_32x256_8_TT_1p8V_25C.lib

Imported 1 cell types from liberty file.

2. Executing Verilog-2005 frontend: /openlane/designs/vsdmemsoc/src/rvmyth.v

Parsing SystemVerilog input from `/openlane/designs/vsdmemsoc/src/rvmyth.v' to AST representation.

Generating RTLIL representation for module `\rvmyth'.

Warning: Replacing memory \CPU_Xreg_value_a5 with list of registers. See /openlane/designs/vsdmemsoc/src/rvmyth_gen.v:676

Warning: Replacing memory \CPU_Xreg_value_a4 with list of registers. See /openlane/designs/vsdmemsoc/src/rvmyth_gen.v:675

Warning: Replacing memory \CPU_Dmem_value_a5 with list of registers. See /openlane/designs/vsdmemsoc/src/rvmyth_gen.v:666

Successfully finished Verilog frontend.

3. Executing Verilog-2005 frontend: /openlane/designs/vsdmemsoc/src/rvmyth_gen.v

/openlane/designs/vsdmemsoc/src/rvmyth_gen.v:12: ERROR: syntax error, unexpected TOK_GENVAR
```

Pa1mantri, 2023-10-07:
Found the bug. Resolved.

closed 2023-10-07 by Pa1mantri

goatgate, 2024-12-12:
how was the error resolved?

Pa1mantri, 2024-12-12:
If i remember correctly, this error was due to sandpiper-saas tool; there should be an issue now.

goatgate, 2024-12-12:
faced similar issue still, can you please tell me what you did?