The-OpenROAD-Project/OpenLane #2015: error generating PDN for the design

Pa1mantri, 2023-10-11:
### Description

I have used FP_PDN_MACRO_HOOKS variable for connecting macros to the power grid. One of the macro instances (cntlr) is not being recognized. I found on the internet to use verilog_files_blackbox variable . If it is, can you help me what all the files that are needed to include in that variable. I have .lef,.gds,.lib files of memory(sram) but none of these files for the controller module. If i remove the cntlr instance completely it results in LVS errors at the end. 

### Expected Behavior

successful flow of the design without lvs errors. Thank you.

### Environment report

```shell
This makes it difficult to rule out issues with your
environment.

Unless instructed specifically to do so, please run this command
outside the OpenLane container.
---

Kernel: Linux v6.2.0-32-generic
Distribution: centos 7
Python: v3.6.8 (OK)
OpenLane Git Version: 00caae2b3186bb6b24a0b7db7ee1a6f056ad1702
pip: INSTALLED
python-venv: INSTALLED
---
PDK Version Verification Status: OK
---
Git Log (Last 3 Commits)

00caae2b 2023-08-22T19:35:09+03:00 Update PDK + Restore `RC_LAYER` variables (#1932) - Kareem Farid -  (HEAD -> master, tag: 2023.08.23, origin/master, origin/HEAD)
eaa6680c 2023-08-22T19:12:10+03:00 `run_designs.py`: Fix `mkdirp` not being called in some scenarios (#1945) - UMW -  ()
17559fa5 2023-08-22T18:54:25+03:00 CI Tweaks (#1951) - Mohamed Gaber -  ()
---
Git Remotes

origin	https://github.com/The-OpenROAD-Project/OpenLane (fetch)
origin	https://github.com/The-OpenROAD-Project/OpenLane (push)
```

### Reproduction material

[issue.tar.gz](https://github.com/The-OpenROAD-Project/OpenLane/files/12866966/issue.tar.gz)

### Relevant log output

```shell
OpenROAD 0cfb9a45bfb256c9af1a0500d4c97da0f145f54f 
This program is licensed under the BSD-3 license. See the LICENSE file for details.
Components of this program may be licensed under more restrictive licenses which must be honored.
[INFO]: Reading ODB at '/openlane/designs/vsdmemsoc/runs/RUN_2023.10.11_08.29.14/tmp/floorplan/6-tapcell.odb'…
define_corners Typical
read_liberty -corner Typical /home/pa1mantri/.volare/sky130A/libs.ref/sky130_fd_sc_hd/lib/sky130_fd_sc_hd__tt_025C_1v80.lib
read_liberty -corner Typical /openlane/designs/vsdmemsoc/src/macros/sky130_sram_1kbyte_1rw1r_32x256_8_TT_1p8V_25C.lib
Using 1e-12 for capacitance...
Using 1e+03 for resistance...
Using 1e-09 for time...
Using 1e+00 for voltage...
Using 1e-03 for current...
Using 1e-09 for power...
Using 1e-06 for distance...
Reading design constraints file at '/openlane/designs/vsdmemsoc/runs/RUN_2023.10.11_08.29.14/tmp/floorplan/3-initial_fp.sdc'…
No regex match found for cntlr defined in FP_PDN_MACRO_HOOKS
```

Pa1mantri, 2023-10-11:
[image]
Here is the command that i used to create the tarball

Pa1mantri, 2023-10-11:
{

    "DESIGN_NAME": "vsdmemsoc",

    "VERILOG_INCLUDE_DIRS": "dir::src/include",

    "VERILOG_FILES": "dir::src/*.v",

    "EXTRA_LEFS":     "dir::src/macros/sky130_sram_1kbyte_1rw1r_32x256_8.lef",

    "EXTRA_GDS_FILES": "dir::src/macros/sky130_sram_1kbyte_1rw1r_32x256_8.gds",

    "EXTRA_LIBS":      "dir::src/macros/sky130_sram_1kbyte_1rw1r_32x256_8_TT_1p8V_25C.lib",

    

            

    "CLOCK_PORT": "CLK",

    "CLOCK_NET": "CLK",

    "CLOCK_PERIOD": 10,

        

    "VDD_NETS": "vccd1",

    "GND_NETS": "vssd1",

     

    "FP_SIZING": "absolute",

    "PL_TARGET_DENSITY": 0.45,

    "RT_MAX_LAYER": "met4",

    "FP_PDN_MACRO_HOOKS": "mem vccd1 vssd1 vccd1 vssd1, cntlr vccd1 vssd1 vccd1 vssd1",

    "DIE_AREA": "0 0 1100 700",

    "MACRO_PLACEMENT_CFG": "dir::/src/macros/macro_placement.cfg",

    "FP_PIN_ORDER_CFG": "dir::/src/macros/pin_order.cfg",

    "BASE_SDC_FILE": "dir::src/vsdmemsoc_synth.sdc",

    "SYNTH_SIZING": 1,

    "SYNTH_FLAT_TOP": 1,

    "SYNTH_STRATEGY": "AREA 0",

    

    "MAGIC_DRC_USE_GDS": false,

    "QUIT_ON_MAGIC_DRC": false,

    "RUN_KLAYOUT_XOR": false,

    "RUN_MAGIC_DRC" :false,

    "DESIGN_IS_CORE": true

}

Pa1mantri, 2023-10-11:
[src.tar.gz](https://github.com/The-OpenROAD-Project/OpenLane/files/12869102/src.tar.gz)

d-m-bailey, 2023-10-11:
`cntlr` is not in the `macros/macro_placement.cfg` file.

Pa1mantri, 2023-10-11:
let me quickly run the flow again by including cntlr in the file.

Pa1mantri, 2023-10-11:
OpenROAD 0cfb9a45bfb256c9af1a0500d4c97da0f145f54f 

This program is licensed under the BSD-3 license. See the LICENSE file for details.

Components of this program may be licensed under more restrictive licenses which must be honored.

Placing the following macros:

{'cntlr': ['1000000', '1500000', 'N'], 'mem': ['1000000', '200000', 'N']}

Design name: vsdmemsoc

Placing mem

Traceback (most recent call last):

  File "/openlane/scripts/odbpy/manual_macro_place.py", line 137, in <module>

    manual_macro_place()

  File "/usr/local/lib/python3.6/site-packages/click/core.py", line 1128, in __call__

    return self.main(*args, **kwargs)

  File "/usr/local/lib/python3.6/site-packages/click/core.py", line 1053, in main

    rv = self.invoke(ctx)

  File "/usr/local/lib/python3.6/site-packages/click/core.py", line 1395, in invoke

    return ctx.invoke(self.callback, **ctx.params)

  File "/usr/local/lib/python3.6/site-packages/click/core.py", line 754, in invoke

    return __callback(*args, **kwargs)

  File "/openlane/scripts/odbpy/reader.py", line 76, in wrapper

    function(**kwargs)

  File "/openlane/scripts/odbpy/manual_macro_place.py", line 131, in manual_macro_place

    assert not macros, ("[ERROR]: Macros not found:", macros)

AssertionError: ('[ERROR]: Macros not found:', {'cntlr': ['1000000', '1500000', 'N']})

This is the error while running floorplan after changing the macro config file to     
cntlr 1000 1500 N
mem 1000 200  N

d-m-bailey, 2023-10-11:
Try adding the lef and gds for the macros to `EXTRA_LEFS` and `EXTRA_GDS_FILES`.

kareefardi, 2023-10-11:
If I understand correctly, this appears to be exactly the same issue which you reported before here: https://github.com/The-OpenROAD-Project/OpenLane/issues/1849#issuecomment-1591095120

Pa1mantri, 2023-10-12:
Yes, there, I removed the 'cntlr' instance, which resulted in LVS errors at the end. I would appreciate your assistance in determining which files need to be included in 'verilog_files_blackbox'. If I intend to blackbox 'cntlr.v', should I generate the '.lef' and '.gds' files for the 'cntlr' module in a separate flow?"

d-m-bailey, 2023-10-12:
@Pa1mantri I believe there are 2 types of verilog macros. soft macros and hard macros. Hard macros have a discrete physical view, while the verilog soft macros are often flattened and included in the parent layout.
For hard macros, you'll use `FP_PDN_MACRO_HOOKS`, for soft macros, I don't think this is needed.
For hard macros, you'll need to also have lef and gds views available. 
Hard macros will be listed in `VERILOG_FILES_BLACKBOX ` while soft macros will not. 
Hard macros can be placed with the `macro_placement.cfg` while soft macros can not.

Pa1mantri, 2023-10-12:
@d-m-bailey  Understood. Thank you for the explanation. Ran the flow again with modified config.json file. It worked without any LVS errors. There seems to be an error at the end of the flow when i ran the generarte_final_summary_report commmand
[errors.log](https://github.com/The-OpenROAD-Project/OpenLane/files/12882389/errors.log)

d-m-bailey, 2023-10-12:
@Pa1mantri There a strange character at the end of the lef filename.
```
FileNotFoundError: [Errno 2] No such file or directory: '/openlane/designs/vsdmemsoc/src/macros/sky130_sram_1kbyte_1rw1r_32x256_8.lef\x16'
```
See the `\x16`? 

You might try `cat -ev config.json` to see any strange characters.

Pa1mantri, 2023-10-13:
resolved the strange character issue, but the manufacturability report, metric report are not generated. Attaching the log files.
[errors.log](https://github.com/The-OpenROAD-Project/OpenLane/files/12888418/errors.log)
[openlane.log](https://github.com/The-OpenROAD-Project/OpenLane/files/12888420/openlane.log)

d-m-bailey, 2023-10-13:
Can you run this and let me know what it returns?
```
find designs/vsdmemsoc/runs/RUN_2023.10.12_10.55.15 -name *runtime.yaml
cat `!!`
```

Pa1mantri, 2023-10-13:
pa1mantri@pa1mantri:~/OpenLane$ find designs/vsdmemsoc/runs/RUN_2023.10.12_10.55.15 -name *runtime.yaml

cat `!!`

designs/vsdmemsoc/runs/RUN_2023.10.12_10.55.15/runtime.yaml

cat `find designs/vsdmemsoc/runs/RUN_2023.10.12_10.55.15 -name *runtime.yaml`

- status: 0 - openlane design prep

  runtime_s: 20.05

  runtime_ts: 0h0m20s53ms

- status: 1 - synthesis - yosys

  runtime_s: 81.01

  runtime_ts: 0h1m21s9ms

- status: 2 - sta - openroad

  runtime_s: 12.18

  runtime_ts: 0h0m12s178ms

- status: 3 - floorplan initialization - openroad

  runtime_s: 2.42

  runtime_ts: 0h0m2s418ms

- status: 4 - io_place - openlane

  runtime_s: 1.75

  runtime_ts: 0h0m1s750ms

- status: 6 - tap/decap insertion - openroad

  runtime_s: 2.0

  runtime_ts: 0h0m1s996ms

- status: 7 - pdn generation - openroad

  runtime_s: 4.02

  runtime_ts: 0h0m4s22ms

- status: 8 - global placement - openroad

  runtime_s: 53.51

  runtime_ts: 0h0m53s512ms

- status: 9 - sta - openroad

  runtime_s: 8.96

  runtime_ts: 0h0m8s955ms

- status: 10 - resizer design optimizations - openroad

  runtime_s: 20.8

  runtime_ts: 0h0m20s801ms

- status: 11 - detailed placement - openroad

  runtime_s: 12.25

  runtime_ts: 0h0m12s252ms

- status: 12 - sta - openroad

  runtime_s: 8.9

  runtime_ts: 0h0m8s903ms

- status: 13 - cts - openroad

  runtime_s: 80.03

  runtime_ts: 0h1m20s31ms

- status: 14 - sta - openroad

  runtime_s: 8.7

  runtime_ts: 0h0m8s698ms

- status: 15 - resizer design optimizations - openroad

  runtime_s: 22.44

  runtime_ts: 0h0m22s440ms

- status: 16 - sta - openroad

  runtime_s: 9.28

  runtime_ts: 0h0m9s281ms

- status: 17 - resizer timing optimizations - openroad

  runtime_s: 33.82

  runtime_ts: 0h0m33s822ms

- status: 18 - sta - openroad

  runtime_s: 9.35

  runtime_ts: 0h0m9s348ms

- status: 20 - write verilog - openroad

  runtime_s: 2.09

  runtime_ts: 0h0m2s85ms

- status: 21 - sta - openroad

  runtime_s: 6.7

  runtime_ts: 0h0m6s695ms

- status: 21 - global routing - openroad

  runtime_s: 6.7

  runtime_ts: 0h0m6s695ms

- status: 22 - fill insertion - openroad

  runtime_s: 4.21

  runtime_ts: 0h0m4s206ms

- status: 23 - detailed_routing - openroad

  runtime_s: 1622.69

  runtime_ts: 0h27m2s694ms

- status: 24 - wire lengths - openlane

  runtime_s: 2.42

  runtime_ts: 0h0m2s424ms

- status: 25 - parasitics extraction - openroad

  runtime_s: 14.39

  runtime_ts: 0h0m14s390ms

- status: 26 - sta - openroad

  runtime_s: 44.94

  runtime_ts: 0h0m44s939ms

- status: 27 - parasitics extraction - openroad

  runtime_s: 12.52

  runtime_ts: 0h0m12s524ms

- status: 28 - sta - openroad

  runtime_s: 44.59

  runtime_ts: 0h0m44s594ms

- status: 29 - parasitics extraction - openroad

  runtime_s: 10.34

  runtime_ts: 0h0m10s336ms

- status: 30 - sta - openroad

  runtime_s: 46.61

  runtime_ts: 0h0m46s606ms

- status: 31 - sta - openroad

  runtime_s: 12.81

  runtime_ts: 0h0m12s814ms

- status: 32 - ir drop report - openroad

  runtime_s: 13.25

  runtime_ts: 0h0m13s250ms

- status: 33 - gdsii - magic

  runtime_s: 63.65

  runtime_ts: 0h1m3s649ms

- status: 33 - lef - magic

  runtime_s: 25.84

  runtime_ts: 0h0m25s835ms

- status: 34 - spice extraction - magic

  runtime_s: 77.63

  runtime_ts: 0h1m17s626ms

- status: 36 - write verilog - openroad

  runtime_s: 5.96

  runtime_ts: 0h0m5s958ms

- status: 36 - write powered verilog - openlane

  runtime_s: 6.17

  runtime_ts: 0h0m6s173ms

- status: 37 - lvs - netgen

  runtime_s: 11.77

  runtime_ts: 0h0m11s766ms

- status: 38 - drc - magic

  runtime_s: 174.11

  runtime_ts: 0h2m54s108ms

- status: 39 - antenna check - openroad

  runtime_s: 8.06

  runtime_ts: 0h0m8s63ms

Pa1mantri, 2023-10-13:
when I run the same design in non-interactive mode, there are errors from the verilator during synthesis.
[linter.log](https://github.com/The-OpenROAD-Project/OpenLane/files/12890407/linter.log)
[errors.log](https://github.com/The-OpenROAD-Project/OpenLane/files/12890606/errors.log)

Some of the errors are "cannot find include file". Changed  the directory of those files(mentioned in the above linter.log) into the same directory where design source files are present and ran the flow again. This time there are 2 errors. 
[linter.log](https://github.com/The-OpenROAD-Project/OpenLane/files/12890669/linter.log)
[errors.log](https://github.com/The-OpenROAD-Project/OpenLane/files/12890677/errors.log)

Should it require any changes in the RTL? Outside the openlane flow when i use iverilog there is nothing wrong with the RTL of the design. Thank you.

d-m-bailey, 2023-10-13:
Sorry,  I don't think I'm going to be of any help here.

kareefardi, 2023-10-23:
@Pa1mantri. The error happens due to `include` statements in the RTL. Inside OpenLane flow these are not needed. You can either entirely remove them or wrap them around an `ifndef` statement and make use of `LINTER_DEFINES`

Pa1mantri, 2023-10-25:
Thank you for the response, removed all the includes and added the respective code. During the flow it results in warnings and timing errors in sram file. Sram file is already black-boxed and included "\\\sta-blackbox" in the file. Attaching the log files.Thank you.
[linter.log](https://github.com/The-OpenROAD-Project/OpenLane/files/13164736/linter.log)

kareefardi, 2023-10-26:
@Pa1mantri is the sram verilog file included under `VERILOG_FILES` or `VERILOG_FILES_BLACKBOX` ?

Pa1mantri, 2023-10-27:
It is under VERILOG_FILES_BLACKBOX. Here is the config.json file                                                                                                               {

    "DESIGN_NAME": "vsdmemsoc",

    "VERILOG_INCLUDE_DIRS": "dir::src/include",

    "VERILOG_FILES": "dir::src/*.v", 

    "EXTRA_LEFS":     "dir::src/macros/sky130_sram_1kbyte_1rw1r_32x256_8.lef",

    "EXTRA_GDS_FILES": "dir::src/macros/sky130_sram_1kbyte_1rw1r_32x256_8.gds",

    "EXTRA_LIBS":      "dir::src/macros/sky130_sram_1kbyte_1rw1r_32x256_8_TT_1p8V_25C.lib",

    "VERILOG_FILES_BLACKBOX": "dir::src/macros/sky130_sram_1kbyte_1rw1r_32x256_8.v",

    

            

    "CLOCK_PORT": "CLK",

    "CLOCK_NET": "CLK",

    "CLOCK_PERIOD": 10,

        

    "VDD_NETS": "vccd1",

    "GND_NETS": "vssd1",

     

    "FP_SIZING": "absolute",

    "PL_TARGET_DENSITY": 0.45,

    "RT_MAX_LAYER": "met4",

    "FP_PDN_MACRO_HOOKS": "mem vccd1 vssd1 vccd1 vssd1",

    "DIE_AREA": "0 0 1100 700",

    "MACRO_PLACEMENT_CFG": "dir::/src/macros/macro_placement.cfg",

    "FP_PIN_ORDER_CFG": "dir::/src/macros/pin_order.cfg",

    "BASE_SDC_FILE": "dir::src/vsdmemsoc_synth.sdc",

    "SYNTH_SIZING": 1,

    "SYNTH_FLAT_TOP": 1,

    "SYNTH_STRATEGY": "AREA 0",

    

    "MAGIC_DRC_USE_GDS": false,

    "QUIT_ON_MAGIC_DRC": false,

    "RUN_KLAYOUT_XOR": false,

    "RUN_MAGIC_DRC" :false,

    "DESIGN_IS_CORE": true

}

kareefardi, 2023-12-24:
@Pa1mantri files included under `VERILOG_FILES_BLACKBOX` should be blackboxed before checking for lint errors starting from `3376e5f897363bed322463e85ad251e90c4057c7` and forward. You are using an older version. Please update and try again.

Pa1mantri, 2023-12-25:
Hi @kareefardi, thank you for the response. I have updated OpenLane and rerun the flow. However, I encountered an error during the placement stage related to the maximum transition time. I am uploading the tarzip file. Thank you.
[issue_reproducible.tar.gz](https://github.com/The-OpenROAD-Project/OpenLane/files/13764645/issue_reproducible.tar.gz)

kareefardi, 2023-12-25:
@Pa1mantri the reproducible is missing some files? Is this a caravel user project design? Perhaps you can pass the source files and I will reproduce the issue

Pa1mantri, 2023-12-25:
Adding all the source files and config file in the attached zip file. Thank you.
[design.tar.gz](https://github.com/The-OpenROAD-Project/OpenLane/files/13765965/design.tar.gz)

kareefardi, 2024-01-16:
@Pa1mantri the max transition error is from ports inside the sram liberty files. If you modify them you should be able to get rid of this problem.

Pa1mantri, 2024-01-29:
@kareefardi Modified the max transition in liberty file. Flow completed without any error. Thank you.

kareefardi, 2024-01-29:
Great. Closing this issue now.

closed 2024-01-29 by kareefardi