The-OpenROAD-Project/OpenLane #2020: open_design.py script fails in caravel_user_project context

mattvenn, 2023-10-19:
### Description

The script works for files inside $OPENLANE_ROOT

But when trying to view a file inside caravel_user_project it fails:

$OPENLANE_ROOT/scripts/klayout/open_design.py --input-lef openlane/user_project_wrapper/runs/23_10_19_10_44/tmp/merged.nom.lef --lyt /home/zerotoasic/asic_tools/pdk/volare/sky130/versions/78b7bc32ddb4b6f14f76883c2e2dc5b5de9d1cbc/sky130A/libs.tech/klayout/tech/sky130A.lyt --lym /home/zerotoasic/asic_tools/pdk/volare/sky130/versions/78b7bc32ddb4b6f14f76883c2e2dc5b5de9d1cbc/sky130A/libs.tech/klayout/tech/sky130A.map --lyp /home/zerotoasic/asic_tools/pdk/volare/sky130/versions/78b7bc32ddb4b6f14f76883c2e2dc5b5de9d1cbc/sky130A/libs.tech/klayout/tech/sky130A.lyp ./openlane/user_project_wrapper/runs/23_10_19_10_44/tmp/floorplan/4-io.def

Error is:

Unable to open file: /home/zerotoasic/asic_tools/caravel_user_project/openlane/user_project_wrapper/runs/23_10_19_10_44/tmp/floorplan/openlane/user_project_wrapper/runs/23_10_19_10_44/tmp/merged.nom.lef (errno=2) in MainWindow.load_layout

The file it's looking for is here: 

openlane/user_project_wrapper/runs/23_10_19_10_44/tmp/merged.nom.lef

The path it's trying to open is wrong

### Expected Behavior

Should work with files in caravel as well as openlane

### Environment report

```shell
Kernel: Linux v6.2.0-34-generic
Distribution: ubuntu 22.04
Python: v3.10.12 (OK)
Container Engine: docker v24.0.5 (OK)
OpenLane Git Version: d054702b2cce04761cc2bc598f6b95c9d8ca7c6c
pip: INSTALLED
python-venv: INSTALLED
---
PDK Version Verification Status: OK
---
Git Log (Last 3 Commits)

d054702 2023-07-19T16:09:15+03:00 remove `unset_propagated_clock` (#1908) - passant5 -  (grafted, HEAD, tag: 2023.07.19)
---
Git Remotes

origin  https://github.com/The-OpenROAD-Project/OpenLane (fetch)
origin  https://github.com/The-OpenROAD-Project/OpenLane (push)
```

### Reproduction material

see above

### Relevant log output

```shell
see above
```

kareefardi, 2023-10-19:
@mattvenn Does it work with absolute paths for the merged lef?

mattvenn, 2023-10-19:
yes it works with absolute paths.

closed 2023-10-19 by mattvenn