The-OpenROAD-Project/OpenLane #2023: Make issue

vamsi-2312, 2023-10-22:
Hi,
While installing the pdks.
I am getting this error
```
[Errno 13] Permission denied: '/home/home/.volare/volare'
make: *** [Makefile:143: pdk] Error 255
```

I have tried,
https://github.com/The-OpenROAD-Project/OpenLane/issues/1770

Can anyone help.

Thank you

donn, 2023-10-22:
Try:

```sh
sudo rm -rf "/home/home/.volare"
make pdk
```

Be careful because if you mess up the first command even slightly you may end up deleting your entire home folder.

SahilPrabhu, 2024-02-17:
> ```shell
> sudo rm -rf "/home/home/.volare"
> ```

i had this error
The read operation timed out
make: *** [Makefile:143: pdk] Error 255
and that got fixed!
Thank you so much!

donn, 2024-02-18:
Given your response and the thumbs up react, I'm going to assume this is resolved. 🙂

closed 2024-02-18 by donn