The-OpenROAD-Project/OpenLane #2025: Global Routing stage fails with ERROR GRT-0076 : Net not properly covered

dineshannayya, 2023-10-25:
### Description

I see recent version of openlane flow fails with GRT error During Global Routing phase.  I need

This database use to work well with older version of openlane : e10820ecf53c08df9ca04447ac8e0428b2141ef8

### Expected Behavior

Flow should route the signal between block to top-level.

### Environment report

```shell
<pre>open_pdks e6f9c8876da77220403014b116761b0b2d79aab4
Kernel: Linux v5.15.0-87-generic
Distribution: ubuntu 20.04
Python: v3.8.10 (OK)
Container Engine: docker v23.0.3 (OK)
OpenLane Git Version: dc5af9827620a2189f65fa42d9965e7a3d7b4695
pip: INSTALLED
python-venv: INSTALLED
---
PDK Version Verification Status: MISMATCH
The version of open_pdks used in building the PDK does not match the version OpenLane was tested on (installed: e6f9c8876da77220403014b116761b0b2d79aab4, tested: dd7771c384ed36b91a25e9f8b314355fc26561be)
This may introduce some issues. You may want to re-install the PDK by invoking `make pdk`.
---
Git Log (Last 3 Commits)

dc5af982 2023-10-20T14:26:03+03:00 ci: fix issue with pip package (#2022) - Vitor Bandeira -  (HEAD -&gt; master, tag: 2023.10.21, origin/master, origin/HEAD)
747bdbbd 2023-10-19T17:34:52+03:00 Makefile Improvements + Banner Update (#2018) - Mohamed Gaber -  (tag: 2023.10.20)
7ea7a2ae 2023-10-15T17:01:59+03:00 Fix `FP_PIN_ORDER_CFG` being overriden (#2017) - Kareem Farid -  (tag: 2023.10.16)
---
Git Remotes

origin	https://github.com/The-OpenROAD-Project/OpenLane (fetch)
origin	https://github.com/The-OpenROAD-Project/OpenLane (push)
</pre>
```

### Reproduction material

[issue_reproducible.tar.gz](https://github.com/The-OpenROAD-Project/OpenLane/files/13162278/issue_reproducible.tar.gz)

### Relevant log output

```shell
[INFO]: Run Directory: /home/dinesha/workarea/opencore/git/openframe_riscduino_dcore/openlane/openframe_project_wrapper/runs/RUN_2023.10.25_06.38.01
[INFO]: Saving runtime environment...
[INFO]: Preparing LEF files for the nom corner...
[INFO]: Preparing LEF files for the min corner...
[INFO]: Preparing LEF files for the max corner...
[INFO]: Running Synthesis (log: openframe_project_wrapper/runs/RUN_2023.10.25_06.38.01/logs/synthesis/1-synthesis.log)...
[INFO]: Running Single-Corner Static Timing Analysis (log: openframe_project_wrapper/runs/RUN_2023.10.25_06.38.01/logs/synthesis/2-sta.log)...
[INFO]: Creating a netlist with power/ground pins.
[INFO]: Running Initial Floorplanning (log: openframe_project_wrapper/runs/RUN_2023.10.25_06.38.01/logs/floorplan/3-initial_fp.log)...
[INFO]: Floorplanned with width 3086.6 and height 4683.84.
[INFO]: Running IO Placement (log: openframe_project_wrapper/runs/RUN_2023.10.25_06.38.01/logs/floorplan/4-io.log)...
[INFO]: Applying DEF template...
[INFO]: Performing Manual Macro Placement (log: openframe_project_wrapper/runs/RUN_2023.10.25_06.38.01/logs/placement/5-macro_placement.log)...
[INFO]: Power planning with power {vccd1 vdda1 } and ground {vssd1 vssa1 }...
[INFO]: Generating PDN (log: openframe_project_wrapper/runs/RUN_2023.10.25_06.38.01/logs/floorplan/6-pdn.log)...
[INFO]: Performing Random Global Placement (log: openframe_project_wrapper/runs/RUN_2023.10.25_06.38.01/logs/placement/7-global.log)...
[INFO]: Skipping Placement Resizer Design Optimizations.
[INFO]: Running Detailed Placement (log: openframe_project_wrapper/runs/RUN_2023.10.25_06.38.01/logs/placement/8-detailed.log)...
[INFO]: Running Single-Corner Static Timing Analysis (log: openframe_project_wrapper/runs/RUN_2023.10.25_06.38.01/logs/placement/9-dpl_sta.log)...
[INFO]: Running Clock Tree Synthesis (log: openframe_project_wrapper/runs/RUN_2023.10.25_06.38.01/logs/cts/10-cts.log)...
[INFO]: Running Single-Corner Static Timing Analysis (log: openframe_project_wrapper/runs/RUN_2023.10.25_06.38.01/logs/cts/11-cts_sta.log)...
[INFO]: Skipping Placement Resizer Timing Optimizations.
[INFO]: Custom DEF template step...
[INFO]: Skipping Global Routing Resizer Design Optimizations.
[INFO]: Skipping Global Routing Resizer Timing Optimizations.
[INFO]: Running Global Routing (log: openframe_project_wrapper/runs/RUN_2023.10.25_06.38.01/logs/routing/12-global.log)...
[ERROR]: during executing openroad script /openlane/scripts/openroad/groute.tcl
[ERROR]: Log: openframe_project_wrapper/runs/RUN_2023.10.25_06.38.01/logs/routing/12-global.log
[ERROR]: Last 10 lines:
[WARNING GRT-0350] Missing route to pin u_qspi_master/spi_oen[3].
[WARNING GRT-0350] message limit reached, this message will no longer print
[INFO GRT-0197] Via related to pin nodes: 18674
[INFO GRT-0198] Via related Steiner nodes: 22
[INFO GRT-0199] Via filling finished.
[INFO GRT-0111] Final number of vias: 26644
[INFO GRT-0112] Final usage 3D: 335253
[ERROR GRT-0076] Net analog_io[15] not properly covered.
Error: grt.tcl, 29 GRT-0076
child process exited abnormally

[ERROR]: Creating issue reproducible...
[INFO]: Saving runtime environment...
[INFO]: Reproducible packaged at 'openframe_project_wrapper/runs/RUN_2023.10.25_06.38.01/issue_reproducible'.
```

kareefardi, 2023-10-25:
@dineshannayya The issue reproducible is missing some files. These are all the files in the tarball:
```
issue_reproducible/
issue_reproducible/env.lldb
issue_reproducible/openlane/
issue_reproducible/openlane/scripts/
issue_reproducible/openlane/scripts/openroad/
issue_reproducible/openlane/scripts/openroad/groute.tcl
issue_reproducible/run.tcl
issue_reproducible/env.gdb
issue_reproducible/run.sh
```
Can you upload the design source files?. Also, how were you running the flow?

dineshannayya, 2023-10-25:
@kareefardi  
I am trying port my riscduino design into efabless openframe structure & I am using efabless openlane flow.
I am facing this issue in top-level, which need .gds from multiple sub-modules. 
I see compressed top-level run folder is around 5GB. Do you want complete Run folder or any specific data from RUN directory ?

dineshannayya, 2023-10-25:
[RUN_2023.10.25_11.01.34.1.tar.gz](https://github.com/The-OpenROAD-Project/OpenLane/files/13165254/RUN_2023.10.25_11.01.34.1.tar.gz)
Here is Run folder excluding synthesis and floor plan data

kareefardi, 2023-10-25:
@dineshannayya I want to be able to rerun your design and reproduce the issue. Since the reproducible is not working this requires the source files (config.json/config.tcl and it's dependencies). If this is pushed to some repo, you can point me to it. This is much easier than working with a run directory.

dineshannayya, 2023-10-25:
My Repo is: https://github.com/dineshannayya/openframe_riscduino_dcore

kareefardi, 2023-10-25:
@dineshannayya The LEF files needed for `openframe_project_wrapper` are not inside the repo (also the GDS but this is not really needed to reproduce the error)

dineshannayya, 2023-10-25:
@kareefardi I have pushed the all the lef files into git repo

dineshannayya, 2023-10-25:
@kareefardi look like i have understood the issue. One of the macro placement was outside the core area. when i moved macro within core area. Openlane flow is completed successfully.
Openlane flow Failing Macro Placement : -  [image]
Openlane passing Macro Placement:- [image]

Tool should have reported this issue during placement stage, instead of routing phase for easy debug

closed 2023-11-19 by dineshannayya